FastLED/FastLED #1570: Documentation Page 404ed

DexterP54, 2023-12-03:
https://fastled.io/docs/3.1/index.html

marmilicious, 2023-12-04:
@DexterP54  Thank you for officially opening an issue for this.

@samguyer @kriegsman  Seems if the link went to [](https://fastled.io/docs/) then it would always pick up the current version. But maybe there's reasons to have to hard coded to a specific version.

These pages need the link updated:
https://github.com/FastLED/FastLED/wiki/Frequently-Asked-Questions
https://github.com/FastLED/FastLED/wiki/Overview
https://github.com/FastLED/FastLED/wiki/API-Reference

I just added a link to https://fastled.io/docs/ on the FastLED reddit side bar.

samguyer, 2023-12-05:
We can update the GitHub Wiki with the more general link. I'm not sure if it's possible to retrieve the old documentation (the old Doxygen pages). I'd have to look into it.

dmadison, 2023-12-10:
It's possible to checkout an older commit and re-generate the documentation from there. We could also check out an older commit of the `gh-pages` branch and re-add the files. Although considering how broken and incomplete the documentation was before the rewrite I'm not sure why we'd want to.

The reason the current documentation is not segmented per version is for consistency (the latest docs are always in the same place) and for Doxygen use in third party libraries, because generating a Doxygen tag file requires a URI without a redirect.

Going forward, in my view if semantic versioning is followed there should be no reason to keep minor versions of the documentation online because there are no breaking changes. It might be wise to keep a set of 3.x documentation around if 4.0 is ever released, but that's a future issue.

nickyonge, 2023-12-15:
Just chiming in to say this is still an issue, just ran into it on the API reference page, had my moment of confusion, and a pre-emptive thankyou to whoever gets the fixed link refs merged in! 🙏💖

zackees, 2024-12-16:
Remind me again what's 404ing?

marmilicious, 2024-12-16:
The links on those pages to the API documentation. They're hard coded to http://fastled.io/docs/3.1/index.html which breaks.

DexterP54, 2024-12-16:
Is someone gonna fix it any year soon ? 😵‍💫

DexterP54, 2024-12-17:
The page is not there !

The link is Broken

Cannot Reach web page

Can't see the information

zackees, 2024-12-18:
Okay i think it's fixed now, let me know if I missed anything.

marmilicious, 2024-12-18:
Looks good, thank you

closed 2024-12-19 by zackees