FastLED/FastLED #1572: ESP32-C2

BeerHunter1337, 2023-12-12:
Hello!

I've just got my hands on the ESP32-C2 board and realized it has neither RTM nor I2S onboard.
(Alpha version of arduino-esp32 just added a support for C2 of [https://github.com/espressif/arduino-esp32/releases/tag/3.0.0-alpha3])

Any chance someone can assist in implementing a fastpin version for ESP32 for clockless chips like it was done for ESP8266?

samguyer, 2023-12-18:
@BeerHunter1337 I'd be happy to take a look at this "port". It would be great to have a bit-banged option for ESP32, not just for devices like the C2.

shlomozippel, 2024-05-02:
espressif actually have a SPI based implementation for ws281x pixels: https://components.espressif.com/components/espressif/led_strip

I've had success combining FastLED with the espressif package. I use FastLED for various pixel manipulation / lib8tion and then output to the strip using `led_strip`. It is pretty limited: can't change color order, or gamma tables, and it's blocking. But it's a good starting point for the C2

samguyer, 2024-05-02:
It's interesting how they are leaning hard into supporting LED strips -- much more than when I started out on this project.

zackees, 2024-08-28:
I attempted a fix on this. But realized that we don't have any clockless bit banged controllers that work.

Very hard to even get the c2 to compile. Arduino does not support it and neither does platformio

zackees, 2024-09-20:
> espressif actually have a SPI based implementation for ws281x pixels: https://components.espressif.com/components/espressif/led_strip
>
>
>
> I've had success combining FastLED with the espressif [...]

Hmmm, I recently ported the led_strip component to C++ in prep for 5.1 driver support, but stripped out the SPI driver and anything related to idf 4.4.

But if SPI support means our C2 board build goes green then I may reconsider this.

zackees, 2024-09-20:
How can I make the arduino / platform io toolchain invoke the C compiler? It seems to ignore c files.

zackees, 2024-09-21:
Yeah okay.

Yeah it sucks.

Arduino cli doesn't compile C either.

Such a tragedy.

zackees, 2024-09-30:
> espressif actually have a SPI based implementation for ws281x pixels: https://components.espressif.com/components/espressif/led_strip
>
>
>
> I've had success combining FastLED with the espressif [...]

I was able to port the led_strip library to C++ then I tore it apart and added all the missing features like async drawing, RMT pools when the number of strips is greater than the number of rmt channels, customizable encoder for all the clickless drivers, etc, DMA, an interrupt for encoding that's higher than 0.

It's in FastLED master branch and going out with the next release.

zackees, 2024-09-30:
led strip 2.2.5 has RGBW.

Surprising they didn't add the async stuff in already. I added it and it was dead simple.

Fun story, I ended up modifying one of their structs in the framework directory, thinking it was part of led_strip component. The new struct passed into their library just fine. But it busted pin 9 and that pin no longer works for rmt 5, but does work for rmt 4.4. Power cycling does not fix the issue.

zackees, 2024-09-30:
Legacy cares about 4.4. Some people have pinned dependencies, mostly in platformio. The problem will go away on its own and my strategy at this point is to pour cement over the 4.4 driver so it doesn't move.

closed 2025-01-13 by zackees