FastLED/FastLED #1580: Is there a way to compare/convert a string into the controllers "class"? definitions?

DanielLester83, 2023-12-27:
I'm unfamilar with the way addLed uses <CLASSNAME?, ...> can I convert a selectable char* to a class name?

sorry about the formatting.

This works but is repetitive:
`if( strncmp ( sled.getValue(), "NEOPIXEL"    , 16 ) == 0 ) FastLED.addLeds<NEOPIXEL     , STRIP_DATA_PIN     >(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "SM16703"     , 16 ) == 0 ) FastLED.addLeds<SM16703      , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "TM1829"      , 16 ) == 0 ) FastLED.addLeds<TM1829       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "TM1812"      , 16 ) == 0 ) FastLED.addLeds<TM1812       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "TM1809"      , 16 ) == 0 ) FastLED.addLeds<TM1809       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "TM1804"      , 16 ) == 0 ) FastLED.addLeds<TM1804       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "TM1803"      , 16 ) == 0 ) FastLED.addLeds<TM1803       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "UCS1903"     , 16 ) == 0 ) FastLED.addLeds<UCS1903      , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "UCS1903B"    , 16 ) == 0 ) FastLED.addLeds<UCS1903B     , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);   
if( strncmp ( sled.getValue(), "UCS1904"     , 16 ) == 0 ) FastLED.addLeds<UCS1904      , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "UCS2903"     , 16 ) == 0 ) FastLED.addLeds<UCS2903      , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "WS2812"      , 16 ) == 0 ) FastLED.addLeds<WS2812       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);  
if( strncmp ( sled.getValue(), "WS2852"      , 16 ) == 0 ) FastLED.addLeds<WS2852       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "WS2812B"     , 16 ) == 0 ) FastLED.addLeds<WS2812B      , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "GS1903"      , 16 ) == 0 ) FastLED.addLeds<GS1903       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);   
if( strncmp ( sled.getValue(), "SK6812"      , 16 ) == 0 ) FastLED.addLeds<SK6812       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "SK6822"      , 16 ) == 0 ) FastLED.addLeds<SK6822       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "APA106"      , 16 ) == 0 ) FastLED.addLeds<APA106       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS); 
if( strncmp ( sled.getValue(), "PL9823"      , 16 ) == 0 ) FastLED.addLeds<PL9823       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "SK6822"      , 16 ) == 0 ) FastLED.addLeds<SK6822       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "WS2811"      , 16 ) == 0 ) FastLED.addLeds<WS2811       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS); 
if( strncmp ( sled.getValue(), "PL9823"      , 16 ) == 0 ) FastLED.addLeds<PL9823       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "SK6822"      , 16 ) == 0 ) FastLED.addLeds<SK6822       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "WS2811"      , 16 ) == 0 ) FastLED.addLeds<WS2811       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS); 
if( strncmp ( sled.getValue(), "WS2813"      , 16 ) == 0 ) FastLED.addLeds<WS2813       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "APA104"      , 16 ) == 0 ) FastLED.addLeds<APA104       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "WS2811_400"  , 16 ) == 0 ) FastLED.addLeds<WS2811_400   , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS); 
if( strncmp ( sled.getValue(), "GE8822"      , 16 ) == 0 ) FastLED.addLeds<GE8822       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "GW6205"      , 16 ) == 0 ) FastLED.addLeds<GW6205       , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "GW6205_400"  , 16 ) == 0 ) FastLED.addLeds<GW6205_400   , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS); 
if( strncmp ( sled.getValue(), "LPD1886"     , 16 ) == 0 ) FastLED.addLeds<LPD1886      , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "LPD1886_8BIT", 16 ) == 0 ) FastLED.addLeds<LPD1886_8BIT , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS); 
if( strncmp ( sled.getValue(), "LPD6803"     , 16 ) == 0 ) FastLED.addLeds<LPD6803      , STRIP_DATA_PIN, STRIP_CLOCK_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "LPD8806"     , 16 ) == 0 ) FastLED.addLeds<LPD8806      , STRIP_DATA_PIN, STRIP_CLOCK_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "WS2801"      , 16 ) == 0 ) FastLED.addLeds<WS2801       , STRIP_DATA_PIN, STRIP_CLOCK_PIN, RGB>(leds, STRIP_NUM_LEDS); 
if( strncmp ( sled.getValue(), "WS2803"      , 16 ) == 0 ) FastLED.addLeds<WS2803       , STRIP_DATA_PIN, STRIP_CLOCK_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "SM16716"     , 16 ) == 0 ) FastLED.addLeds<SM16716      , STRIP_DATA_PIN, STRIP_CLOCK_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "P9813"       , 16 ) == 0 ) FastLED.addLeds<P9813        , STRIP_DATA_PIN, STRIP_CLOCK_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "DOTSTAR"     , 16 ) == 0 ) FastLED.addLeds<DOTSTAR      , STRIP_DATA_PIN, STRIP_CLOCK_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "APA102"      , 16 ) == 0 ) FastLED.addLeds<APA102       , STRIP_DATA_PIN, STRIP_CLOCK_PIN, RGB>(leds, STRIP_NUM_LEDS);
if( strncmp ( sled.getValue(), "SK9822"      , 16 ) == 0 ) FastLED.addLeds<SK9822       , STRIP_DATA_PIN, STRIP_CLOCK_PIN, RGB>(leds, STRIP_NUM_LEDS);`

ideally I'd like something like this:
`
/
switch( sled.getValue ){
  case "SK9822" :
  case "DOTSTAR" :
  case "P9813" :
  case "SM16716" : 
  case "WS2803" :
  case "WS2801" :
  case "LPD8806" :
  case "LPD6803" :
  case "APA102" :  FastLED.addLeds<  ??sled.getValue??  , STRIP_DATA_PIN, STRIP_CLOCK_PIN, RGB>(leds, STRIP_NUM_LEDS);
                             break;
 case "NEOPIXEL": FastLED.addLeds<NEOPIXEL     , STRIP_DATA_PIN     >(leds, STRIP_NUM_LEDS);
                              break;
 default                : FastLED.addLeds<  ??sled.getValue??   , STRIP_DATA_PIN, RGB>(leds, STRIP_NUM_LEDS); 
}
`

dmadison, 2024-01-03:
The first way you wrote it is exactly how I'd do it, although I'd put the relevant `strncmp` block into a macro and use the preprocessor to do the heavy lifting:

```cpp
#define ADD_LED_TYPE(x) if( strncmp(sled.getValue(), #x, 16 ) == 0 ) FastLED.addLeds<x, STRIP_DATA_PIN>(leds, STRIP_NUM_LEDS)
```

Then usage is:

```cpp
ADD_LED_TYPE(WS2812B);
ADD_LED_TYPE(SK6812);
...
```

---

To do it the second way you propose you'd need a go-between that would associate the strings with the LED datatypes. You can build that, but the middleman is going to be another function that looks more-or-less like a chain of `strncmp` functions anyways.

kriegsman, 2024-01-15:
The preprocessor seems like a helpful tool here.
For general questions about FastLED and help programming with it, check out https://www.reddit.com/r/FastLED/ -- hope to see you there!

closed 2024-01-15 by kriegsman