FastLED/FastLED #1587: Panic on 32MB 8MB ESP32-S3-DevKitC1 (Illegal memory access)

GameTec-live, 2024-01-07:
Hardware: 32MB FLASH 8MB PRAM ESP32-S3-DevKitC1(ESP32-S3-DevKitC-1-N32R8V)

Im trying to get nightdriver LED running on my ESP32-S3 but it seems like the current error is tracing back to a illegal memory access performed by FastLED (thats why im opening this issue)

Log:
https://github.com/PlummersSoftwareLLC/NightDriverStrip/files/13836222/debug-log.log

Nightdriver issue:
https://github.com/PlummersSoftwareLLC/NightDriverStrip/issues/580

samguyer, 2024-01-15:
Hello @GameTec-live ! I was just reviewing the discussion on the Nightdriver issue board. Were you able to confirm whether or not mCur is running off the end of the mPixelData array? The array is sized to hold all of your pixel data, and mSize is the size in bytes. So, for RGB pixels, that should be 3X the number of pixels (in your case, 4500 bytes). mCur keeps track of the position in that array as the data is sent.

GameTec-live, 2024-01-16:
Ah, that makes sense, thanks! I'll try and confirm that this evening...

GameTec-live, 2024-01-16:
Ok, so mCur goes up to (and including) 4500 and then the for loop ends (L497) and it starts to go into the following code ("mWhichHalf++")..

GameTec-live, 2024-02-01:
@samguyer ?

GameTec-live, 2024-09-05:
I tried with the latest fastled version (3.7.5) today, i still get a Panic, so this doesnt seem to be resolved?

zackees, 2024-09-06:
Do you have the built in RMT driver specifically enabled by any chance? If not, can you also try running this with that driver enabled so we can get a better understanding on which half of the RMT driver interface is messing up?

Also, can you upload your sketch to try and repro on our end?

zackees, 2024-09-06:
and how many leds are you pushing. Something is exceptional to cause this issue. It's either the number of leds, driving multiple strips or a unique pin set you are using. I can think of any other reason.

GameTec-live, 2024-09-06:
Well, as said, im using NightdriverLED / NightDriverStrip and I want to drive a 50x30 Matrix, so a strip with 1500 LEDs. The pin used is the default, Pin 5.

zackees, 2024-09-06:
Gotcha, so the exceptional case here is 1500 pixels.

One possibility here is the block size.

https://github.com/FastLED/FastLED/blob/master/src/platforms/esp/32/idf4_rmt_impl.cpp#L320C33-L320C39

I see here a default malloc allocation. Exactly where this memory is being allocated from isn't explicit nor clear. It's possible that past a certain size with the default allocator there are issues if the allocator chooses a different memory zone to allocate from. 

Have you ever used the esp heap alloc function which allows you to specify the area of memory that is allocated from?

I suspect that heap allocating from the DMA zone might give you better results. There is also PSRAM mode.

Btw are you just one pin? That many leds from
one pin will give you frame rate issues and the RMT driver is designed to drive leds in parallel. For the WS2812 chipset you can do 550 pixels at 60 fps. At 1500 pixels you are going to see 20fps if you drive it with just one pin.

I recommend trying to clone our repo and hitting the compile button and poking at the memory allocation function and seeing if being more explicit about the memory type fixes your problem. And if it does please let me know and I'll merge in your changes into the next release.

zackees, 2024-09-06:
Can't repro this issue on my XIAO s3. I pushed it to 3k pixels. Used pin5. Works fine.

At this point there are two options:

  1. Bribe me with a board and full access to a repo to reproduce this issue -or-
  2. Clone our repo, modify [dev/dev.ino](dev/dev.ino) to match your project and hit compile and upload and then poke at the FastLED [code](src/platforms/esp32/idf4_rmt_impl.cpp) until it works and send a PR request. I will merge in a fix pretty fast.

zackees, 2024-10-01:
You can try the new RMT5 driver that's been merged into main and see if that fixes your issue.

closed 2024-11-14 by zackees