FastLED/FastLED #1590: deftype simplepatternlist not defining?

jester101YT, 2024-01-22:
i want to make an andon/stack light i got the code down where i can controle each pixel individualy but the rainbow pattern doesnt want to define i have copy pasted every thing from the demo sketch but still no dice can somebody help me clean up the code and fix the defining error
[image]

marmilicious, 2024-01-22:
We mainly use GitHub for bug reports and improvement requests. For general help and coding questions please use the FastLED reddit group. 

https://www.reddit.com/r/FastLED

marmilicious, 2024-01-24:
@samguyer or @kriegsman   This one can be closed as @jester101YT found us and posted on reddit. 

(or @jester101YT you can also close this)

kriegsman, 2024-03-12:
Thanks!  Closing this issue here.

closed 2024-03-12 by kriegsman