FastLED/FastLED #1591: Arduino Local + Web editor not compiling smallest test code / clockless_esp8266.h:85:18: error: expected ';' at end of member declaration

h0jeZvgoxFepBQ2C, 2024-01-25:
I try to run the mentioned test code in the readme:

```c
#include <FastLED.h>
#define NUM_LEDS 60
CRGB leds[NUM_LEDS];
void setup() { FastLED.addLeds<NEOPIXEL, 6>(leds, NUM_LEDS); }
void loop() {
	leds[0] = CRGB::White; FastLED.show(); delay(30);
	leds[0] = CRGB::Black; FastLED.show(); delay(30);
}
```

But I only get this output (in web editor)

```output
/usr/local/bin/arduino-cli compile --fqbn esp8266:esp8266:generic:CrystalFreq=26,FlashFreq=40,FlashMode=dout,ResetMethod=ck,baud=115200,dbg=Disabled,eesz=512K,exception=disabled,ip=lm2f,led=2,lvl=None____,vt=flash,wipe=none,xtal=80 --build-cache-path /tmp --output-dir /tmp/1550360353/build --build-path /tmp/arduino-build-0DB3197658358C37F42AD4B21DA3F26A /tmp/1550360353/sketch_jan3a

In file included from /home/builder/Arduino/libraries/fastled_3_6_0/src/platforms/esp/8266/fastled_esp8266.h:9:0,

from /home/builder/Arduino/libraries/fastled_3_6_0/src/platforms.h:40,

from /home/builder/Arduino/libraries/fastled_3_6_0/src/FastLED.h:62,

from /tmp/1550360353/sketch_jan3a/sketch_jan3a.ino:1:

/home/builder/Arduino/libraries/fastled_3_6_0/src/platforms/esp/8266/clockless_esp8266.h:85:18: error: expected ';' at end of member declaration

static uint32_t IRAM_ATTR showRGBInternal(PixelController<RGB_ORDER> pixels) {

^

/home/builder/Arduino/libraries/fastled_3_6_0/src/platforms/esp/8266/clockless_esp8266.h:85:77: error: ISO C++ forbids declaration of 'showRGBInternal' with no type [-fpermissive]

static uint32_t IRAM_ATTR showRGBInternal(PixelController<RGB_ORDER> pixels) {

^

In file included from /home/builder/Arduino/libraries/fastled_3_6_0/src/platforms/esp/8266/fastled_esp8266.h:10:0,

from /home/builder/Arduino/libraries/fastled_3_6_0/src/platforms.h:40,

from /home/builder/Arduino/libraries/fastled_3_6_0/src/FastLED.h:62,

from /tmp/1550360353/sketch_jan3a/sketch_jan3a.ino:1:

/home/builder/Arduino/libraries/fastled_3_6_0/src/platforms/esp/8266/clockless_block_esp8266.h:111:18: error: expected ';' at end of member declaration

static uint32_t IRAM_ATTR showRGBInternal(PixelController<RGB_ORDER, LANES, PORT_MASK> &allpixels) {

^

/home/builder/Arduino/libraries/fastled_3_6_0/src/platforms/esp/8266/clockless_block_esp8266.h:111:99: error: ISO C++ forbids declaration of 'showRGBInternal' with no type [-fpermissive]

static uint32_t IRAM_ATTR showRGBInternal(PixelController<RGB_ORDER, LANES, PORT_MASK> &allpixels) {

^

In file included from /home/builder/Arduino/libraries/fastled_3_6_0/src/FastLED.h:75:0,

from /tmp/1550360353/sketch_jan3a/sketch_jan3a.ino:1:

/home/builder/Arduino/libraries/fastled_3_6_0/src/fastspi.h:157:23: note: #pragma message: No hardware SPI pins defined. All SPI access will default to bitbanged output

# pragma message "No hardware SPI pins defined. All SPI access will default to bitbanged output"

^

In file included from /home/builder/Arduino/libraries/fastled_3_6_0/src/FastLED.h:58:0,

from /tmp/1550360353/sketch_jan3a/sketch_jan3a.ino:1:

/home/builder/Arduino/libraries/fastled_3_6_0/src/fastpin.h: In instantiation of 'class FastPin<6u>':

/home/builder/Arduino/libraries/fastled_3_6_0/src/platforms/esp/8266/clockless_esp8266.h:21:49: required from 'class ClocklessController<6, 20, 50, 30, (EOrder)66u, 0, false, 85>'

/home/builder/Arduino/libraries/fastled_3_6_0/src/chipsets.h:654:7: required from 'class WS2812Controller800Khz<6u, (EOrder)66u>'

/home/builder/Arduino/libraries/fastled_3_6_0/src/FastLED.h:119:34: required from 'class NEOPIXEL<6u>'

/home/builder/Arduino/libraries/fastled_3_6_0/src/FastLED.h:366:28: required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = NEOPIXEL; unsigned char DATA_PIN = 6u]'

/tmp/1550360353/sketch_jan3a/sketch_jan3a.ino:4:59: required from here

/home/builder/Arduino/libraries/fastled_3_6_0/src/fastpin.h:261:2: error: static assertion failed: Invalid pin specified

static_assert(validpin(), "Invalid pin specified");

^

Error during build: exit status 1
```

Any idea how to solve this?

marmilicious, 2024-01-25:
It says "Invalid pin specified". Have a look at this info.

https://github.com/FastLED/FastLED/wiki/ESP8266-notes

h0jeZvgoxFepBQ2C, 2024-01-25:
ah i found the issue myself, i had an old https://github.com/esp8266/Arduino version installed. After updating, it worked fine without problems.

closed 2024-01-25 by h0jeZvgoxFepBQ2C