FastLED/FastLED #1593: FastLED on RP2040 cli redefined

Gavin-Perry, 2024-02-06:
This warning seems to be new. (3.x) I now get a warning:

>  In file included from C:\Users\gavin\OneDrive\Documents\Arduino\libraries\FastLED\src/led_sysdefs.h:38,
>                  from C:\Users\gavin\OneDrive\Documents\Arduino\libraries\FastLED\src/FastLED.h:51,
>                  from C:\Users\gavin\OneDrive\Documents\Arduino\RP2040-7Pixel\RP2040-7Pixel.ino:10:
> C:\Users\gavin\OneDrive\Documents\Arduino\libraries\FastLED\src/platforms/arm/rp2040/led_sysdefs_arm_rp2040.h:87: warning: "cli" redefined
>    87 | #define cli() (saved_interrupt_status = save_and_disable_interrupts())
>       | 
> In file included from C:\Users\gavin\AppData\Local\Temp\arduino\sketches\2E7CED4A54FD27679058BBB361D7A6DC\sketch\RP2040-7Pixel.ino.cpp:1:
> C:\Users\gavin\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.3.1\cores\rp2040/Arduino.h:74: note: this is the location of the previous definition
>    74 | #define cli() noInterrupts()

Same for sei()

Also 
> C:\Users\gavin\OneDrive\Documents\Arduino\libraries\FastLED\src/fastspi.h:172:23: note: '#pragma message: Forcing software SPI - no hardware SPI for you!'
>  172 | #      pragma message "Forcing software SPI - no hardware SPI for you!"

Should have worked with PIO. Did I mess something up?

Gavin-Perry, 2024-02-14:
Although the title doesn't match it looks like this issue is covered in previous issue here:
["Error: invalid offset, value too big" when building for Raspberry Pi Pico (RP2040)](https://github.com/FastLED/FastLED/issues/1481#top)
#1481

JonnyHaystack, 2024-02-14:
> Although the title doesn't match it looks like this issue is covered in previous issue here: ["Error: invalid offset, value too big" when building for Raspberry Pi Pico (RP2040)](https://github.com/FastLED/FastLED/issues/1481#top) #1481

This issue is actually a duplicate of #1541 which should be fixed by #1575

Gavin-Perry, 2024-02-14:
Thanks for the pointer to the fix.