FastLED/FastLED #1594: addLeds pass DATA_PIN from function argument

Sladerix, 2024-02-07:
Hi everyone I'm writing my wrapper class that uses FastLED

I made `init` method that should initialize everything, including the FastLED strip.

## const
```cpp
void Strips::init(std::string input, const uint8_t pin) {
    strips = getStrips(input);
    led_number = getNumLeds(strips);
    leds = new CRGB[led_number];
    FastLED.addLeds<WS2812B, pin, GRB>(leds, led_number);  // <-------- this line
    FastLED.setBrightness(5); //TODO 
    fill_solid(leds, led_number, CRGB::Black);
    FastLED.show();
}
```

usage:
```cpp
#define DATA_PIN 27
strips.init(input, DATA_PIN);
```

The problem is that I cannot pass a "variable" to `addLeds` even if it has the `const` flag.

## template
I know understood that those things are done at compile time, in fact I also tried using a template approach like this one:
```cpp
template<uint8_t DATA_PIN> void Strips::init(std::string input, const uint8_t pin) {
    strips = getStrips(input);
    led_number = getNumLeds(strips);
    leds = new CRGB[led_number];
    FastLED.addLeds<WS2812B, DATA_PIN, GRB>(leds, led_number);  // <-------- this line
    FastLED.setBrightness(5); //TODO 
    fill_solid(leds, led_number, CRGB::Black);
    FastLED.show();
}
```
But it fails during linking  for `undefined reference to 'void Strips::init<(unsigned char)27>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'`

So, is there any method to pass this value to the function in order to make a decent wrapper O_O?

vortigont, 2024-02-21:
you can check [this](https://github.com/vortigont/LedFB/blob/main/ledfb/w2812-rmt.hpp) header file with wrapper class templates, it allows run-time configuration for esp32 family chips. Not sure which MCU you are using.

Sladerix, 2024-03-06:
> ```c++
> template<uint8_t DATA_PIN> void Strips::init(std::string input, const uint8_t pin)
> ```
> 
> Are you defining the function in the source? The template function either needs to be declared inline as part of the class in the header, or you need to enumerate all possible versions of the template (all pins) in the header to satisfy the linker.

**At the moment** I paused this work, in fact I'm not replying to the others (thanks anyway).

However @dmadison I already tried this approach, and it cannot link.

Sladerix, 2024-03-06:
Thank you for the example, but I'm pretty sure that this thing will not link (at least on platformio I got errors by doing like so, those went away only when I hard-coded the pin value)

> If you no longer need help with this, consider closing this issue as resolved.

Actually, my problem is not solved yet. I will leave this open.

dmadison, 2024-03-06:
Have you tried the MRE I posted above? Without a testable sample of your code demonstrating the issue it's difficult to provide suggestions.

I've used templated classes for per-pin functionality in a few of my own projects. I can assure you that there is a way to make this work.

samguyer, 2024-03-07:
Hello! This feature is something we have been talking about recently. In the past, it wasn't really possible because we relied on template parameters to compile down into very tight assembly code, particularly for the AVR family of microcontrollers. These days, though, a lot of the popular MCUs are much more powerful and could easily handle having the pin be a variable. I suspect that the way we would handle it is to add a "setPin" method on the controller -- something like that -- so that the existing interface doesn't need to change.

zackees, 2024-08-31:
A potential work around is to use explicit instantiation on a templated constructor and destructor for each pin for each driver.

The when you want say WS2912 on pin there you use

function foo
  switch arg-chipset:
    switch EP32:
      case 0: ,..

closed 2024-12-18 by zackees