FastLED/FastLED #1596: FastLED generate an noise signal on LED_PIN for no reason

UchihaYuki, 2024-02-07:
I use two different esp32-c3 dev boards. The LED strip is connected to pin 1. I observed the same noise signal pattern again and again, which caused my LED strip to be turned off. Following is the code:
```cpp
#include <FastLED.h>

const int NumberOfLeds = 4;
const int LedPin = 1;

CRGB leds[NumberOfLeds];

void turnLightOn()
{
    for (int i = 0; i < NumberOfLeds; ++i)
    {
        leds[i].setRGB(1, 0, 0);
    }
    FastLED.show();
}

void setup()
{
    Serial.begin(9600);
    while (!Serial)
    { // wait for serial port to connect. Needed for native USB port only
    }
    delay(1000);
    Serial.println("Show Time");

    FastLED.addLeds<WS2812B, LedPin, GRB>(leds, NumberOfLeds);
    turnLightOn();
}

void loop()
{
    delay(5000);
    analogRead(0); // This line caused the LED to be turned off after 5s. But without this line, you won't observe the noise signal. You can also observe the noise signal if you connect to Wi-Fi. I guess there are other cases where the noise signal will appear. If I don't use FastLed at all, nothing will be observed, so I think it must be related to this library.
}
```
I tried many pins for both the LED strip and analogRead function, the result is the same. The noise will appear on the LED pin even when you don't connect it to anything. The noise will appear if you use it with FastLED.

Following is the noise signal I observed on the LED pin, if I don't use FastLED nothing will be seen on this pin.
[image]

This noise pattern will follow the above. There might be other noise signals that my hand is not quick enough to capture them.
[image]

UchihaYuki, 2024-02-08:
Please also refer to this post:
https://www.reddit.com/r/arduino/comments/1al0y1e/when_i_read_pin_0_an_interference_signal_is/

According to others and my observation, this might be caused by watch dog. But only pins used by FastLED will be affected.

UchihaYuki, 2024-02-08:
Now I'm pretty sure, the bug is on FastLED side. I implemented the function myself by referencing this page:

https://github.com/espressif/arduino-esp32/blob/2.0.14/libraries/ESP32/examples/RMT/RMTWriteNeoPixel/RMTWriteNeoPixel.ino

The problem is gone.

samguyer, 2024-03-12:
Hi @UchihaYuki ! How are things going? I've got some time to work on FastLED, and I want to understand this problem more clearly. Is it still a problem?

closed 2024-08-30 by zackees