FastLED/FastLED #1597: Arduino Nano Connect RP2040 work issues

crusardri, 2024-02-11:
Hi!!

First of all, i'm spanish, and english is not my main languaje.

I'm working with an arduino UNO, and wrote a code that works, but recently, i buy an Arduino Nano Connect RP2040
Code compila and send to Arduino Connect, but for some reason, the digital PWR pins don't work.
I've bought 2 Arduino Nano Connect, and i have the same issue.

The diference between Arduino UNO and Connect, it show me this error

`
In file included from C:\Users\Crusa\Documents\Arduino\libraries\FastLED\src/FastLED.h:75:0,
                 from C:\Users\Crusa\Documents\Arduino\RGBA2_nano\RGBA2_nano.ino:1:
C:\Users\Crusa\Documents\Arduino\libraries\FastLED\src/fastspi.h:172:23: note: #pragma message: Forcing software SPI - no hardware SPI for you!
 #      pragma message "Forcing software SPI - no hardware SPI for you!"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
`

This is part of the code

`
CRGB rightDoorStrip[RIGHT_DOOR_LEDS];
CRGB leftDoorStrip[LEFT_DOOR_LEDS];
CRGB frontStrip[FRONT_LEDS];

void setup() {
  Serial.begin(9600);

  FastLED.addLeds<WS2812, RIGHT_DOOR_PIN, GRB>(rightDoorStrip, RIGHT_DOOR_LEDS);
  FastLED.addLeds<WS2812, LEFT_DOOR_PIN, RGB>(leftDoorStrip, LEFT_DOOR_LEDS);
  FastLED.addLeds<WS2812, FRONT_PIN, GRB>(frontStrip, FRONT_LEDS);

  startupAnimation();
}`

crusardri, 2024-02-12:
OK, mistery resolved

The pins are not maped like an arduino UNO
```C++
#define D2  25
#define D3  15
#define D4  16
#define D5  17
#define D6  18
#define D7  19
#define D8  20
#define D9  21
#define D10 5
#define D11 7
#define D12 4

#define A5  13
#define A4  12
#define A3  29
#define A2  28
#define A1  27
#define A0  26

#define NUM_LEDS 30

CRGB leds[NUM_LEDS];

void setup() {
  Serial.begin(9600);
  //FastLED.addLeds<WS2812, 1, GRB>(leds, 30);  //NONE
  //FastLED.addLeds<WS2812, 2, GRB>(leds, 30);  //NONE
  //FastLED.addLeds<WS2812, 3, GRB>(leds, 30);  //NONE
  //FastLED.addLeds<WS2812, 4, GRB>(leds, 30);  //D12
  //FastLED.addLeds<WS2812, 5, GRB>(leds, 30);  //D10
  //FastLED.addLeds<WS2812, 6, GRB>(leds, 30);  //NONE
  //FastLED.addLeds<WS2812, 7, GRB>(leds, 30);  //D11
  //FastLED.addLeds<WS2812, 8, GRB>(leds, 30);  //NONE
  //FastLED.addLeds<WS2812, 9, GRB>(leds, 30);  //NONE
  //FastLED.addLeds<WS2812, 10, GRB>(leds, 30); //NONE
  //FastLED.addLeds<WS2812, 11, GRB>(leds, 30); //NONE 
  //FastLED.addLeds<WS2812, 12, GRB>(leds, 30); //A4
  //FastLED.addLeds<WS2812, 13, GRB>(leds, 30); //A5
  //FastLED.addLeds<WS2812, 14, GRB>(leds, 30); //NONE
  //FastLED.addLeds<WS2812, 15, GRB>(leds, 30); //D3
  //FastLED.addLeds<WS2812, 16, GRB>(leds, 30); //D4
  //FastLED.addLeds<WS2812, 17, GRB>(leds, 30); //D5
  //FastLED.addLeds<WS2812, 18, GRB>(leds, 30); //D6
  //FastLED.addLeds<WS2812, 19, GRB>(leds, 30); //D7
  //FastLED.addLeds<WS2812, 20, GRB>(leds, 30); //D8
  //FastLED.addLeds<WS2812, 21, GRB>(leds, 30); //D9
  //FastLED.addLeds<WS2812, 22, GRB>(leds, 30); //NONE
  //FastLED.addLeds<WS2812, 23, GRB>(leds, 30); //NONE
  //FastLED.addLeds<WS2812, 24, GRB>(leds, 30); //NONE
  //FastLED.addLeds<WS2812, 25, GRB>(leds, 30); //D2
  //FastLED.addLeds<WS2812, 26, GRB>(leds, 30); //A0
  //FastLED.addLeds<WS2812, 27, GRB>(leds, 30); //A1
  //FastLED.addLeds<WS2812, 28, GRB>(leds, 30); //A2
  //FastLED.addLeds<WS2812, 29, GRB>(leds, 30); //A3
}
```

zackees, 2024-08-31:
I want to put these in an #ifndef guard but:
  * I'm not sure if this bug still exists
  * I can't find the define for nano

If you still see this bug then re-open this issue and I'll take a look.

closed 2024-08-31 by zackees