FastLED/FastLED #1598: RPi Pico freezes when changing color

markthomascampbell, 2024-02-14:
I have a Raspberry Pi Pico that is attached to a custom board I created that is running SMD sk6812 LEDs. Essentially, my board is working as a VU meter for audio. And the Pico is doing the calculations on the audio strength and how many LEDs to light up. I have the circuit working all day long with the color red at brightness 40, but as soon as I try to change to another color, or to another pattern, or even to a different brightness level, the Pico locks up nearly immediately, and the activity light repeatedly flashes 4 slow and 4 fast.  My research tells me that means that there's an incompatible board type.  But I know that that's not entirely true, because it is working fine with one color, but I also used a simple test program using FastLED to loop through all the colors on all the LEDs and it did so perfectly fine.  So that makes me think there's something software causing this somehow.

Specs for reproduction:
- Raspberry Pi Pico
- Arduino IDE 2.3.0
- Raspberry Pi Pico Mbed 4.1.1 (have also tried Philhower's rp2040 3.7.0, but running into issues with multiple pins for LED arrays)
- FastLED 3.6.0

Is this an appropriate place to post my sketch?

marmilicious, 2024-02-14:
I would suggest starting with posting this info and a link to your code on the reddit group.

https://www.reddit.com/r/FastLED

zackees, 2024-09-04:
Yes please post the sketch and attempt to debug yourself a bit by adding print statements and seeing what the last line that it executes is so that we can narrow it down. The fastled devs aren't using this platform so any support in narrowing it down will help us immensely.

zackees, 2024-09-08:
And please open the serial monitor - your strip might be crashing.

ledereyes, 2024-11-16:
Sounds like the same issue as in https://github.com/FastLED/FastLED/issues/1497

Try commenting out your FastLED.show() call to see if that avoids the lockup or crash.

zackees, 2025-11-05:
Fix: https://github.com/FastLED/FastLED/issues/1497#issuecomment-1516406183

You dont' have the right platform files installed.