FastLED/FastLED #1600: Gradient color palettes

Serega88kos, 2024-02-19:
Hi all.
Is it possible to implement a temperature gradient, for example -50 to 50 degrees?
``` c++
DEFINE_GRADIENT_PALETTE( heatmap_gp ) {
-30, CRGB:: MediumSlateBlue,
-20, CRGB:: Blue, 
  0, CRGB:: DeepSkyBlue,,
 10, CRGB:: MediumSpringGreen,
 20, CRGB:: Lime,
 30, CRGB:: DarkOrange
};
```

Serega88kos, 2024-02-19:
It turned out this way

``` с++
 DEFINE_GRADIENT_PALETTE(Temperature) {
         0,  123,   104,   238,   //MediumSlateBlue
128 -30 *2,    0,     0,   255,   //Blue
128 -20 *2,    0,   191,   255,   //DeepSkyBlue
128  +0 *2,  176,   224,   230,   //PowderBlue
128 +10 *2,  238,   232,   170,   //PaleGoldenrod
128 +20 *2,  255,   255,     0,   //Yellow
       255,  255,     0,     0 }; //Red

CRGBPalette256 MyPalette = Temperature;
```

marmilicious, 2024-02-19:
Hello @Serega88kos 
Looks good. If you're sorted out feel free to close this issue. We use github "issues" here mostly just for bugs or improvements in the library itself. For all general discussions, help, and how-to, please use the FastLED Reddit group here: https://www.reddit.com/r/FastLED

closed 2024-02-22 by Serega88kos