FastLED/FastLED #1602: Store and Use HTMLColorCode in NVS Memory

VigneshVicky97, 2024-03-02:
```
struct CRGB
{
	/// Predefined RGB colors
    typedef enum {
        AliceBlue=0xF0F8FF,             ///< @htmlcolorblock{F0F8FF}
        Amethyst=0x9966CC,              ///< @htmlcolorblock{9966CC}
        AntiqueWhite=0xFAEBD7,          ///< @htmlcolorblock{FAEBD7}
        Aqua=0x00FFFF,                  ///< @htmlcolorblock{00FFFF}
        Aquamarine=0x7FFFD4,            ///< @htmlcolorblock{7FFFD4}
        Azure=0xF0FFFF,                 ///< @htmlcolorblock{F0FFFF}
        Beige=0xF5F5DC,                 ///< @htmlcolorblock{F5F5DC}
        Bisque=0xFFE4C4,                ///< @htmlcolorblock{FFE4C4}
        Black=0x000000,                 ///< @htmlcolorblock{000000}
        BlanchedAlmond=0xFFEBCD,        ///< @htmlcolorblock{FFEBCD}
        Blue=0x0000FF,                  ///< @htmlcolorblock{0000FF
		.
		.
		.
    } HTMLColorCode;
}
```

The above given code is a part in FastLED library to control RGB LED. The code to flash RGB LED is:

```
#include <FastLED.h>
#define NUM_LEDS 1
CRGB leds[NUM_LEDS];
void setup() 
{ 
    FastLED.addLeds<NEOPIXEL, 15>(leds, NUM_LEDS); 
}
void loop() 
{
    leds[0] = CRGB::White; 
    FastLED.show(); 
    delay(30);
    leds[0] = CRGB::Black; 
    FastLED.show(); 
    delay(30);
}
```

I want to store the color of RGB in Flash Memory using NVS library such as White, Black, so that, it has to take the color from the memory.How can I make that happen?

marmilicious, 2024-03-02:
Hello @VigneshVicky97   
We use github "issues" here mostly just for bugs or improvements in the library itself.  For all general discussions, help, and how-to, please use the FastLED Reddit group here: 

https://www.reddit.com/r/FastLED

Before posting your question there you have have a look at these posts and see if any of them are useful to you.
https://www.reddit.com/r/FastLED/comments/s0obmk/convert_hex_colour_codes_to_rgb/
https://www.reddit.com/r/FastLED/comments/dnj8o2/crgb_html_color_index_as_variable/
https://www.reddit.com/r/FastLED/comments/ompee1/custom_color_variable_help/
https://www.reddit.com/r/FastLED/comments/sdcny2/using_crgbcolor_as_a_function_argument_in_the/

VigneshVicky97, 2024-03-05:
Thankyou @marmilicious . I think, I found a way to set RGB Color through MQTT. Thankyou for your inputs.

closed 2024-03-05 by VigneshVicky97