FastLED/FastLED #1606: Compiling FastLED blink sketch using FastLED 3.6.0 for an ESP32-C3 with Arduino IDE 1.8.19 seems to fail.

HenkHoldijk, 2024-03-07:
Trying to build the FastLED blink example for an ESP32-C3 using the Arduino IDE 1.8.19 results in:

- rmt.h that must be replaced with rmt_tx.h and/or rmt_rx.h
- periph_ctrl.h not being used anymore
- Several warnings regarding volatile operands being deprecated
- cpu_ll_get_cycle_count() being deprecated
- incompatible types

When disabling all warnings, still the following is reported:

...\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp: In member function 'void ESP32RMTController::startOnChannel(int)':
...\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:266:29: error: 'RMTMEM' was not declared in this scope
  266 |         mRMT_mem_start = & (RMTMEM.chan[mRMT_channel].data32[0].val);
      |                             ^~~~~~

...\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp: In static member function 'static void ESP32RMTController::doneOnChannel(rmt_channel_t, void*)':
...\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:347:5: error: 'gpio_matrix_out' was not declared in this scope; did you mean 'gpio_iomux_out'?
  347 |     gpio_matrix_out(pController->mPin, 0x100, 0, 0);
      |     ^~~~~~~~~~~~~~~
      |     gpio_iomux_out

HenkHoldijk, 2024-03-07:
Hi Sam (?), Adding ugly #ifdefs is not a long term solution. Sooner or later this kind of "swamp" will catch up with you. The ESP32 series has several variations with different behavior (and definitely more will come) an additional platform layer would make sense. I already can see that IO mapping of a C3 is handled different from some other EPS32's. The ESP8266 ("First Of Class") is EOL,
but I guess the ESP32 series will continue for a while (before the ESP64 series is announced ;) ). Several changes in the architecture will appear while using the same well known ESP32 name (exploit "ESP32"). In the end ESP32 is just a name and the SOC architecture will change while still called ESP32. This really calls for some sort of adaptation layer. Cheers, Henk

samguyer, 2024-03-08:
@HenkHoldijk The problem is that we are controlling the RMT device at a pretty low level. ESP-IDF offers higher level abstractions, but they take away the control we need. At least that was the situation the last time I really looked at it. If the architectures get sufficiently different (and need different drivers) we will probably split them up (like we do with the ESP8266)

HenkHoldijk, 2024-03-08:
I'm also using the v3.0.0-alpha3 (as I'm using the ESP32-C3). It is definitely related to the changes made there. But creating an abstraction layer while still using low level facilities will be a tough challenge.

samguyer, 2024-03-08:
The ESP-IDF does have a nice hardware abstraction layer. I just need to make sure it lets us do what we need to do.

HenkHoldijk, 2024-03-08:
Yep, I found this as I was looking for an Adafruit_NeoPixel alternative, as the Adafruit lib does not handle interrupts (at least for the ESP32-C3) in an even decent way. NeoPixelBus runs into the same issue. Although it's difficult to blame Expressive, as sometimes things just have to change due to unavoidable refactoring. B.t.w. The Adafruit_NeoPixel lib still works, as long as you don't call it's functions in an interrupt handler (which I'm trying to do :) ).

samguyer, 2024-03-08:
Ok, I'm going to make it a priority to work on this. I've also got some ideas on how to make it even more robust to interrupts.

J-Lips, 2024-03-11:
I just started a project using ESP32-C6, for which I am seeing similar errors when including the FastLED library. (running Arduino IDE 1.8.19, FastLED 3.6.0)

...\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:330:6: error: #error Not yet implemented for unknown ESP32 target
clockless_rmt_esp32.cpp:394:6: error: #error Not yet implemented for unknown ESP32 target
clockless_rmt_esp32.cpp:440:10: error: #error Not yet implemented for unknown ESP32 target
clockless_rmt_esp32.cpp:266:29: error: 'RMTMEM' was not declared in this scope
clockless_rmt_esp32.cpp:347:5: error: 'gpio_matrix_out' was not declared in this scope; did you mean 'gpio_iomux_out'
clockless_rmt_esp32.cpp:445:31: error: 'tx_next_bit' was not declared in this scope
clockless_rmt_esp32.cpp:451:35: error: 'tx_done_bit' was not declared in this scope
Error compiling for board DFRobot FireBeetle 2 ESP32-C6.

It might be exactly the same error Henk is reporting, but since some error lines did not show up in the log he shared, I also share mine here. 
Also on the ESP32-C6 Adafruit-NeoPixel works so far.

Cheers, Johannes

samguyer, 2024-03-12:
@J-Lips I think it's the same problem -- more to do with software changes in the ESP-IDF than the specific hardware part.

Ronso007, 2024-03-16:
> @J-Lips I think it's the same problem -- more to do with software changes in the ESP-IDF than the specific hardware part.

Is there a way to contact you? I am trying to find a fast workaround for this problem

DillonSimeone, 2024-04-25:
Aye, I also have a lot of ESP32-C6s. They're amazing, I was looking forward using them to play with FastLED and ESP32Now... Oof! 

Like a cold water dumped onto me when I reilized that FastLED won't compile on ESP32-C6s, and that it's not a five-minute fix.

egnor, 2024-08-05:
ESP32-C6 support seems to be tracked here: https://github.com/FastLED/FastLED/issues/1623
The RMT issue with arduino-esp32 v3.x / ESP-IDF v5.x seems to be tracked (mostly?) in: https://github.com/FastLED/FastLED/issues/1628 and https://github.com/FastLED/FastLED/issues/1631
Fixes for both seem to be merged in git (not yet released formally)?

zackees, 2024-08-17:
Fixed in the 3.7.1 and 3.7.3 release

closed 2024-08-17 by zackees

nup002, 2024-09-16:
FastLED 3.7.3 for ESP32C6 compiles fine, but leads to the following runtime exception:

```driver_ng is not allowed to be used with the legacy driver```

zackees, 2024-09-16:
Yup. Got that too. Trying to fix it.