FastLED/FastLED #1609: Inefficient SPI Data Transmission for APA102 / SK9822 LEDs on ESP32

coryking, 2024-03-10:
## Summary

When using FastLED with ESP32 to drive APA102 and SK9822 LED strips, significant inefficiencies have been observed in SPI data transmission. The implementation appears to conduct separate DMA transactions for each byte of data, introducing noticeable gaps between 8-bit bursts. This behavior considerably diminishes the effective data transmission rate, closely mirroring the performance limitations characteristic of older one-wire LED models like the WS2812b.

## Environment

- **Microcontroller**: ESP32
- **LED Types**: APA102 and SK9822
- **LED Count**: Tested with 30 LEDs, but the issue is applicable to any strip length
- **Data Rate**: Configured for 5MHz and 20MHz
- **Library Version**: FastLED (latest as of writing)

## Observations

- **Oscilloscope Measurements**:
  - With a data rate set to 5MHz, updating a 30-LED strip takes about 422 microseconds.
  - Setting the data rate to 20MHz reduces the update sequence to approximately 244 microseconds, which, despite showing improvement, falls short of expected performance levels.
  - Gaps between each 8-bit data burst are measured to be roughly 1.7 microseconds, aligning with the setup overhead for DMA transactions described in ESP32 documentation.

## Expected vs. Actual Behavior

Expected: For high-speed SPI communication, especially at higher data rates, FastLED should utilize continuous DMA transfers for the entire data packet of LED strip data, minimizing CPU overhead and maximizing throughput for quick and efficient updates.

[image]
This is what I'm calling the "inner byte gap". It's roughly 1.7 or so microseconds. It transmits 8 bits, takes a break, transmits 8 more bits... etc.

Actual: The implementation seems to trigger individual DMA transactions for every byte sent to the APA102 and SK9822 LEDs, leading to unnecessary overhead, reduced transmission rate, and increased update times, particularly for longer LED strips.

## Investigation and Documentation

This issue aims to document and investigate the SPI data transmission inefficiency for APA102 and SK9822 LEDs on the ESP32 using FastLED. The goal is to explore this issue further, propose optimizations, and prevent duplication of efforts across multiple issues or pull requests.

[image]
The above image shows how long it took to transmit a 30 LED strip. Roughly 230uS at a "20MHz" setting in fastLED.

Additionally, this concern echoes a [comment](https://github.com/FastLED/FastLED/issues/1552#issuecomment-1870581313) from @SCarter858 on Dec 27, 2023, which detailed similar inefficiencies and questioned the actual utilization of DMA for SPI communication within FastLED's ESP32 HW SPI support.

## Additional Context

Note that this being caused by separate DMA transactions is only hypothetical. It might not be using DMA at all!

This inefficiency not only affects APA102 LEDs but also their SK9822 counterparts, which are often used interchangeably in projects. The potential lack of full DMA utilization for SPI transmission significantly throttles the performance capabilities of these chips, making high-density LED installations as slow as older one-wire LED protocols like the WS2801. An optimized implementation would greatly enhance performance, leveraging the ESP32's capabilities for LED projects using FastLED.

coryking, 2024-03-11:
Here is some results when I used a completely different APA102 driver I found online: [DMA Findings.md](https://github.com/coryking/FastLED-Test-Jig/blob/DMA_Driver_Test/FINDINGS.md)

As you can see from the oscilloscope readings in the linked document, when you use DMA it looks nothing like the ones in this ticket. Plus the speeds are significantly faster (though they appear to be "rounded" for reasons I don't fully understand.

coryking, 2024-03-17:
So I need everybody's opinion on how to provide the "SPIOutput" class the number of LEDs / bytes required for a DMA buffer.

 In order to get DMA working we need to allocate a buffer on the heap but to do so we need the size of the buffer. I think the best place for this is in the [init()]([ESP32SPIOutput](https://github.com/coryking/FastLED/blob/2be8b401ee78961b1c594b92fc24b29244234baf/src/platforms/esp/32/fastspi_esp32_dma.h#L137)) function. The problem is these "SPIOutput" classes do not know how many LEDs they are supposed to be driving, thus I have no clue how big of a buffer to allocate.  So I need to get the number of LED's plumbed in and probably the number of "extra" bytes needed for start and end frames.

The problem is these SPI handlers are all supposed to look like "[SPIOutput](https://github.com/coryking/FastLED/blob/2be8b401ee78961b1c594b92fc24b29244234baf/src/fastspi.h#L66)" even though they are all completely different classes. So I suspect if I make a change to init() to pass in the number of leds (or perhaps just bytes) I'll have to change all the other SPIOutput classes as well (which may not be a bad idea because any of the other SPI classes that do DMA need the same info).

So, in short:

1. Is SPIOutput the correct place to manage the DMA buffer?
2. What is the best way to provide SPIOutput with the size of said DMA buffer?
3. Should it get the number of LEDs or the number of bytes?  I'm leaning toward the number of bytes, as I suspect that SPIOutput is a "bit pusher" and isn't supposed to care about the various LED protocols.

Pinging @samguyer , @Jueff (who looks to have done some DMA stuff) and @kriegsman for their thoughts.

coryking, 2024-03-17:
Also I should add I'm going to borrow quite liberally from the [led_strip_spi component](https://github.com/UncleRus/esp-idf-lib/tree/master/components/led_strip_spi) from the [UncleRus esp-if-lib component library](https://github.com/UncleRus/esp-idf-lib). I'll make sure to adhere to its MIT License when pushing my changes.

zackees, 2024-08-17:
Any update on this?

zackees, 2024-11-14:
I'm adding a new feature for bulk transfer through the SPI device on ESP32. You can enable it via `#define FASTLED_ESP32_SPI_BULK_TRANSFER 1` before the inclusion of FastLED.

It will go out with 3.9.4. Right now it's just in master. I've compile tested, but haven't run it yet.

coryking, 2024-11-14:
Thanks! I haven't had time to look into this for quite some time,
unfortunately life events got in the way!

SCarter858, 2024-12-31:
Thanks, Zackees!  I finally got back to this project and turning on your new bulk transfer flag doesn't seem to change the speed/operation of the SPI output.  But I haven't dug into things very far yet...

zackees, 2025-01-01:
> Thanks, Zackees! I finally got back to this project and turning on your new bulk transfer flag doesn't seem to change the speed/operation of the SPI output. But I haven't dug into things very far yet...

You are using a build define and not an include define right?

SCarter858, 2025-01-01:
Using this at the start of my main.cpp:

#define FASTLED_ESP32_SPI_BULK_TRANSFER 1
#define FASTLED_ALL_PINS_HARDWARE_SPI
#include <FastLED.h>

Without debugger and scope hooked up, I see that it does indeed seem to invoke your changes, since I can console print that FASTLED_ESP32_SPI_BULK_TRANSFER_SIZE is set to 64 (your default).  

But the refresh rate that I'm seeing (a single strand of 128 APA102s can refresh only 7 frames/sec) is exactly the same with or without your bulk transfer invoked.  

As a quick debug, I put a couple Serial.print lines in the writePixels template where you implemented your fix, and it looks like that template is never being called?

zackees, 2025-01-01:
I don't know why it wouldn't be called. I just did an audit.

Massively increase the size of FASTLED_ESP32_SPI_BULK_TRANSFER_SIZE and see if that helps.

Try to put some print statements to confirm what is happening. It should be calling my new api.

SCarter858, 2025-01-01:
Part of the problem might be in the initial conditional compile setup:  (all this in file fastspi_esp32.h)

Putting some "#pragma message" statements in, it looks like (when compiling for an ESP32S3), it correctly goes into the #if of lines 70-75 to set the bus to FSPI (though I'm not sure what that term is - it's not standard ESP nomenclature?).  But just a few lines later, it compiles lines 83-87, as if the bus was set to VSPI.  It does not compile lines 93-97 as expected.  Later, in the early part of the main.cpp program, when I print the value of FASTLED_ESP32_SPI_BUS to the console, it shows as "0".  

Perhaps VSPI is a reserved word, while FSPI is not?  (color coding of them in VSCode is different).  

Putting in some Serial.print lines in the main program as pixel writes are happening shows that the "select()" method in the file we're talking about is called frequently, but I can't find any of the other writeByte, writePixel, etc, functions ever being called.  But this is all downstream of the initial setup discussed above...

zackees, 2026-09-03:
Reopening: this issue was closed automatically by a stale-bot run, not by a maintainer decision. That auto-close behavior was a mistake and has been disabled — FastLED issues will no longer be closed for inactivity. Apologies for the noise.

SCarter858, 2026-10-26:
Circling back on this years later — finally dug into it properly.

Root cause: `ESP32SPIOutput`'s direct `select()`/`writeByte()`/`writeWord()`/`release()` path (used by APA102/SK9822/WS2801-style clocked chipsets) issues one `spi_device_polling_transmit()` per byte, each of which blocks on the SPI peripheral lock as @zackees suspected back in Nov 2024. `FASTLED_ESP32_SPI_BULK_TRANSFER` doesn't help because it's wired into a different code path (`writePixels`) that this chipset class never calls — matches what I found testing it back in Jan 2025 in the thread above.

Filed with a working fix (ring-buffered async DMA via `spi_device_queue_trans()`, confirmed on real hardware): #4918 / #4919.

SCarter858, 2026-10-26:
Closing - fixed by #4919, merged 2026-10-26. Thanks @zackees for the fix path and the hardware verification on the PR.

SCarter858, 2026-10-26:
Looks like neither the reporter nor I have permission to close this ourselves - @zackees, could you close this out now that #4919's merged? Thanks again for the hardware verification.

closed 2026-10-26 by zackees