FastLED/FastLED #1612: FastLED v3.6.0 not sending complete LED data on ESP8266 with WiFi enabled

achronoxx, 2024-03-20:
I've included a simple Sketch to demonstrate a problem with certain versions of FastLED while WiFi is enabled
    - this is running on an ESP8266_NODEMCU_ESP12E
    - I observed that v3.6.0 of FastLED does not send the complete set of data to the LEDs when WiFi is connected
    - eg. this sketch controls 150 LEDS, but  FastLED is only sending 1 or 2 bytes of data, but should be sending 450 bytes (150LEDs * 3 bytes/LED)
    - FastLED.Show() is called every ~100ms, and data is sent every time,  but mostly it is 1 or 2 bytes - occasionally all 450 bytes are sent
    - I've confirmed this by looking at the data signal on a digital oscilloscope
    - it is also very evident when looking at the LED strip, as the pattern "stalls" 
    
    The problem goes away in the following 2 cases:
      1. Disable the WiFi (set wifiGUI = 0 in sketch below to test)
      2. Use v3.3.3 of FastLED (test by building with this older version)
          - I'm not sure if other versions work, as I was using v3.3.3 before I updated straight to 3.6.0

    These are the versions of all the Libraries I used:
    - Using library FastLED at version 3.6.0
    - Using library ESP8266WiFi at version 1.0 
    - Using library Hash at version 1.0 
    - Using library ESP8266mDNS at version 1.2
    - Using library ArduinoOTA at version 1.0
    - Using library NTPClient at version 3.2.0

Is there something NEW in v3.6.0 that I need to adjust to make this work, or is this a bug?

Any help would be appreciated!

Andrew

Here is the sketch that demonstrates the problem:

```
/*************************************************************************************************************************************************************
   Mar 20/24
 
    This is a simple Sketch to demonstrate problem with certain versions of FastLED while WiFi is enabled
    - this is running on an ESP8266_NODEMCU_ESP12E
    - I observed that v3.6.0 of FastLED does not send the complete set of data to the LEDs when WiFi is connected
    - eg. this sketch controls 150 LEDS, but  FastLED is only sending 1 or 2 bytes of data, but should be sending 450 bytes (150LEDs * 3 bytes/LED)
    - FastLED.Show() is called every ~100ms, and data is sent every time,  but mostly its 1 or 2 bytes - occasionally all 450 bytes are sent
    - I've confirmed this by looking at the data signal on a digital oscilloscope
    - it is also very evident when looking at the LED strip, as the pattern "stalls" 
    
    The problem goes away in the following 2 cases:
      1. Disable the WiFi (set wifiGUI = 0 in sketch below to test)
      2. Use v3.3.3 of FastLED (test by building with this older version)
          - I'm not sure if other versions work, as I was using v3.3.3 before I updated straight to 3.6.0

    These are the versions of all the Libraries I used:
    - Using library FastLED at version 3.6.0
    - Using library ESP8266WiFi at version 1.0 
    - Using library Hash at version 1.0 
    - Using library ESP8266mDNS at version 1.2
    - Using library ArduinoOTA at version 1.0
    - Using library NTPClient at version 3.2.0

**************************************************************************************************************************************************************/

char swVer[] = "24.03.20";

#define wifiGUI 0           // to enable WiFI

#if wifiGUI
#define OTA_ENABLE 1
#else
#define OTA_ENABLE 0        // make sure OTA is disabled if WiFi off
#endif

#define logger (&Serial)
#define ledESP D4
#define NUM_LEDS 150

[... 260 lines cut ...]
  if (server.hasClient()) {
    serverClient.stop();            // disconnect current client, as sometimes connection is lost, and reconnect does not work as it thinks it's still connected

    // find free/disconnected spot
    if (!serverClient.connected()) {
      serverClient = server.available();
      logger->println(F("\nNew client connected...\n"));
    }

    // no free/disconnected spot so reject
    else {
      server.available().println("busy");
      // hints: server.available() is a WiFiClient with short-term scope
      // when out of scope, a WiFiClient will
      // - flush() - all data will be sent
      // - stop() - automatically too
      logger->printf("server is busy with active connection...\n");
    }
  } // if(server.hasClient)

} // end newClient()

#endif // for WiFi ==============================================================================================================

```

zackees, 2024-09-20:
Hi Andrew, this is an issue with the 4.4 RMT driver. Unfortunately this is a legacy driver that is going away, and we have no plans on touching this code much more than we already have as it has been finely tuned to work the way it has.

It's entirely possible that the issues you are describing have also been addressed in an update to FastLED, since the version you are referencing is quite old.

achronoxx, 2024-09-30:
Hi Zach,

Thanks for the response.  I will try the latest version in the next few days and report back.

Andrew

zackees, 2024-10-01:
RMT5 has been released on master.

You'll need idf5.1 or higher.

zackees, 2024-11-02:
Closing this big as i think it's a solved issue with the release of 3.9.0

closed 2024-11-03 by zackees