FastLED/FastLED #1618: 3.6.0 as installed in Arduino IDE doesn't work with SEEED Xiao RP2040

np703, 2024-04-27:
The 3.6.0 version installed by the library manager in Arduino IDE 2.3.2 doesn't work with the Seeed Xiao RP2040. I've seen other reports on the web that the problem is that certain RP2040 definitions are missing in the Arduino IDE library and that the 3.6.0 version in the Arduino library manager isn't the same as the 3.6.0 version here. Perhaps consider updating the version number to eliminate the confusion that seems somehow to have crept in?

samguyer, 2024-05-27:
@np703 What is the error or behavior that you're seeing?

navivfr, 2024-06-06:
In file included from /Users/mathieu/Documents/Arduino/libraries/FastLED/src/FastLED.h:79,
                 from ...
/Users/mathieu/Documents/Arduino/libraries/FastLED/src/fastspi.h:172:23: note: '#pragma message: Forcing software SPI - no hardware SPI for you!'
  172 | #      pragma message "Forcing software SPI - no hardware SPI for you!"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
{standard input}: Assembler messages:
{standard input}:5414: Error: invalid offset, value too big (0x00000534)
{standard input}:5441: Error: invalid offset, value too big (0x000004F0)

zackees, 2024-08-31:
I believe this is now fixed. We have an RP2040 that is part of the build suite.

closed 2024-08-31 by zackees