FastLED/FastLED #1621: 3.6.0 gives wrong WS2812B colours on Blue Pill

Tressie246, 2024-05-07:
I compiled FastLED 3.6.0 for an STM32F103C8T6 okay.

The neopixels I am using work fine with an earlier version of FastLED on an atmega328p, using FastLED.addLeds<NEOPIXEL, DATA_PIN>(leds, NUM_LEDS);

When I use the same line on the STM32F103C8T6, I get invariant, solid white LED outputs.

If I use FastLED.addLeds<WS2812, DATA_PIN, RGB>(leds, NUM_LEDS);, I get wrong colours, but that may be blinked.

FastLED.addLeds<WS2812B, DATA_PIN, RGB>(leds, NUM_LEDS); produces a similar result, but with different colours.

I write Black, and get I get blue. I write White, I get pink.

?

samguyer, 2024-05-27:
@Tressie246 Try GRB for the color order -- the WS281x chips do not use RGB color order.

zackees, 2024-09-07:
If you are writing black and getting a color then this isn't a color ordering issue.

Please reproduce your problem by cloning our repo and editing the [dev/dev.ino](dev/dev.ino).

If you can post a fork then this will give us a repro case and we can fix the issue.

zackees, 2024-09-20:
Pin definitions were recently updated on nano every by community support so I'm marking this bug as resolved.

closed 2024-09-20 by zackees