FastLED/FastLED #1622: Problem with SM16703

ahmetcemturan, 2024-05-13:
Hi,
my problem is that I can not get my SM16703 strip to run correctly.
It seems as it is atotally wrong driver.
I tried to paly around with the timing but don't really know what I am doing there...
Can anyone help?

samguyer, 2024-05-13:
I've never heard of that chip, but the protocol looks fairly similar to WS2812. Have you tried just using the WS2812 chip and seeing if it works?
Also: this is a good question for the FastLED subreddit.

zackees, 2024-09-20:
> Hi,
> 
> my problem is that I can not get my SM16703 strip to run correctly.
> 
> It seems as it is atotally wrong driver.
> 
> I tried to paly around with the timing but don't really know what I am doing there...
> 
> Can anyone help?

The T1, T2, T3 timings are a little non standard but are roughly equivalent to the TH0...TL1 timings used in data sheets.

If you look at chipsets.h you'll find a python script embedded in the comments that will calculated these T1...T3 values from the TH0...TL1 values found in data sheets.

zackees, 2026-09-01:
Resolved by the centralized SM16703 timing/controller fix in commit 678b3d04bb (merged in PR #2347). Current source has a single `SM16703Controller` using `fl::TIMING_SM16703` (300/600/300 ns), and the public `SM16703` route uses that controller. The remaining report lacks device/batch markings, expected-vs-observed color order, and logic-analyzer evidence needed to diagnose any hardware-specific follow-up.

closed 2026-09-01 by zackees