FastLED/FastLED #1624: All LEDs Flash White when power-on or ESP12 resets

gadjet, 2024-05-21:
I'm using an ESP12 to drive 5 WS2813b RGB LEDs, in the code I use the Fastled library and switch off the LEDS straight away but there's always a white flash of all the LEDs on power-up and reset/wake up from deepsleep (the code uses deepsleep and wakes itself up every few minutes).
Code:
```
#include <FastLED.h>
#define DATA_PIN 2
#define NUM_LEDS 5
//#define CLOCK_PIN 13
#define COLOR_ORDER RGB
#define BRIGHTNESS 5
// Define the array of leds
CRGB leds[NUM_LEDS];

void setup() {
  //Setup lEDs
  FastLED.addLeds<WS2812B, DATA_PIN, COLOR_ORDER>(leds, NUM_LEDS);
  FastLED.setBrightness(BRIGHTNESS);
}

void loop() {
    //Turn off LEDs
  FastLED.clear();  // clear all pixel data
  FastLED.show();
  ESP.deepSleep(10e6);
}
```
The LED connections are as per the datasheet with a 100nF cap across each LED supply. 
[image]

The Data line is connected to GPIO2 and once running the LEDs all work correctly, I can control the colours and brightness etc.

I have yet to put an oscilloscope on GPIO2 to see what signals may be on there during boot.

I thought the LEDs should all remain off on power-up.

If I disconnect the data line and apply power then the LEDs remains off.

The only possibly contentious thing I'm doing is supplying the LEDs with 3.3V and not 5V I know this isn't ideal but they work perfectly well apart from the power-up/reset glitch.

gadjet, 2024-05-22:
It appears to be data on GPIO2 whenever the ESP12 is powered up or reset, nothing to do with FastLED.

samguyer, 2024-05-27:
@gadjet Yeah, sometimes the pins have weird special behavior on powerup. Is there more to do here, or can we close this issue?

gadjet, 2024-05-27:
Hi, it can be closed, I thought I'd closed it with my last comment! my bad.

closed 2024-05-27 by gadjet