FastLED/FastLED #1625: Compiler error on Version 3.6.0 - ambiguous overload for 'operator*' - while using esp-idf version 5.1.3

BoltSwith, 2024-05-23:
Greetings.

I get the following compiler errors originating from the noise.cpp file with the following error messages:

- ambiguous overload for 'operator*' (operand types are 'q44' {aka 'qfx<unsigned char, 4, 4>'} and 'int')
- ambiguous overload for 'operator*' (operand types are 'int' and 'q88' {aka 'qfx<short unsigned int, 8, 8>'})
- ambiguous overload for 'operator*' (operand types are 'q44' {aka 'qfx<unsigned char, 4, 4>'} and 'int')

All three above errors come with supporting error messages - 'more than one operator "*" matches these operands:'
The lines in the Noise.cpp file producing these errors respectively are:

- 628: fill_raw_2dnoise8(pData, width, height, octaves-1, freq44, amplitude, skip+1, x*freq44, freq44 * scalex, y*freq44, freq44 * scaley, time);
- 666: fill_raw_2dnoise16(pData, width, height, octaves-1, freq88, amplitude, skip, x *freq88 , scalex *freq88, y * freq88, scaley * freq88, time);
- 704: fill_raw_2dnoise16into8(pData, width, height, octaves-1, freq44, amplitude, skip+1, x*freq44, scalex *freq44, y*freq44, scaley * freq44, time);

From investigation, it appears the error message - "ambiguous overload for 'operator*' - indicates that the compiler is unable to determine which version of the multiplication operator (*) to use when you try to multiply an object of type q44 or q88 with an integer.

Seems these errors may not be flagged by IDF versions below 5.1.3, or probably the errors appear as a result of some other settings in my Menu Config (Compiler Optimization set to O2 - Optimize for performance)

Any advice will go a long way. Thanks.

tueddy, 2024-05-24:
I can compile successfully with PR https://github.com/FastLED/FastLED/pull/1619

BoltSwith, 2024-05-27:
Many thanks @tueddy for the swift response. I have been able to compile my project.

closed 2024-05-27 by BoltSwith