FastLED/FastLED #1628: FastLED 3.7.0 compile error

niiiksh, 2024-05-31:
Hi! I've updated FastLED to new version and I'm getting this when trying to build project.

c:\Users\niiik\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp: In member function 'void ESP32RMTController::tx_start()':
c:\Users\niiik\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:326:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'mem_rd_rst_n'
  326 |     RMT.chnconf0[mRMT_channel].mem_rd_rst_n = 1;
      |                                ^~~~~~~~~~~~
c:\Users\niiik\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:327:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'mem_rd_rst_n'
  327 |     RMT.chnconf0[mRMT_channel].mem_rd_rst_n = 0;
      |                                ^~~~~~~~~~~~
c:\Users\niiik\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:328:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'apb_mem_rst_n'
  328 |     RMT.chnconf0[mRMT_channel].apb_mem_rst_n = 1;
      |                                ^~~~~~~~~~~~~
c:\Users\niiik\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:329:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'apb_mem_rst_n'
  329 |     RMT.chnconf0[mRMT_channel].apb_mem_rst_n = 0;
      |                                ^~~~~~~~~~~~~
c:\Users\niiik\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:335:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'conf_update_n'
  335 |     RMT.chnconf0[mRMT_channel].conf_update_n = 1;
      |                                ^~~~~~~~~~~~~
c:\Users\niiik\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:336:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'tx_start_n'
  336 |     RMT.chnconf0[mRMT_channel].tx_start_n = 1;
      |                                ^~~~~~~~~~
c:\Users\niiik\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp: In static member function 'static void ESP32RMTController::doneOnChannel(rmt_channel_t, void*)':
c:\Users\niiik\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:389:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'tx_stop_n'
  389 |     RMT.chnconf0[channel].tx_stop_n = 1;
      |                           ^~~~~~~~~
c:\Users\niiik\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:390:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'conf_update_n'
  390 |     RMT.chnconf0[channel].conf_update_n = 1;
      |                           ^~~~~~~~~~~~~
c:\Users\niiik\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:392:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'mem_rd_rst_n'
  392 |     RMT.chnconf0[channel].mem_rd_rst_n = 1;
      |                           ^~~~~~~~~~~~
c:\Users\niiik\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:393:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'mem_rd_rst_n'
  393 |     RMT.chnconf0[channel].mem_rd_rst_n = 0;
      |                           ^~~~~~~~~~~~
c:\Users\niiik\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:394:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'apb_mem_rst_n'
  394 |     RMT.chnconf0[channel].apb_mem_rst_n = 1;
      |                           ^~~~~~~~~~~~~
c:\Users\niiik\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:395:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'apb_mem_rst_n'
  395 |     RMT.chnconf0[channel].apb_mem_rst_n = 0;
      |                           ^~~~~~~~~~~~~

Uspizig, 2024-05-31:
Same here: After Updating to ESP 32 Arduino Core 3.0 with an ESP32-S3 chip(IDF5.1)  ... 
      |                                                                        ^
c:\Users\User\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp: In member function 'void ESP32RMTController::tx_start()':
c:\Users\User\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:326:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'mem_rd_rst_n'
  326 |     RMT.chnconf0[mRMT_channel].mem_rd_rst_n = 1;
      |                                ^~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:327:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'mem_rd_rst_n'
  327 |     RMT.chnconf0[mRMT_channel].mem_rd_rst_n = 0;
      |                                ^~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:328:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'apb_mem_rst_n'
  328 |     RMT.chnconf0[mRMT_channel].apb_mem_rst_n = 1;
      |                                ^~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:329:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'apb_mem_rst_n'
  329 |     RMT.chnconf0[mRMT_channel].apb_mem_rst_n = 0;
      |                                ^~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:335:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'conf_update_n'
  335 |     RMT.chnconf0[mRMT_channel].conf_update_n = 1;
      |                                ^~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:336:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'tx_start_n'
  336 |     RMT.chnconf0[mRMT_channel].tx_start_n = 1;
      |                                ^~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp: In static member function 'static void ESP32RMTController::doneOnChannel(rmt_channel_t, void*)':
c:\Users\User\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:389:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'tx_stop_n'
  389 |     RMT.chnconf0[channel].tx_stop_n = 1;
      |                           ^~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:390:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'conf_update_n'
  390 |     RMT.chnconf0[channel].conf_update_n = 1;
      |                           ^~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:392:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'mem_rd_rst_n'
  392 |     RMT.chnconf0[channel].mem_rd_rst_n = 1;
      |                           ^~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:393:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'mem_rd_rst_n'
  393 |     RMT.chnconf0[channel].mem_rd_rst_n = 0;
      |                           ^~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:394:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'apb_mem_rst_n'
  394 |     RMT.chnconf0[channel].apb_mem_rst_n = 1;
      |                           ^~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:395:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'apb_mem_rst_n'
  395 |     RMT.chnconf0[channel].apb_mem_rst_n = 0;

Uspizig, 2024-05-31:
I guess something simlar then here: https://github.com/adafruit/Adafruit_NeoPixel/pull/367

niiiksh, 2024-06-01:
@Uspizig I can confirm it is related to esp32 from Espressif v3.0.0. Reverting back to 2.0.17 (also reverted to FastLED 3.6.0 idk if 3.7.0 will work) is a workaround until FastLED developers check a build against new esp32 libraries.

Contact-Aschmu, 2024-06-16:
@niiiksh and @Uspizig The same problems are occurring with the new Espressif v3.0.0 and v3.0.1 changing back to v2.0.17 works with FastLED for v3.6.0 and v3.7.0

zhaoker, 2024-07-11:
> @niiiksh and @Uspizig The same problems are occurring with the new Espressif v3.0.0 and v3.0.1
> 
> To fix the problem: as @Uspizig said changing back to v2.0.17 works with FastLED for v3.6.0, I found that it also works for v3.7.0

THANKS ,  When I changing back , everything goes right...

scottchiefbaker, 2024-07-20:
I can also confirm compiling FastLED 3.7.0 on an ESP32-S2 with Espressif v3.0.1 gives the above error message. Reverting to Espressif v2.0.17 solves the problem. 

For now I will use the older version of Espressif, but it would be nice to be able to use FastLED with the version 3.0.x.

DKasle, 2024-08-04:
I encounter the errors while using a Tiny S3 microcontroller with the Arduino IDE.  Unfortunately, I cannot revert to esp32 v2.0.17 because that version does not recognize the Tiny S3 board; it has to be v3.0.0 or later.  I am using Arduino IDE v2.3.2, FastLED v3.7.0 and esp32 v3.0.3. 
Any suggestions on how to fix this situation would be much appreciated!

zackees, 2024-08-04:
Seems like an easy-ish fix.

Can someone provide repro instructions? I don't ever use the Arduino IDE anymore and didn't know you could upgrade the ESP32 toolchain like this.

DKasle, 2024-08-04:
To reproduce this issue: 
Try this sample code to compile with UM TinyS3 board selected in Arduino IDE 2.3.2, FastLED 3.7.0, esp32 3.0.3 and FreeRTOS 11.1.0-2

#include <FastLED.h>

# include <stdio.h>
# include <stdlib.h>

# include "Arduino.h"
# include "math.h"

#include <FreeRTOS.h>
#include <SPI.h>         

/*********
  Rui Santos
  Complete project details at https://randomnerdtutorials.com  
*********/

TaskHandle_t Task1;
TaskHandle_t Task2;

// LED pins
const int led1 = 2;
const int led2 = 4;

void setup() {
  Serial.begin(115200); 
  pinMode(led1, OUTPUT);
  pinMode(led2, OUTPUT);

  //create a task that will be executed in the Task1code() function, with priority 1 and executed on core 0
  xTaskCreatePinnedToCore(
                    Task1code,   /* Task function. */
                    "Task1",     /* name of task. */
                    10000,       /* Stack size of task */
                    NULL,        /* parameter of the task */
                    1,           /* priority of the task */
                    &Task1,      /* Task handle to keep track of created task */
                    0);          /* pin task to core 0 */                  
  delay(500); 

  //create a task that will be executed in the Task2code() function, with priority 1 and executed on core 1
  xTaskCreatePinnedToCore(
                    Task2code,   /* Task function. */
                    "Task2",     /* name of task. */
                    10000,       /* Stack size of task */
                    NULL,        /* parameter of the task */
                    1,           /* priority of the task */
                    &Task2,      /* Task handle to keep track of created task */
                    1);          /* pin task to core 1 */
    delay(500); 
}

//Task1code: blinks an LED every 1000 ms
void Task1code( void * pvParameters ){
  Serial.print("Task1 running on core ");
  Serial.println(xPortGetCoreID());

  for(;;){
    digitalWrite(led1, HIGH);
    delay(1000);
    digitalWrite(led1, LOW);
    delay(1000);
  } 
}

//Task2code: blinks an LED every 700 ms
void Task2code( void * pvParameters ){
  Serial.print("Task2 running on core ");
  Serial.println(xPortGetCoreID());

  for(;;){
    digitalWrite(led2, HIGH);
    delay(700);
    digitalWrite(led2, LOW);
    delay(700);
  }
}

void loop() {
  
}

zackees, 2024-08-04:
Uh... so hard to jump through code through this arduino system. Platform IO allows jumping to the symbol declaration to see what's there.

Here's a platform.ini file to use the new esp-idf 5.X to compile.

If you can make a simple demo of this it will help greatly (notice in this example I'm using the ESP32C6).

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32c6]
platform = https://github.com/zackees/platform-espressif32#Arduino/IDF5
framework = arduino
board = esp32-c6-devkitc-1

monitor_speed = 115200

;platform = espressif32@^6.3.2

upload_speed = 500000

upload_protocol = esptool

monitor_filters = 
	default
	esp32_exception_decoder

board_build.f_cpu = 80000000L

```

zackees, 2024-08-04:
Here's a platformio repo reproducing this bug:

https://github.com/zackees/fastled_bug_s3_idf5x

DKasle, 2024-08-05:
zackees: Thanks for your suggestions.  I'm not familiar with Platform IO so this will take some time to research.

zackees, 2024-08-05:
I've got a fix, but I can't test it on live hardware.

I'll create a fastled branch and stash the fix there. I'll update the code repo above to point to the branch. Then I'll comment on what you need to do to test it.

The fix is actually very easy - espressif simply changed the names of their variables (#*#@*@) and there needs to simply be an #if-#else-#endif to check which version of the toolchain is in use.

# Side Note

PlatformIO is WAAAAAAAAAAY better than the Arduino IDE. You get all the auto-complete stuff, co-pilot and jumping to code declarations/defintions with a right click because it's just a plugin to VSCode. It's much easier to point to a github fork of whatever dependency you are working. For example I can just point it to github branch of fastled and it pull in all the dependencies and away you go. The platformio.ini, which defines your project dependencies, is stashed with the project. ArduinoIDE wants to make everything global. It's such a mess. Once you use platformIO once for a project, you'll never want to ever go back to the ArduinoIDE, which will seem like a toy in comparison to a real build system.

zackees, 2024-08-05:
Okay, I've created a fix in a branch. This is what you'll need to do to test it:

  1. Install VSCode
  2. Open up the extensions panel and install platformio
  3. `git clone https://github.com/zackees/fastled_bug_s3_idf5x`
  4. Open up `fastled_bug_s3_idf5x` directory in VSCode
  5. Wait a couple of minutes for the dependencies to install. If things take longer than 10 mins then cancel the update terminal (it's safe) and hit the compile button, it's the check mark button on the bottom menu bar:
  
  
[image]

Now you should see that your project compiles fine. If the dependencies weren't completely installed it will do it now then compile your program.

  6. Hit the right arrow (see picture above) which will upload the firmware to your s3 device.
  7. Verify that it works (or doesn't) and report back to here, tagging me with the @zackees symbol so that i get a notification.

Also please ping this pull request and mention that this change does infact work on a real hardware device.

https://github.com/FastLED/FastLED/pull/1652

zackees, 2024-08-05:
The faster this fix is tested, the faster we can mint a new fastled version so the fix can go in everywhere.

zackees, 2024-08-08:
Compiler error fixed in FastLED 3.7.1 bugfix release.

closed 2024-08-08 by zackees

DKasle, 2024-08-08:
@zackees: Thank you!  I can confirm this fixes the compile error mentioned previously for the UM Tiny S3 microcontroller.

zackees, 2024-08-09:
> @zackees: Thank you! I can confirm this fixes the compile error mentioned previously for the UM Tiny S3 microcontroller.

That microcontroller is actually the target board in the git repo above.

RubenTeLoo, 2024-08-21:
Hi I am currently still experiencing the same issue with the newest (FastLED 3.7.3) on a lolin s2 mini. Can that be possible or do I miss something. Tried to install the specific version (3.7.1) you said the bug fix was in just to be sure, but it did not help.

RubenTeLoo, 2024-08-21:
> Compiler error fixed in FastLED 3.7.1 bugfix release.

Hi I am currently still experiencing the same issue with the newest (FastLED 3.7.3) on a lolin s2 mini. Can that be possible or do I miss something. Tried to install the specific version (3.7.1) you said the bug fix was in just to be sure, but it did not help.

zackees, 2024-08-30:
Okay - thanks for your build tips. I've repro'd the failure. I put in a compiler fix but it's likely tragically wrong. If you are using this for the WS2812 chipset let me know if it doesn't work.

RubenTeLoo, 2024-08-30:
I am using the WS2811 chipset

zackees, 2024-08-30:
My only advice is to pop in to the code base and try to fix it and issue a pull request.

RubenTeLoo, 2024-08-30:
I will try but C isn't my best language :)

RubenTeLoo, 2024-08-30:
Okey, I think it is just some weird shit from arduino IDE. (as you recommended) I switched to platformIO, loaded up the firstlight example and it compiled with no error. Gonna test if uploading also works.

zackees, 2024-08-30:
Not sure if you are using the repo or not, but I recommend you clone this repo and open it up in VSCode. Everything is all setup to compile and edit code.

RubenTeLoo, 2024-08-30:
Yeah, just downloaded it from platformIO as library

zackees, 2024-08-30:
If you do it the way I'm recommending then FastLED code will be symlinked into your code base and any changes will punch through to the actual library instead of a copy of the library in your build directory that you have to remember to copy. This is how I'm able to test and iterate on things so quickly.

If you import FastLED as a library then you only get a copy of the library. Sometimes platformio will delete your build directory with changes to the platformio.ini file and then you lose whatever you were working on.

When you are done you can just issue a pull request directly from the repo.

RubenTeLoo, 2024-08-30:
Ow sorry I meant the library works fine, I didn't change anything. I only switched from AruinoIDE to platformIO and then it compiled with no error.