FastLED/FastLED #1629: FastLED 3.7.0 compile error with Arduino/Pico

CliveMcF, 2024-06-01:
Compiling version 3.7.0 for a Raspberry Pi Pico W using Earle F. Philhower's ArduinoPico core in the Arduino IDE gives these errors:
{standard input}: Assembler messages:
{standard input}:9169: Error: invalid offset, value too big (0x00000530)
{standard input}:9196: Error: invalid offset, value too big (0x000004EC)
Rolling back to version 3.6.0 works fine so it's not a major issue for me, posting this in case it helps the developer troubleshoot the problem.

voidptr, 2024-06-07:
Also experiencing this. Downgrading to 3.6.0 restores compilation.

CraCaNN, 2024-06-12:
I'm also having this issue and downgrading to 3.6.0 also fixes the issue for me. 
In my project I'm using a Pico with earlephilhower Arduino-pico board manager with Adafruit_tinyUSB USB stack instead of the standard Pico-SDK USB Stack. 
I've worked out that I can compile with the library included but as soon as I want to add an LED in setup such as `FastLED.addLeds<WS2812B, DATA_PIN, RGB>(leds, NUM_LEDS);` the compiler fails using FastLED 3.7.0 with an almost identical error message to what CliveMcF shows.

bchwtz, 2024-08-06:
I can also confirm this error when using version 3.7.0; downgrading solves this problem. No library example compiles for me. Is there any fix so that the most recent version works with the [arduino-pico](https://github.com/earlephilhower/arduino-pico) core again?

darrenbeckley, 2024-08-08:
I ran into the same problem with FastLED 3.7.0 and [ardiuo-pico](https://github.com/earlephilhower/arduino-pico). I am using PlatformIO with [platform-raspberrypi](https://github.com/maxgerhardt/platform-raspberrypi/) and was getting a similar assembler error. 

There is a setting added in #1576 which I used to solve this problem in my build. Try setting FASTLED_RP2040_CLOCKLESS_M0_FALLBACK to zero before including FastLED.h:

~~~
#include <Arduino.h>
#define FASTLED_RP2040_CLOCKLESS_M0_FALLBACK 0
#include <FastLED.h>
~~~

If using PlatformIO you could instead set this in platformio.ini:

~~~
build_flags =
    -DFASTLED_RP2040_CLOCKLESS_M0_FALLBACK=0
~~~

My project now builds without the assembler error. I have tested on a Pimoroni Plasma Stick 2040 W, which is essentially a Raspberry Pi Pico W with some extra hardware, as well as a Pimoroni Plasma 2040.

pierrotm777, 2024-08-09:
Hello, 

I try your 3.7.1 with my **Waveshare RP2040 zero** and have always a compilation error:

In file included from c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\FastLED\src/FastLED.h:79,
                 from C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\Sound_Christian\Sound_Christian_RP2040_Ph\Sound_Christian_RP2040_Ph.ino:56:
c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\FastLED\src/fastspi.h:172:23: note: '#pragma message: Forcing software SPI - no hardware SPI for you!'
  172 | #      pragma message "Forcing software SPI - no hardware SPI for you!"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\Sound_Christian\Sound_Christian_RP2040_Ph\Sound_Christian_RP2040_Ph.ino: In function 'void xanyFilter()':
C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\Sound_Christian\Sound_Christian_RP2040_Ph\Sound_Christian_RP2040_Ph.ino:934:29: warning: comparison is always true due to limited range of data type [-Wtype-limits]
  934 |   if ((arg != NULL) && (Val8>=0 && Val8<=3))
      |                         ~~~~^~~
C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\Sound_Christian\Sound_Christian_RP2040_Ph\Sound_Christian_RP2040_Ph.ino: In function 'void unrecognized(const char*)':
C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\Sound_Christian\Sound_Christian_RP2040_Ph\Sound_Christian_RP2040_Ph.ino:997:32: warning: unused parameter 'command' [-Wunused-parameter]
  997 |  void unrecognized(const char *command) {
      |                    ~~~~~~~~~~~~^~~~~~~
C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\Sound_Christian\Sound_Christian_RP2040_Ph\PPM_Read.ino: In function 'void interruptHook()':
C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\Sound_Christian\Sound_Christian_RP2040_Ph\PPM_Read.ino:35:52: warning: comparison of constant '20000' with boolean expression is always true [-Wbool-compare]
   35 |     if (PAUSE_PULSE_MIN_LENGTH < pulses[pulses_nr] < PAUSE_PULSE_MAX_LENGTH) {
      |                                                    ^
C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\Sound_Christian\Sound_Christian_RP2040_Ph\PPM_Read.ino:35:32: warning: comparisons like 'X<=Y<=Z' do not have their mathematical meaning [-Wparentheses]
   35 |     if (PAUSE_PULSE_MIN_LENGTH < pulses[pulses_nr] < PAUSE_PULSE_MAX_LENGTH) {
      |                                ^
C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\Sound_Christian\Sound_Christian_RP2040_Ph\PPM_Read.ino: In function 'void setup_CPPM(uint8_t)':
C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\Sound_Christian\Sound_Christian_RP2040_Ph\PPM_Read.ino:63:25: warning: unused parameter 'InputCppmPin' [-Wunused-parameter]
   63 | void setup_CPPM(uint8_t InputCppmPin) {
      |                 ~~~~~~~~^~~~~~~~~~~~
{standard input}: Assembler messages:
{standard input}:5519: Error: invalid offset, value too big (0x0000052C)
{standard input}:5546: Error: invalid offset, value too big (0x000004E8)
{standard input}:6748: Error: invalid offset, value too big (0x0000052C)
{standard input}:6775: Error: invalid offset, value too big (0x000004E8)

I return to the 3.6.0 version.

zackees, 2024-09-04:
I think this is now fixed. Sometimes compiled sizes of functions overflow their address space, especially in debug mode. I've gone ahead and forced optimization to minimize compiled code size for this one function header and all the code it includes.

If this bug is still happening then go ahead and re-open it and we'll take another look.

CliveMcF, 2024-09-04:
I downloaded and installed the 3.7.5 zip file from Github but still get the same compile errors:

{standard input}: Assembler messages:
{standard input}:10361: Error: invalid offset, value too big (0x00000528)
{standard input}:10388: Error: invalid offset, value too big (0x000004E4)
{standard input}:11589: Error: invalid offset, value too big (0x0000052C)
{standard input}:11616: Error: invalid offset, value too big (0x000004E8)

zackees, 2024-09-04:
@maxgerhardt

zackees, 2024-09-04:
Looks like a bug in the toolchain

pierrotm777, 2024-09-04:
Same issue with my RP2040 zero:

{standard input}: Assembler messages:
{standard input}:5696: Error: invalid offset, value too big (0x0000052C)
{standard input}:5723: Error: invalid offset, value too big (0x000004E8)
{standard input}:6925: Error: invalid offset, value too big (0x0000052C)
{standard input}:6952: Error: invalid offset, value too big (0x000004E8)

zackees, 2024-09-04:
Does setting the following define resolve the issue?

FASTLED_RP2040_CLOCKLESS_M0_FALLBACK=1

pierrotm777, 2024-09-04:
> Does setting the following define resolve the issue?
> 
> FASTLED_RP2040_CLOCKLESS_M0_FALLBACK=1

Where i add that ?

zackees, 2024-09-04:
If the above solves the issue then the problem might be the template expansions getting too large for their code blocks.

https://github.com/FastLED/FastLED/blob/master/src/platforms/arm/rp2040/clockless_arm_rp2040.h#L86C5-L86C138

I see a next inner template of a template class. It's possible that this is getting backed in to the same code address as the enclosing class.

If so then I think the inner template function of the template class can be made to just a template function:

instead of

template class C:
  template function f

we just have

template function: f

template class C: uses f<>

pierrotm777, 2024-09-04:
Add **#define FASTLED_RP2040_CLOCKLESS_M0_FALLBACK 1** not solve the problem.

CliveMcF, 2024-09-04:
Same for me, still the same errors. For reference I'm using:

```cpp
#define FASTLED_INTERNAL  // Disable message with version during compile
#define FASTLED_RP2040_CLOCKLESS_M0_FALLBACK 1

FASTLED_USING_NAMESPACE  //Memory allocation for LED strip
```

pierrotm777, 2024-09-04:
I think have found my solution [here](https://github.com/FastLED/FastLED/issues/1629#issuecomment-2275336557) .
#define FASTLED_RP2040_CLOCKLESS_M0_FALLBACK 0
#include <FastLED.h>

I can now compile the 3.7.5 lib without error :-)

CliveMcF, 2024-09-04:
Genius! Thanks @pierrotm777, setting FASTLED_RP2040_CLOCKLESS_M0_FALLBACK 0 worked for me too.

zackees, 2024-09-04:
Okay I've gone ahead and made this FASTLED_RP2040_CLOCKLESS_M0_FALLBACK default disabled. This should fix the issue.

The comments mention that the default is disabled, but the actual define is default enabled.

closed 2024-09-04 by zackees