FastLED/FastLED #1631: Issues with ESP32-S2 Release v3.0.1(fastled 3.7)

marcobrianza, 2024-06-08:
Hello, building the blink sketch on a loin esp32-s2 does not work anymore upgrading to ESP32 core 3.0.1 and fasteld 3.7

the errors are the following:

/Users/marco/Nextcloud-marco/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp: In member function 'void ESP32RMTController::tx_start()':
/Users/marco/Nextcloud-marco/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:339:37: error: 'volatile union rmt_chnconf1_reg_t' has no member named 'mem_rd_rst'
  339 |     RMT.conf_ch[mRMT_channel].conf1.mem_rd_rst = 1;
      |                                     ^~~~~~~~~~
/Users/marco/Nextcloud-marco/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:340:37: error: 'volatile union rmt_chnconf1_reg_t' has no member named 'mem_rd_rst'
  340 |     RMT.conf_ch[mRMT_channel].conf1.mem_rd_rst = 0;
      |                                     ^~~~~~~~~~
/Users/marco/Nextcloud-marco/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:347:37: error: 'volatile union rmt_chnconf1_reg_t' has no member named 'tx_start'
  347 |     RMT.conf_ch[mRMT_channel].conf1.tx_start = 1;
      |                                     ^~~~~~~~
/Users/marco/Nextcloud-marco/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp: In static member function 'static void ESP32RMTController::doneOnChannel(rmt_channel_t, void*)':
/Users/marco/Nextcloud-marco/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:400:32: error: 'volatile union rmt_chnconf1_reg_t' has no member named 'tx_stop'
  400 |     RMT.conf_ch[channel].conf1.tx_stop = 1;
      |                                ^~~~~~~
/Users/marco/Nextcloud-marco/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:402:32: error: 'volatile union rmt_chnconf1_reg_t' has no member named 'mem_rd_rst'
  402 |     RMT.conf_ch[channel].conf1.mem_rd_rst = 1;
      |                                ^~~~~~~~~~
/Users/marco/Nextcloud-marco/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:403:32: error: 'volatile union rmt_chnconf1_reg_t' has no member named 'mem_rd_rst'
  403 |     RMT.conf_ch[channel].conf1.mem_rd_rst = 0;
      |                                ^~~~~~~~~~
Using library FastLED at version 3.7.0 in folder: /Users/marco/Nextcloud-marco/Arduino/libraries/FastLED 
exit status 1
Error compiling for board LOLIN S2 Mini.

samguyer, 2024-06-08:
@marcobrianza Thanks for the heads-up. I have seen a few other issues with the new versions of FastLED and ESP-IDF. I'll take a look as soon as I can.

jbry, 2024-06-12:
Looks like it works if you just update the following:

Rename mem_rd_rst to mem_rd_rst_chn
Rename tx_start to tx_start_chn
Rename tx_stop to tx_stop_chn

marcobrianza, 2024-06-14:
@acalatrava with this fix S3 code compiles but S2 still produces errors in my test.

tresler, 2024-06-27:
Thanks for jbry, Now on my ESP32S3 after changing clockless_rmt_esp32.cpp with mem_rd_rst_n to mem_rd_rst_chn I can comple and upload fastled example scatch. But now I have new  problem:

'''E (93) rmt(legacy): CONFLICT! driver_ng is not allowed to be used with the legacy driver
abort() was called at PC 0x4200807b on core 0
E (122) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
E (130) esp_core_dump_elf: Elf write init failed!
E (135) esp_core_dump_common: Core dump write failed with error=-1'''

jbry, 2024-07-09:
@tresler  - I just pulled that commit into the file (as I was having that issue once I tried something  besides one of the demos) - and that seemed to fix the issue. Read the comments at the top of the example though on how to clear the build cache to make sure you get the build_opt.h taken into account.

zackees, 2024-08-08:
Compiler error fixed in FastLED 3.7.1 bugfix release.

closed 2024-08-08 by zackees