FastLED/FastLED #1632: Pragma warning in FastLED 3.7.0 even with #FASTLED_INTERNAL defined

nomakewan, 2024-06-10:
I am compiling for ESP32, and am using the Expressif Systems ESP32 dev library 2.0.16 (not 3.0.x). In my project I have `#FASTLED_INTERNAL` defined before calling `#include <FastLED.h>`. With FastLED 3.6.0, no warnings appear in my compilation report. After installing FastLED 3.7.0 however, I get the following warning:

```cpp
In file included from C:\Arduino\libraries\FastLED\src\FastLED.h:79,
                 from C:\Arduino\libraries\FastLED\src\five_bit_hd_gamma.cpp:1:
C:\Arduino\libraries\FastLED\src\fastspi.h:157:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

This appears to be due to the new "Five bit HD gamma" code that was added in 3.7.0. Perhaps an error in the order of includes, or a missing include guard?

zackees, 2024-08-31:
Fixed in latest. Will go out on monday.

closed 2024-08-31 by zackees

nomakewan, 2024-08-31:
Thank you so much! I really appreciate it. Have a great one!