FastLED/FastLED #1633: blend8 maths is wrong

notzed, 2024-06-13:
Using real numbers this is the actual calculation desired, where m=[0.0, 1.0] is the interpolation factor.

    result = round(a + (b - a) * m)                          (1)

But your blend function uses this:

    result = (  A*(255-M) + A + B*M + B      ) / 256

But where did "B*M + B" even come from?  It seems to have no mathematical basis.

The root of the problem is that with a scaling range of M=[0,255] with an 8 bit shift the actual range is M = [0,(255/256)] = [0.0,0.996] rather than m = [0.0,1.0].  One easy fix is to simply allow M the range [0,256].

A more serious problem is that it's truncating the result rather than rounding  so for small values of A and B the interpolation is very poor.  e.g. If A=0, B=1, the result is mostly 0.  i.e. for M=[0,254] you will get 0, and M=255 you will get 1.  I've only played with 1 led strip but this makes the already bad low intensity performance even worse.

This calculation will not reach the full range for some values if M=[0,255], but it will be smoother for all of them:

    result = (((A << 8) + (B - A) * M + 0x80) >> 8           (2)

A more accurate fixed point version of (1) requires more bits and a couple more steps:

    result = ((A << 16) + (B - A) * M * 257 + 0x8000) >> 16  (3)

Here (M * 257) is an 8.8 fixed point approximation of (M * 256 / 255) to scale M=[0,255] to the range [0,256] and 0x8000 is half of 1<<16 for rounding.

zackees, 2024-08-26:
Unit testing this type of code is interesting.

I just went through this yesterday with another project and my notes are pretty much this: you have to have a sub folder of pure portable code which includes no arduino.h, esp, avr etc headers.

This pure portable code can then be unit tested with your favorite framework.

The next question is how to deploy the gcc compiler framework so it works on mac/win/linux without sudo access, as system package managers need this.

The answer to this surprisingly is to use the zig C++ compiler and install it via pip, which doesn't require elevated privileges.

***

As for the blend function, a lot of this stuff was written when the Arduino uno was all the rage. The strategy we have is that we have upgraded these little functors with a better version for 32 bit chips but still use the old version for avr chips, controlled by a define.

I've done a lot of work to make FastLED as easy as possible to hack on it. Now you can just clone the repo and then open it up in VSCode (+platformio) and hit compile / upload.

zackees, 2024-08-26:
@notzed was this fix suggestion hypothetical or did you use it in code and verify that it works correctly?

zackees, 2024-08-27:
> Platformio has a framework to run unit tests on the device and/or natively.

I've used native. Anything including "Arduino.h" will fail to compile. So unless you make a fake "Arduino.h", the value is debatable on exactly what you are testing, especially since you have to verify your expectations **by capturing the stdout from the device and checking string values**.

Hence the reason having a lib of just pure functions / classes not calling into any platform stuff. You get to use fancy expectation stuff in the testing frameworks and possibly get to use gdb to step through the code while it executes.

What would be better is to use qemu to fake an esp32 and read back the printfs. I've looked that qemu for esp32 and there is a github action to run it. The downside is that it wants partition.csv and a lot of other stuff that is abstracted away from platformio when using the arduino core. It could probably work with esp-idf framework for platformio, though.

zackees, 2024-08-27:
> Testing race conditions in DMA interrupts and flickering is just a different problem than confirming that a saturating add didn't sign extend in a dumb place.

It's doable with a custom github runner that tunnels back to a real machine. I've done it before with android. The great thing is once you break out of the VM and into real hardware you can pretty much do whatever you want. For example, the WS2812 protocol running on the ESP32 could be read back either via another micro or some USB data signal sniffer to verify that no timings are missed.

NiekLambert, 2024-12-19:
Back to the original issue: blend8 simply does not have the right behaviour. An iterative blend makes it even worse: it will typically never reach the overlay color and can even easily go outside the range of the two input colors. Rounding could be made at least neutral by adding 128 before the >>8 operation but that still does not solve all issues.

For a (possibly slow) transition from color p1 to p2 you could alternatively use a bounded difference as a more linear transition function CRGB(
p1.r + min(max(-amountOfP2, p2.r - p1.r), amountOfP2),
p1.g + min(max(-amountOfP2, p2.g - p1.g), amountOfP2),
p1.b + min(max(-amountOfP2, p2.b - p1.b), amountOfP2))
This is visually OK to me, has utterly robust behavior (stays within range p1 .. p2 and always iterates fully to p2) and executes fast. If you not want to break the original blend() definition you could perhaps add it as a new function (fadeTo() or so) and refer to it in the blend documentation?

zackees, 2024-12-19:
Okay great, i'll put this in.

closed 2025-12-18 by zackees