FastLED/FastLED #1635: Compilation Error

pierrotm777, 2024-06-18:
Hello,
I use your FastLed lib with success with a Waveshare RP2040 Zero board and live it.
Bit if y use the 3.7.0 version I have a compilation error.
3.6.0 is OK for me.
Can you solve that please ?

My use:
#include <FastLED.h>// la version 3.7.0 ne compile pas. la 3.6.0 est OK
#define NUM_LEDS 1
#define DATA_PIN 16
CRGB leds[NUM_LEDS];

On my setup:
FastLED.addLeds<WS2812B, DATA_PIN, RGB>(leds, NUM_LEDS);

An a blink function by example:
void blinkNTimeYELLOW(int count, int onInterval, int offInterval)
{
  byte i;
  for (i = 0; i < count; i++) 
  {
    FastLED.delay(offInterval);
    leds[0] = CRGB::OrangeRed;
    FastLED.delay(onInterval);
    leds[0] = CRGB::Black;
  }
}

samguyer, 2024-06-18:
@pierrotm777 What is the compiler error you are seeing?

samguyer, 2024-06-18:
Ah ok, you're getting the "invalid offset" error. We've had a bunch of reports of this problem. We're looking into it. Thanks!

pierrotm777, 2024-06-18:
Thanks to you :-)

pierrotm777, 2024-08-13:
I try your new 3.7.1 and have always the same error :-)
`
FQBN: rp2040:rp2040:waveshare_rp2040_zero
Using board 'waveshare_rp2040_zero' from platform in folder: C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.9.5
Using core 'rp2040' from platform in folder: C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.9.5

Detecting libraries used...
C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\2.2.0-d04e724/bin/arm-none-eabi-g++ -I C:\Users\pierrot\AppData\Local\Temp\arduino\sketches\7903C059F93DA6EA3277D1933AFC52C2/core -c -Werror=return-type -Wno-psabi -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSBD_PID=0x0003 -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=500 -DUSB_MANUFACTURER="Waveshare" -DUSB_PRODUCT="RP2040 Zero" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=1 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -DARDUINO_VARIANT="waveshare_rp2040_zero" -DTARGET_RP2040 -DPICO_FLASH_SIZE_BYTES=2097152 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS -iprefixC:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.9.5/ @C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.9.5/lib/platform_inc.txt -IC:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.9.5/include -fno-rtti -std=gnu++17 -g -pipe -w -x c++ -E -CC -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_WAVESHARE_RP2040_ZERO -DBOARD_NAME="WAVESHARE_RP2040_ZERO" -DARDUINO_ARCH_RP2040 -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE -IC:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.9.5\cores\rp2040 -IC:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.9.5\variants\waveshare_rp2040_zero C:\Users\pierrot\AppData\Local\Temp\arduino\sketches\7903C059F93DA6EA3277D1933AFC52C2\sketch\Sound_Christian_RP2040_Ph.ino.cpp -o nul
Alternative pour EEPROM.h: [EEPROM@1.0]
ResolveLibrary(EEPROM.h)
  -> candidates: [EEPROM@1.0]
C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\2.2.0-d04e724/bin/arm-none-eabi-g++ -I C:\Users\pierrot\AppData\Local\Temp\arduino\sketches\7903C059F93DA6EA3277D1933AFC52C2/core -c -Werror=return-type -Wno-psabi -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSBD_PID=0x0003 -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=500 -DUSB_MANUFACTURER="Waveshare" -DUSB_PRODUCT="RP2040 Zero" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=1 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -DARDUINO_VARIANT="waveshare_rp2040_zero" -DTARGET_RP2040 -DPICO_FLASH_SIZE_BYTES=2097152 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS -iprefixC:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.9.5/ @C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.9.5/lib/platform_inc.txt -IC:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.9.5/include -fno-rtti -std=gnu++17 -g -pipe -w -x c++ -E -CC -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_WAVESHARE_RP2040_ZERO -DBOARD_NAME="WAVESHARE_RP2040_ZERO" -DARDUINO_ARCH_RP2040 -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE -IC:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.9.5\cores\rp2040 -IC:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.9.5\variants\waveshare_rp2040_zero -IC:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.9.5\libraries\EEPROM\src C:\Users\pierrot\AppData\Local\Temp\arduino\sketches\7903C059F93DA6EA3277D1933AFC52C2\sketch\Sound_Christian_RP2040_Ph.ino.cpp -o nul
Alternative pour RcBusRx.h: [RcNavyRcBusRx]
ResolveLibrary(RcBusRx.h)
  -> candidates: [RcNavyRcBusRx]
C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\2.2.0-d04e724/bin/arm-none-eabi-g++ -I C:\Users\pierrot\AppData\Local\Temp\arduino\sketches\7903C059F93DA6EA3277D1933AFC52C2/core -c -Werror=return-type -Wno-psabi -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSBD_PID=0x0003 -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=500 -DUSB_MANUFACTURER="Waveshare" -DUSB_PRODUCT="RP2040 Zero" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=1 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -DARDUINO_VARIANT="waveshare_rp2040_zero" -DTARGET_RP2040 -DPICO_FLASH_SIZE_BYTES=2097152 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS -iprefixC:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.9.5/ @C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.9.5/lib/platform_inc.txt -IC:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.9.5/include -fno-rtti -std=gnu++17 -g -pipe -w -x c++ -E -CC -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_WAVESHARE_RP2040_ZERO -DBOARD_NAME="WAVESHARE_RP2040_ZERO" -DARDUINO_ARCH_RP2040 -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE -IC:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.9.5\cores\rp2040 -IC:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.9.5\variants\waveshare_rp2040_zero -IC:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.9.5\libraries\EEPROM\src -Ic:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\RcNavyRcBusRx C:\Users\pierrot\AppData\Local\Temp\ [... cut ...]
Utilisation de la bibliothèque DFPlayer_Mini_Mp3 prise dans le dossier: C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\DFPlayer_Mini_Mp3 (héritage)
Utilisation de la bibliothèque PPM-reader version 1.2.0 dans le dossier: C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\PPMReader 
Utilisation de la bibliothèque RcNavyRcRxSerial prise dans le dossier: C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\RcNavyRcRxSerial (héritage)
exit status 1

Compilation error: exit status 1`

Thanks for your help

zackees, 2024-08-28:
We need the whole sketch.

This assembler message says the jump instruction is set be larger than the allowable address range. The function is question incoming from your code, which you haven't posted.

zackees, 2024-08-28:
> We need the whole sketch.
> 
> This assembler message says the jump instruction set is larger than the allowable address range. The function is question incoming from your code, which isn't posted.

We've reprod this on the RPXXXX platforms. And are currently getting this on the RP2350.

So we are at least tracking it. Looks like a platform issue with assembly generation

zackees, 2024-08-31:
We are now running RPXXXX boards in test. This problem seems to have gone away. If this is not the case then please re-open.

closed 2024-08-31 by zackees