FastLED/FastLED #1638: Compile error

Haijo18, 2024-06-24:
Hello,
I get an error using FastLED with an Arduino Nano Every. I'm using version 3.7.0. Even when switching back to 3.6.0, I get the same error. I use it in combination with Arduino IDE 2.3.2. Here's a minimal code example:

    #include <FastLED.h>
    #define NUM_LEDS 1
    #define DATA_PIN 14
    
    CRGB leds[NUM_LEDS];
    
    void setup() {
      FastLED.addLeds<WS2812B, DATA_PIN, RGB>(leds, NUM_LEDS);
      leds[0] = CRGB::Green;
      FastLED.show();
    }

If I remove `FastLED.addLeds<WS2812B, DATA_PIN, RGB>(leds, NUM_LEDS);`, the sketch compiles. The error message displayed is as follows:

    In file included from C:\Users\haijo\OneDrive\Dokumente\Arduino\libraries\FastLED\src/platforms/avr/fastled_avr.h:4:0,
                     from C:\Users\haijo\OneDrive\Dokumente\Arduino\libraries\FastLED\src/platforms.h:49,
                     from C:\Users\haijo\OneDrive\Dokumente\Arduino\libraries\FastLED\src/FastLED.h:66,
                     from C:\Users\haijo\OneDrive\Dokumente\Arduino\sketch_aug28a\sketch_aug28a.ino:1:
    C:\Users\haijo\OneDrive\Dokumente\Arduino\libraries\FastLED\src/platforms/avr/fastpin_avr.h: In instantiation of 'static void _AVRPIN<PIN, _MASK, _PORT, _DDR, _PIN>::setOutput() [with unsigned char PIN = 14; unsigned char _MASK = 8; _PORT = __gen_struct_VPORTD_OUT; _DDR = __gen_struct_VPORTD_DIR; _PIN = __gen_struct_VPORTD_IN]':
    C:\Users\haijo\OneDrive\Dokumente\Arduino\libraries\FastLED\src/platforms/avr/clockless_trinket.h:114:31:   required from 'void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::init() [with unsigned char DATA_PIN = 14; int T1 = 4; int T2 = 10; int T3 = 6; EOrder RGB_ORDER = (EOrder)10; int XTRA0 = 0; bool FLIP = false; int WAIT_TIME = 10]'
    C:\Users\haijo\OneDrive\Dokumente\Arduino\sketch_aug28a\sketch_aug28a.ino:11:1:   required from here
    C:\Users\haijo\OneDrive\Dokumente\Arduino\libraries\FastLED\src/platforms/avr/fastpin_avr.h:22:42: error: incomplete type '__gen_struct_VPORTD_DIR' used in nested name specifier
      inline static void setOutput() { _DDR::r() |= _MASK; }
                                       ~~~~~~~^~
    In file included from C:\Users\haijo\OneDrive\Dokumente\Arduino\libraries\FastLED\src/platforms/avr/fastled_avr.h:4:0,
                     from C:\Users\haijo\OneDrive\Dokumente\Arduino\libraries\FastLED\src/platforms.h:49,
                     from C:\Users\haijo\OneDrive\Dokumente\Arduino\libraries\FastLED\src/FastLED.h:66,
                     from C:\Users\haijo\OneDrive\Dokumente\Arduino\sketch_aug28a\sketch_aug28a.ino:1:
    C:\Users\haijo\OneDrive\Dokumente\Arduino\libraries\FastLED\src/platforms/avr/fastpin_avr.h: In instantiation of 'static volatile uint8_t* _AVRPIN<PIN, _MASK, _PORT, _DDR, _PIN>::port() [with unsigned char PIN = 14; unsigned char _MASK = 8; _PORT = __gen_struct_VPORTD_OUT; _DDR = __gen_struct_VPORTD_DIR; _PIN = __gen_struct_VPORTD_IN; _AVRPIN<PIN, _MASK, _PORT, _DDR, _PIN>::port_ptr_t = volatile unsigned char*; uint8_t = unsigned char]':
    C:\Users\haijo\OneDrive\Dokumente\Arduino\libraries\FastLED\src/platforms/avr/clockless_trinket.h:362:44:   required from 'static void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::showRGBInternal(PixelController<RGB_ORDER>&) [with unsigned char DATA_PIN = 14; int T1 = 4; int T2 = 10; int T3 = 6; EOrder RGB_ORDER = (EOrder)10; int XTRA0 = 0; bool FLIP = false; int WAIT_TIME = 10]'
    C:\Users\haijo\OneDrive\Dokumente\Arduino\libraries\FastLED\src/platforms/avr/clockless_trinket.h:126:19:   required from 'void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::showPixels(PixelController<RGB_ORDER>&) [with unsigned char DATA_PIN = 14; int T1 = 4; int T2 = 10; int T3 = 6; EOrder RGB_ORDER = (EOrder)10; int XTRA0 = 0; bool FLIP = false; int WAIT_TIME = 10]'
    C:\Users\haijo\OneDrive\Dokumente\Arduino\sketch_aug28a\sketch_aug28a.ino:11:1:   required from here
    C:\Users\haijo\OneDrive\Dokumente\Arduino\libraries\FastLED\src/platforms/avr/fastpin_avr.h:39:84: error: incomplete type '__gen_struct_VPORTD_OUT' used in nested name specifier
      inline static port_ptr_t port() __attribute__ ((always_inline)) { return &_PORT::r(); }
                                                                                ~~~~~~~~^~
    
    exit status 1
    
    Compilation error: exit status 1
    
I like this library very much and want to use it again, please help. Thank you!

hugoviolin, 2024-06-28:
The library is not working if ESP32 version 3.0.2.

emintz, 2024-07-06:
I see. Thank you so much. Perhaps I can help fix this?

emintz, 2024-07-06:
Confirmed. It compiles perfectly with ESP32 library version 2.0.17. Thank you so much.

zackees, 2024-08-31:
This is not an ESP, how can the ESP library fix anything?

zackees, 2024-08-31:
I'm sorry I tried to make changes but failed to make this board compile. I don't know what's wrong with it.

zackees, 2024-09-04:
So one solution here is to just move pins.

The code for the AVR hardware pins here is knarly. It's templates on macros.

zackees, 2024-09-04:
Here's what I found so far.

This is from [src\platforms\avr\fastpin_avr.h](src\platforms\avr\fastpin_avr.h}

There's this spi platform definition for the pins.

As far as I can tell _FL_DEFPIN(...) stands for "FastLED Define Pin". It looks like it's enumerating pins 1...22 to be PORT(A/B/C/D/E/F) + channel (?).

```C++
#elif defined(ARDUINO_AVR_NANO_EVERY)

#define MAX_PIN 22
_FL_DEFPIN(0, 5, C); _FL_DEFPIN(1, 4, C); _FL_DEFPIN(2, 0, A); _FL_DEFPIN(3, 5, F);
_FL_DEFPIN(4, 6, C); _FL_DEFPIN(5, 2, B); _FL_DEFPIN(6, 4, F); _FL_DEFPIN(7, 1, A);
_FL_DEFPIN(8, 3, E);
_FL_DEFPIN(9, 0, B);
_FL_DEFPIN(10, 1, B); _FL_DEFPIN(11, 0, E);
_FL_DEFPIN(12, 1, E); _FL_DEFPIN(13, 2, E);
_FL_DEFPIN(14, 3, D);
_FL_DEFPIN(15, 2, D);
_FL_DEFPIN(16, 1, D); _FL_DEFPIN(17, 0, D); _FL_DEFPIN(18, 2, A); _FL_DEFPIN(19, 3, A);
_FL_DEFPIN(20, 4, D); _FL_DEFPIN(21, 5, D); _FL_DEFPIN(22, 2, A);

// To confirm for the SPI interfaces
#define SPI_DATA 11
#define SPI_CLOCK 13
#define SPI_SELECT 8
#define AVR_HARDWARE_SPI 1
#define HAS_HARDWARE_PIN_SUPPORT 1
```

zackees, 2024-09-04:
Here is the repro case. Clone the FastLED repo, install ci/requirements.txt and run at the project root.

`pio ci --board nano_every --lib=ci --lib=src --keep-build-dir --build-dir=.build/nano_every examples/Apa102HD/*ino`

zackees, 2024-09-07:
More insight on this issue:

Ok i understand what's going on better.

The difficult thing here is that the static compile-time-validation pattern that's part of FastLEDs design uses a static type system so the compiler can only generate code for valid pins.

These pins are defined by macros like _FL_DEFPIN which attach valid templates clauses which are parameteized against template functions and template classes.

The end result is absolutely nasty compiler errors, like what you are seeing here.

I just went through whole painful exercise for the NRF52 board for the adafruit feather sense boards.

What makes this more painful is that these invalid pin definitions are hitting our own compiler suite as well.

The way I solved this on the NRF52 is that I went into the board pin definitions and added a macro and fake pin class defining that subclasses to fake thing that printed a warning and paused for 2 second during first run with

zackees, 2024-09-20:
Fixed in 3.7.8

closed 2024-09-20 by zackees