FastLED/FastLED #1640: FastLED does not work for ESP32-Wroom-DA

tjodork1960, 2024-07-06:
The following program works flawlessly on esp8266 nodemcu
but will not work on esp32 wroom da
It seemed at times i was able to blankLEDs but when I went to turn on an individual....it will not work..
I had to move to esp32 for my project because the 8266 didnt have enough io pins.
PLEASE help this to work...its critical to my project.
Thanks
note I did try a few earlier releases to no avail

`#include <FastLED.h>

// How many leds are in the strip?
#define NUM_LEDS 14
#define DATA_PIN 2

// This is an array of leds.  One item for each led in your strip.
CRGB leds[NUM_LEDS];

// This function sets up the ledsand tells the controller about them
void setup() {
  //Setup LEDs
  FastLED.addLeds<WS2811, DATA_PIN, RGB>(leds, NUM_LEDS);
  blankLEDs();
  delay(2000);
  LEDon(5);
}

// This function runs over and over, and is where you do the magic to light
// your leds.
void loop() {
}

void LEDon(int led) {
  leds[led] = CRGB::Blue;
  FastLED.setBrightness(100);
  FastLED.show();
}

void blankLEDs() {
  for (int i = 0; i < NUM_LEDS; i++) {
    leds[i] = CRGB::Black;
    FastLED.show();
    FastLED.setBrightness(100);
  }
  delay(100);
}
`

zackees, 2024-08-08:
How does this not work?

Please post the output or a screenshot on what's not working.

zackees, 2024-08-31:
Following up on this, can you create a platformio.ini package? Doing this myself will be a lot of work. We have a suite of tests and I'd be a little surprised if this isn't fixed right now.

zackees, 2024-12-15:
Okay I still can't reproduce this bug with a ESP32Dev/Wroom/S3/C3/C6.

This is the first attempt I'm making to fix it while "flying blind".

The code is located on the `fix-rmt` branch, which disables the RMT recycling.

https://github.com/FastLED/FastLED/tree/fix-rmt

You can download this here as a zip file and load it up in your arduino ide.

https://github.com/FastLED/FastLED/archive/refs/heads/fix-rmt.zip

You can also use platformio and set the FastLED dependency to the branch like this:

```
lib_deps = FastLED@https://github.com/FastLED/FastLED.git#fix-rmt
```

Here's an example:

```
[env:dev]
platform = espressif32
framework = arduino
board = esp32dev

monitor_speed = 115200
upload_protocol = esptool

monitor_filters = 
	default
	esp32_exception_decoder  ; Decode exceptions so that they are human readable.

lib_deps = FastLED@https://github.com/FastLED/FastLED.git#fix-rmt
```

zackees, 2024-12-19:
Try the newest release that just went out.

zackees, 2024-12-30:
Fixed in 3.9.8. Feel free to re-open if this is not the case.