FastLED/FastLED #1642: when FASTLED_RMT_BUILTIN_DRIVER is enabled ESP32RMTController::convertByte has a nullptr error

briandilley, 2024-07-11:
The code in question is:

```cpp
void ESP32RMTController::convertByte(uint32_t byteval)
{
    // -- Write one byte's worth of RMT pulses to the big buffer
    byteval <<= 24;
    for (FASTLED_REGISTER uint32_t j = 0; j < 8; j++) {
        mBuffer[mCurPulse] = (byteval & 0x80000000L) ? mOne : mZero;
        byteval <<= 1;
        mCurPulse++;
    }
}
```

I have 8 data pins that i'm driving LEDs with, and mBuffer is consistently null with the third pin every time (GPIO32).

If i disable FASTLED_RMT_BUILTIN_DRIVER everything works... but I'd like to try the FASTLED_RMT_BUILTIN_DRIVER given that without it the system can be unreliable with WiFi connections active.

briandilley, 2024-07-25:
I'm using platformIO  (in CLion) with ESP-IDF - what's the best/easiest way to attach a debugger?

zackees, 2024-08-05:
Yeah we are going to need repro steps. Please upload demo code to github and link it here.

zackees, 2024-10-01:
RMT5 is now live in FastLED 3.8 beta. Just point your FastLED code to the master branch. Hopefully, this fixes your issue.

briandilley, 2024-10-01:
What is RMT5? I know what the RMT peripheral is - just curious on the 5.

zackees, 2024-10-02:
ESP32 features a major change in their DMA like controller called the RMT controller. It fixes a lot of problems.

It came out in esp idf 5.1 a few years ago and arduino rolled it into a core update a few months ago.

zackees, 2024-11-01:
Since 3.9.0 we are now using the RMT driver. This was happening in the one-shot/built-in rmt driver which we no longer use by default. Therefore, closing this bug.

closed 2024-11-01 by zackees