FastLED/FastLED #1646: I need a way to store LED order in preferences and load it at runtime

scottchiefbaker, 2024-07-23:
My application will interface with different types of LEDs (RGB, BGR, GRB, etc) so I'm planning on storing the configuration for the order in [preferences](https://docs.espressif.com/projects/arduino-esp32/en/latest/tutorials/preferences.html). 

The problem is that the only way I can see to configure the color order is at _compile time_ via the `addLeds<>()` template, which won't work if I need to read the variable from preferences at _runtime_. Is there a way to change the LED color order at runtime so that I can allow my user to configure the LEDs for their particular LED strand?

smurphyUSI, 2024-07-28:
The idea of the _compile time_ order is that you *should* know the order of your LEDs at the time you wire them to the controller.

Other people have done what you are asking, but often as a form of visual effect instead of preference.

You will need to remap all write operations to your led array, either inline, or just before writing.

Your should have a simple array definition like this:
CRGB leds[NUM_LEDS];

In reality, you can swap or reorder anything in that array at any time before you call FastLED.show();

@todbot has a good example of remapping the entire array before writing:
https://gist.github.com/todbot/eafe8467099bdb641a744b539af915d2

zackees, 2024-08-04:
This feature can be easily done on the client side with a few lines of code.

You may not agree with decisions made in the past on compatibility but last time I tried, FastLED still compiles and correctly runs on the original Arduino UNO, which is still VERY popular.

closed 2024-08-04 by zackees

GPSBabelDeveloper, 2024-08-04:
Ok, since you've now heard an official answer, I'll steer you more strongly to a project that works sensibly on the cpu you asked about. 

https://components.espressif.com/components/espressif/led_strip

scottchiefbaker, 2024-08-04:
@zackees as the OP can you provide some direction on how I can do it on the client side?

zackees, 2024-08-05:
Sure

```
for (int i = 0; i < N_LEDS; ++i) {
  // swaps RGB -> BRG
  CRGB c = leds[i];
  leds[i].r = c.b;
  leds[i].g = c.r;
  leds[i].b = c.g;
}
FastLED.show();
// optional - unswap the colors after draw if you need to keep your frame buffer.
```