FastLED/FastLED #1649: compile error on Arduino nano esp32

TxJakob, 2024-07-26:
Hi, I am currently trying to make an LED control with FastLED with my Arduino nano esp32. On my old nano every this script worked, but unfortunately it is broken and I wanted to test it on my new esp32 as well. I only get this error and have no idea what to do.

The code I use:
```

#include "FastLED.h"

#define NUM_LEDS 35
#define LED_DATA_PIN D3
#define NUM_BYTES (NUM_LEDS*3) // 3 colors  

#define BRIGHTNESS 100
#define UPDATES_PER_SECOND 100

#define TIMEOUT 3000

#define MODE_ANIMATION 0
#define MODE_AMBILIGHT 1
uint8_t mode = MODE_ANIMATION;

byte MESSAGE_PREAMBLE[] = { 0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09 };
uint8_t PREAMBLE_LENGTH = 10;
uint8_t current_preamble_position = 0;

unsigned long last_serial_available = -1L;

uint8_t led_counter = 0;
uint8_t byte_counter = 0;

CRGB leds[NUM_LEDS];
byte buffer[NUM_BYTES];

// Filler animation attributes
CRGBPalette16 currentPalette = RainbowColors_p;
TBlendType    currentBlending = LINEARBLEND;
uint8_t startIndex = 0;

void setup()
{
  Serial.begin(115200);
  FastLED.clear(true);
  FastLED.addLeds<WS2811, LED_DATA_PIN, GRB>(leds, NUM_LEDS);
  FastLED.setBrightness(BRIGHTNESS);
}

void loop()
{
	switch (mode) {
		case MODE_ANIMATION: 
			fillLEDsFromPaletteColors();
			break;
		
		case MODE_AMBILIGHT:
			processIncomingData();
			break;
	}
	
}

void processIncomingData()
{
	if (waitForPreamble(TIMEOUT))
	{
		Serial.readBytes(buffer, NUM_BYTES);

		/* DEBUG
		for (int i = 0; i < NUM_BYTES; i++)
		{
			Serial.write((char)buffer[i]);
		}
		*/

		while (byte_counter < NUM_BYTES)
		{
			byte blue = buffer[byte_counter++];
			byte green = buffer[byte_counter++];
			byte red = buffer[byte_counter++];

			leds[led_counter++] = CRGB(red, green, blue);
		}

		FastLED.show();

		byte_counter = 0;
		led_counter = 0;
	}
	else
	{
		mode = MODE_ANIMATION;
	}
}

bool waitForPreamble(int timeout)
{
	last_serial_available = millis();
	while (current_preamble_position < PREAMBLE_LENGTH)
	{
		if (Serial.available() > 0)
		{
			last_serial_available = millis();

			if (Serial.read() == MESSAGE_PREAMBLE[current_preamble_position])
			{
				current_preamble_position++;
			}
			else
			{
				current_preamble_position = 0;
			}
		}

		if (millis() - last_serial_available > timeout)
		{
			return false;
		}
	}
	current_preamble_position = 0;
	return true;
}

void fillLEDsFromPaletteColors()
{
	startIndex++; // speed

	uint8_t colorIndex = startIndex;
	for( int i = 0; i < NUM_LEDS; i++) {
        leds[i] = ColorFromPalette(currentPalette, colorIndex, BRIGHTNESS, currentBlending);
		colorIndex += 3;
    }

	FastLED.delay(1000 / UPDATES_PER_SECOND);

	if (Serial.available() > 0)
	{
		mode = MODE_AMBILIGHT;
	}
}

```

The error when I'm trying to compile the code:

In file included from C:\Users\Jakob\Documents\Arduino\libraries\FastLED\src/FastLED.h:79,
                 from C:\Users\Jakob\Desktop\Bambilight-master\Bambilight-master\Arduino\bambilight\bambilight.ino:1:
C:\Users\Jakob\Documents\Arduino\libraries\FastLED\src/fastspi.h:157:23: note: '#pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output'
  157 | #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Jakob\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp: In member function 'void ESP32RMTController::tx_start()':
C:\Users\Jakob\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:326:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'mem_rd_rst_n'
  326 |     RMT.chnconf0[mRMT_channel].mem_rd_rst_n = 1;
      |                                ^~~~~~~~~~~~
C:\Users\Jakob\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:327:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'mem_rd_rst_n'
  327 |     RMT.chnconf0[mRMT_channel].mem_rd_rst_n = 0;
      |                                ^~~~~~~~~~~~
C:\Users\Jakob\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:328:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'apb_mem_rst_n'
  328 |     RMT.chnconf0[mRMT_channel].apb_mem_rst_n = 1;
      |                                ^~~~~~~~~~~~~
C:\Users\Jakob\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:329:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'apb_mem_rst_n'
  329 |     RMT.chnconf0[mRMT_channel].apb_mem_rst_n = 0;
      |                                ^~~~~~~~~~~~~
C:\Users\Jakob\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:335:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'conf_update_n'
  335 |     RMT.chnconf0[mRMT_channel].conf_update_n = 1;
      |                                ^~~~~~~~~~~~~
C:\Users\Jakob\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:336:32: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'tx_start_n'
  336 |     RMT.chnconf0[mRMT_channel].tx_start_n = 1;
      |                                ^~~~~~~~~~
C:\Users\Jakob\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp: In static member function 'static void ESP32RMTController::doneOnChannel(rmt_channel_t, void*)':
C:\Users\Jakob\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:389:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'tx_stop_n'
  389 |     RMT.chnconf0[channel].tx_stop_n = 1;
      |                           ^~~~~~~~~
C:\Users\Jakob\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:390:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'conf_update_n'
  390 |     RMT.chnconf0[channel].conf_update_n = 1;
      |                           ^~~~~~~~~~~~~
C:\Users\Jakob\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:392:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'mem_rd_rst_n'
  392 |     RMT.chnconf0[channel].mem_rd_rst_n = 1;
      |                           ^~~~~~~~~~~~
C:\Users\Jakob\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:393:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'mem_rd_rst_n'
  393 |     RMT.chnconf0[channel].mem_rd_rst_n = 0;
      |                           ^~~~~~~~~~~~
C:\Users\Jakob\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:394:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'apb_mem_rst_n'
  394 |     RMT.chnconf0[channel].apb_mem_rst_n = 1;
      |                           ^~~~~~~~~~~~~
C:\Users\Jakob\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:395:27: error: 'volatile union rmt_chnconf0_reg_t' has no member named 'apb_mem_rst_n'
  395 |     RMT.chnconf0[channel].apb_mem_rst_n = 0;
      |                           ^~~~~~~~~~~~~
In file included from C:\Users\Jakob\Documents\Arduino\libraries\FastLED\src\FastLED.h:79,
                 from C:\Users\Jakob\Documents\Arduino\libraries\FastLED\src\five_bit_hd_gamma.cpp:1:
C:\Users\Jakob\Documents\Arduino\libraries\FastLED\src\fastspi.h:157:23: note: '#pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output'
  157 | #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

softhack007, 2024-07-29:
I Think the "nano esp32" is actually a ESP32-S3, correct?

Which version of arduino-esp32 are you using?
I think that arduino-esp32 v 3.0.x is not supported by fastled yet -> did you try with an older board package like arduino-esp32 v 2.0.14 or v2.0.17?

egnor, 2024-08-05:
(Yes, this looks like a dup of the arduino-esp32 v3.x / ESP-IDF v5.x RMT issue, see https://github.com/FastLED/FastLED/issues/1628 and https://github.com/FastLED/FastLED/issues/1631 for more discussion & fixes)

zackees, 2024-08-08:
Compiler error fixed in FastLED 3.7.1 bugfix release.

closed 2024-08-08 by zackees