FastLED/FastLED #1653: FastLED does not compile for ESP32S3 under Arduino IDE 2.3.2

DasElvetritsche, 2024-08-09:
If i compile the sketch without line 1 "#define FASTLED_ALL_PINS_HARDWARE_SPI" i get the message rearding bit banging and all is OK. If i include it i get the following errors:

##################################################################
In file included from c:\TMP\Arduino\libraries\FastLED\src/platforms/esp/32/fastled_esp32.h:6,
                 from c:\TMP\Arduino\libraries\FastLED\src/platforms.h:42,
                 from c:\TMP\Arduino\libraries\FastLED\src/FastLED.h:66,
                 from C:\TMP\Arduino\LED-colors\LED-colors.ino:2:
c:\TMP\Arduino\libraries\FastLED\src/platforms/esp/32/fastspi_esp32.h:2:17: note: #pragma message: ESP32 Hardware SPI support added
 #pragma message "ESP32 Hardware SPI support added"
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from c:\TMP\Arduino\libraries\FastLED\src/platforms/esp/32/fastled_esp32.h:6,
                 from c:\TMP\Arduino\libraries\FastLED\src/platforms.h:42,
                 from c:\TMP\Arduino\libraries\FastLED\src/FastLED.h:66,
                 from C:\TMP\Arduino\LED-colors\LED-colors.ino:2:
c:\TMP\Arduino\libraries\FastLED\src/platforms/esp/32/fastspi_esp32.h:54:18: note: #pragma message: Targeting ESP32S3, which has better SPI support. Configuring for flexible pin assignment.
  #pragma message "Targeting ESP32S3, which has better SPI support. Configuring for flexible pin assignment."
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\TMP\Arduino\libraries\FastLED\src/platforms/esp/32/fastspi_esp32.h: In instantiation of 'static void ESP32SPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::writeWord(uint16_t) [with unsigned char DATA_PIN = 40; unsigned char CLOCK_PIN = 39; unsigned int SPI_SPEED = 10000000; uint16_t = short unsigned int]':
c:\TMP\Arduino\libraries\FastLED\src/chipsets.h:447:25:   required from 'void P9813Controller<DATA_PIN, CLOCK_PIN, RGB_ORDER, SPI_SPEED>::writeBoundary() [with unsigned char DATA_PIN = 40; unsigned char CLOCK_PIN = 39; EOrder RGB_ORDER = (EOrder)136; unsigned int SPI_SPEED = 10000000]'
c:\TMP\Arduino\libraries\FastLED\src/chipsets.h:466:3:   required from 'void P9813Controller<DATA_PIN, CLOCK_PIN, RGB_ORDER, SPI_SPEED>::showPixels(PixelController<RGB_ORDER>&) [with unsigned char DATA_PIN = 40; unsigned char CLOCK_PIN = 39; EOrder RGB_ORDER = (EOrder)136; unsigned int SPI_SPEED = 10000000]'
c:\TMP\Arduino\libraries\FastLED\src/chipsets.h:463:15:   required from here
c:\TMP\Arduino\libraries\FastLED\src/platforms/esp/32/fastspi_esp32.h:118:69: error: cannot call member function 'void ESP32SPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::writeByte(uint8_t) [with unsigned char DATA_PIN = 40; unsigned char CLOCK_PIN = 39; unsigned int SPI_SPEED = 10000000; uint8_t = unsigned char]' without object
  static void writeWord(uint16_t w) __attribute__((always_inline)) { writeByte(w>>8); writeByte(w&0xFF); }
                                                                     ^~~~~~~~~
c:\TMP\Arduino\libraries\FastLED\src/platforms/esp/32/fastspi_esp32.h:118:86: error: cannot call member function 'void ESP32SPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::writeByte(uint8_t) [with unsigned char DATA_PIN = 40; unsigned char CLOCK_PIN = 39; unsigned int SPI_SPEED = 10000000; uint8_t = unsigned char]' without object
  static void writeWord(uint16_t w) __attribute__((always_inline)) { writeByte(w>>8); writeByte(w&0xFF); }
                                                                                      ^~~~~~~~~
In file included from c:\TMP\Arduino\libraries\FastLED\src/FastLED.h:80,
                 from C:\TMP\Arduino\LED-colors\LED-colors.ino:2:
c:\TMP\Arduino\libraries\FastLED\src/chipsets.h: In instantiation of 'static void LPD8806Controller<DATA_PIN, CLOCK_PIN, RGB_ORDER, SPI_SPEED>::LPD8806_ADJUST::postBlock(int) [with unsigned char DATA_PIN = 40; unsigned char CLOCK_PIN = 39; EOrder RGB_ORDER = (EOrder)136; unsigned int SPI_SPEED = 12000000]':
c:\TMP\Arduino\libraries\FastLED\src/platforms/esp/32/fastspi_esp32.h:189:15:   required from 'void ESP32SPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::writePixels(PixelController<RGB_ORDER>) [with unsigned char FLAGS = 0; D = LPD8806Controller<40, 39, (EOrder)136, 12000000>::LPD8806_ADJUST; EOrder RGB_ORDER = (EOrder)136; unsigned char DATA_PIN = 40; unsigned char CLOCK_PIN = 39; unsigned int SPI_SPEED = 12000000]'
c:\TMP\Arduino\libraries\FastLED\src/chipsets.h:113:3:   required from 'void LPD8806Controller<DATA_PIN, CLOCK_PIN, RGB_ORDER, SPI_SPEED>::showPixels(PixelController<RGB_ORDER>&) [with unsigned char DATA_PIN = 40; unsigned char CLOCK_PIN = 39; EOrder RGB_ORDER = (EOrder)136; unsigned int SPI_SPEED = 12000000]'
c:\TMP\Arduino\libraries\FastLED\src/chipsets.h:112:15:   required from here
c:\TMP\Arduino\libraries\FastLED\src/chipsets.h:96:27: error: cannot call member function 'void ESP32SPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::writeBytesValueRaw(uint8_t, int) [with unsigned char DATA_PIN = 40; unsigned char CLOCK_PIN = 39; unsigned int SPI_SPEED = 12000000; uint8_t = unsigned char]' without object
    SPI::writeBytesValueRaw(0, ((len*3+63)>>6));
    ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
In file included from c:\TMP\Arduino\libraries\FastLED\src/platforms/esp/32/fastled_esp32.h:6,
                 from c:\TMP\Arduino\libraries\FastLED\src/platforms.h:42,
                 from c:\TMP\Arduino\libraries\FastLED\src/FastLED.h:66,
                 from C:\TMP\Arduino\LED-colors\LED-colors.ino:2:
c:\TMP\Arduino\libraries\FastLED\src/platforms/esp/32/fastspi_esp32.h: In instantiation of 'static void ESP32SPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::writeWord(uint16_t) [with unsigned char DATA_PIN = 40; unsigned char CLOCK_PIN = 39; unsigned int SPI_SPEED = 24000000; uint16_t = short unsigned int]':
c:\TMP\Arduino\libraries\FastLED\src/chipsets.h:231:3:   required from 'void APA102Controller<DATA_PIN, CLOCK_PIN, RGB_ORDER, SPI_SPEED, GAMMA_CORRECTION_MODE, START_FRAME, END_FRAME>::startBoundary() [with unsigned char DATA_PIN = 40; unsigned char CLOCK_PIN = 39; EOrder RGB_ORDER = (EOrder)136; unsigned int SPI_SPEED = 24000000; FiveBitGammaCorrectionMode GAMMA_CORRECTION_MODE = (FiveBitGammaCorrectionMode)1; unsigned int START_FRAME = 0; unsigned int END_FRAME = 0]'
c:\TMP\Arduino\libraries\FastLED\src/chipsets.h:322:3:   required from 'void APA102Controller<DATA_PIN, CLOCK_PIN, RGB_ORDER, SPI_SPEED, GAMMA_CORRECTION_MODE, START_FRAME, END_FRAME>::showPixelsDefault(PixelController<RGB_ORDER>&) [with unsigned char DATA_PIN = 40; unsigned char CLOCK_PIN = 39; EOrder RGB_ORDER = (EOrder)136; unsigned int SPI_SPEED = 24000000; FiveBitGammaCorrectionMode GAMMA_CORRECTION_MODE = (FiveBitGammaCorrectionMode)1; unsigned int START_FRAME = 0; unsigned int END_FRAME = 0]'
c:\TMP\Arduino\libraries\FastLED\src/chipsets.h:284:5:   required from 'void APA102Controller<DATA_PIN, CLOCK_PIN, RGB_ORDER, SPI_SPEED, GAMMA_CORRECTION_MODE, START_FRAME, END_FRAME>::showPixels(PixelController<RGB_ORDER>&) [with unsigned char DATA_PIN = 40; unsigned char CLOCK_PIN = 39; EOrder RGB_ORDER = (EOrder)136; unsigned int SPI_SPEED = 24000000; FiveBitGammaCorrectionMode GAMMA_CORRECTION_MODE = (FiveBitGammaCorrectionMode)1; unsigned int START_FRAME = 0; unsigned int END_FRAME = 0]'
c:\TMP\Arduino\libraries\FastLED\src/chipsets.h:281:15:   required from here
c:\TMP\Arduino\libraries\FastLED\src/platforms/esp/32/fastspi_esp32.h:118:69: error: cannot call member function 'void ESP32SPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::writeByte(uint8_t) [with unsigned char DATA_PIN = 40; unsigned char CLOCK_PIN = 39; unsigned int SPI_SPEED = 24000000; uint8_t = unsigned char]' without object
  static void writeWord(uint16_t w) __attribute__((always_inline)) { writeByte(w>>8); writeByte(w&0xFF); }
                                                                     ^~~~~~~~~
c:\TMP\Arduino\libraries\FastLED\src/platforms/esp/32/fastspi_esp32.h:118:86: error: cannot call member function 'void ESP32SPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::writeByte(uint8_t) [with unsigned char DATA_PIN = 40; unsigned char CLOCK_PIN = 39; unsigned int SPI_SPEED = 24000000; uint8_t = unsigned char]' without object
  static void writeWord(uint16_t w) __attribute__((always_inline)) { writeByte(w>>8); writeByte(w&0xFF); }
                                                                                      ^~~~~~~~~
[... cut ...]
  static uint8_t hue = 0;
  // Set the first led to the current hue
  leds[0] = CHSV(hue, 255, 255);
  // Show the leds
  FastLED.show();  
  // Increase the hue
  hue++;
  // Wait a little bit before the next loop
  delay(20); // This delay controls the speed at which hues change, lower value will cause faster hue rotation.
}
-------------------------------------------------------------------------------
The target hardware is a LilyGo T-Dongle-S3 -> ESP32S3 Dev Module

Anyone a idea what goes wrong in the library?

zackees, 2024-08-17:
I just repro'd your bug and added a test case for you.

Note to core-devs, you can repro this bug by cloning the repro then using this command:

`ci/ci-compile --boards esp32-s3-devkitc-1 --examples _bugs/1653_S3_Apa102_Hardware_Pins`

closed 2024-08-17 by zackees

DasElvetritsche, 2024-08-20:
Hi, i updated to the lates lib, but still not working. I modified som files in the old lib and then it is working.
[ModifiedLibraryFiles.txt](https://github.com/user-attachments/files/16667507/ModifiedLibraryFiles.txt)
[fastspi_esp32.h.txt](https://github.com/user-attachments/files/16667554/fastspi_esp32.h.txt)
[chipsets.h.txt](https://github.com/user-attachments/files/16667557/chipsets.h.txt)
[fastspi.h.txt](https://github.com/user-attachments/files/16667559/fastspi.h.txt)
i named them .txt to upload.
Hope it helps.

zackees, 2024-08-30:
Looking closer into this I noticed that this is an unsupported feature.

If you want me to consider this PR then I need you to do a pull request. There are development instructions in the readme but it boils down to

  1. Forking FastLED into your repo
  2. Cloning it locally
  3. Open up the repo in VSCode (make sure you have the platformio extension)
  4. Make your changes then push
  5. Open up the pull request url generated by the push command

DasElvetritsche, 2024-08-31:
The latest file i provided fix all issues for me but i dont know what it breaks for others. Thats why i didn’t dare to make a pull request. All changes i made are marked with BK. In the one file the preprocessor does not work properly with the #ifdefine. Only works for the SPI mode in the first if, later it does not work in the else. If you still want i will make it as proposed with the latest lib version. I will then remove my debug outputs. With my latest mods it even works with 2nd SPI used in TFT_eSPI lib and latest arduino IDF and espressif bord def.Bernd

If you want me to consider this change then I need you to do a pull request. There are development instructions in the readme but it boils down to

Cloning it locally

zackees, 2024-08-31:
Please make a new PR. I will take care of the build breakages because I've implemented a fleet of builders that will test it on every platform we support.

Can I count on you to implement this?

DasElvetritsche, 2024-08-31:
Next weeks i will do!Bernd