FastLED/FastLED #1654: Effect in a loop

saddys, 2024-08-10:
All various effect are in a loop with a for i = 0, so they are always in movement. I put a trigger that change type of effect, but when press the trigger, the new effect dont start from led 0. I can change the effect but cant force to restart new effect from led 0, it started in the position of the loop

There is a way to force the new effect to start from led 0 when I push a trigger?

Thanks

marmilicious, 2024-08-16:
Hello @saddys  We use github "issues" here mostly just for bugs or improvements in the library itself.  For all general discussions, help, and how-to, please use the FastLED Reddit group here: https://www.reddit.com/r/FastLED

Also please include a link to your code on pastebin.com or gist.github.com

saddys, 2024-08-17:
Ok, sorry, Thanks!

closed 2024-08-18 by zackees