FastLED/FastLED #1655: FAST_LED Issue (Failure to run) with ESP32_S3 compilation and running.. (using Fast LED 3.7.1)

bsohn73, 2024-08-12:
I made mention on Reddit about still having an issue after the 3.7.1 Fast_LED update that DID actually fix the compilation through bypassed warnings as it actually compiled and uploaded to the Waveshare ESP32s3 LED Matrix Development Board that I have.  However when it does try to run the code I get a Debug Output over serial indicating a program failure and reboot.. 

So here is all of the information regarding this issue:

Attached is the Compile Output TXT.

[ESP32s3matrix Compile output.txt](https://github.com/user-attachments/files/16577381/ESP32s3matrix.Compile.output.txt)

Debug Output.

[ESP32s3matrix Debug Output.txt](https://github.com/user-attachments/files/16577386/ESP32s3matrix.Debug.Output.txt)

and the .ino (in txt form) used (note the commented out neopixelWrite(PIN_NEOPIXEL, red, green, blue)) can be commented back in while commenting out all of the FAST_LED items and the code works as intended.

[blink ino.txt](https://github.com/user-attachments/files/16577412/blink.ino.txt)

The Issues with running the program occur with just the #include of the Fast_LED library even if there is no actual fast LED code in the ino.

zackees, 2024-08-12:
Please inline your log files.

Also, if you want to debug at all then you can use platformio and it will use intellisense to jump right to the fastled code which you can edit for debugging purposes. Here is a minimal boilerplate to convert your ino. You can probably just stick everything you want in the ino file as well, but I'm not sure the intellisense will work for it.

https://github.com/zackees/fastled_bug_s3_idf5x

bsohn73, 2024-08-12:
I'm not experienced enough to probably figure out exactly what is going on here but here are the files inline.

Compiler Output
```
FQBN: esp32:esp32:esp32s3:USBMode=default,CDCOnBoot=cdc
Using board 'esp32s3' from platform in folder: /Users/bsohn/Library/Arduino15/packages/esp32/hardware/esp32/3.0.3
Using core 'esp32' from platform in folder: /Users/bsohn/Library/Arduino15/packages/esp32/hardware/esp32/3.0.3

/usr/bin/env bash -c "[ ! -f \"/Users/bsohn/Library/Arduino15/packages/esp32/hardware/esp32/3.0.3/libraries/ESP32/examples/GPIO/BlinkRGB\"/partitions.csv ] || cp -f \"/Users/bsohn/Library/Arduino15/packages/esp32/hardware/esp32/3.0.3/libraries/ESP32/examples/GPIO/BlinkRGB\"/partitions.csv \"/private/var/folders/v3/ybxp1f_x0s31nzhdgbckd5k80000gn/T/arduino/sketches/4C6D8DF9738D2C73F1C51C62C29B5D9D\"/partitions.csv"
/usr/bin/env bash -c "[ -f \"/private/var/folders/v3/ybxp1f_x0s31nzhdgbckd5k80000gn/T/arduino/sketches/4C6D8DF9738D2C73F1C51C62C29B5D9D\"/partitions.csv ] || [ ! -f \"/Users/bsohn/Library/Arduino15/packages/esp32/hardware/esp32/3.0.3/variants/esp32s3\"/partitions.csv ] || cp \"/Users/bsohn/Library/Arduino15/packages/esp32/hardware/esp32/3.0.3/variants/esp32s3\"/partitions.csv \"/private/var/folders/v3/ybxp1f_x0s31nzhdgbckd5k80000gn/T/arduino/sketches/4C6D8DF9738D2C73F1C51C62C29B5D9D\"/partitions.csv"
/usr/bin/env bash -c "[ -f \"/private/var/folders/v3/ybxp1f_x0s31nzhdgbckd5k80000gn/T/arduino/sketches/4C6D8DF9738D2C73F1C51C62C29B5D9D\"/partitions.csv ] || cp \"/Users/bsohn/Library/Arduino15/packages/esp32/hardware/esp32/3.0.3\"/tools/partitions/default.csv \"/private/var/folders/v3/ybxp1f_x0s31nzhdgbckd5k80000gn/T/arduino/sketches/4C6D8DF9738D2C73F1C51C62C29B5D9D\"/partitions.csv"
/usr/bin/env bash -c "[ -f \"/Users/bsohn/Library/Arduino15/packages/esp32/hardware/esp32/3.0.3/libraries/ESP32/examples/GPIO/BlinkRGB\"/bootloader.bin ] && cp -f \"/Users/bsohn/Library/Arduino15/packages/esp32/hardware/esp32/3.0.3/libraries/ESP32/examples/GPIO/BlinkRGB\"/bootloader.bin \"/private/var/folders/v3/ybxp1f_x0s31nzhdgbckd5k80000gn/T/arduino/sketches/4C6D8DF9738D2C73F1C51C62C29B5D9D\"/BlinkRGB.ino.bootloader.bin || ( [ -f \"/Users/bsohn/Library/Arduino15/packages/esp32/hardware/esp32/3.0.3/variants/esp32s3\"/bootloader.bin ] && cp \"/Users/bsohn/Library/Arduino15/packages/esp32/hardware/esp32/3.0.3/variants/esp32s3\"/bootloader.bin \"/private/var/folders/v3/ybxp1f_x0s31nzhdgbckd5k80000gn/T/arduino/sketches/4C6D8DF9738D2C73F1C51C62C29B5D9D\"/BlinkRGB.ino.bootloader.bin || \"/Users/bsohn/Library/Arduino15/packages/esp32/tools/esptool_py/4.6\"/esptool --chip esp32s3 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o \"/private/var/folders/v3/ybxp1f_x0s31nzhdgbckd5k80000gn/T/arduino/sketches/4C6D8DF9738D2C73F1C51C62C29B5D9D\"/BlinkRGB.ino.bootloader.bin \"/Users/bsohn/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-dc859c1e67/esp32s3\"/bin/bootloader_qio_80m.elf )"
esptool.py v4.6
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
/usr/bin/env bash -c "[ ! -f \"/Users/bsohn/Library/Arduino15/packages/esp32/hardware/esp32/3.0.3/libraries/ESP32/examples/GPIO/BlinkRGB\"/build_opt.h ] || cp -f \"/Users/bsohn/Library/Arduino15/packages/esp32/hardware/esp32/3.0.3/libraries/ESP32/examples/GPIO/BlinkRGB\"/build_opt.h \"/private/var/folders/v3/ybxp1f_x0s31nzhdgbckd5k80000gn/T/arduino/sketches/4C6D8DF9738D2C73F1C51C62C29B5D9D\"/build_opt.h"
/usr/bin/env bash -c "[ -f \"/private/var/folders/v3/ybxp1f_x0s31nzhdgbckd5k80000gn/T/arduino/sketches/4C6D8DF9738D2C73F1C51C62C29B5D9D\"/build_opt.h ] || : > \"/private/var/folders/v3/ybxp1f_x0s31nzhdgbckd5k80000gn/T/arduino/sketches/4C6D8DF9738D2C73F1C51C62C29B5D9D\"/build_opt.h"
/usr/bin/env bash -c ": > '/private/var/folders/v3/ybxp1f_x0s31nzhdgbckd5k80000gn/T/arduino/sketches/4C6D8DF9738D2C73F1C51C62C29B5D9D/file_opts'"
Detecting libraries used...
/Users/bsohn/Library/Arduino15/packages/esp32/tools/esp-xs3/2302/bin/xtensa-esp32s3-elf-g++ -c @/Users/bsohn/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-dc859c1e67/esp32s3/flags/cpp_flags -w -Os -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32S3_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32S3_DEV" -DARDUINO_VARIANT="esp32s3" -DARDUINO_PARTITION_default -DARDUINO_HOST_OS="macosx" -DARDUINO_FQBN="esp32:esp32:esp32s3:UploadSpeed=921600,USBMode=default,CDCOnBoot=cdc,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,CPUFreq=240,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=none,PSRAM=disabled,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_MODE=0 -DARDUINO_USB_CDC_ON_BOOT=1 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 @/Users/bsohn/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-dc859c1e67/esp32s3/flags/defines -I/Users/bsohn/Library/Arduino15/packages/esp32/hardware/esp32/3.0.3/libraries/ESP32/examples/GPIO/BlinkRGB -iprefix /Users/bsohn/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-dc859c1e67/esp32s3/include/ @/Users/bsohn/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-dc859c1e67/esp32s3/flags/includes -I/Users/bsohn/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-dc859c1e67/esp32s3/qio_qspi/include -I/Users/bsohn/Library/Arduino15/packa [... cut ...]
      case 0: { red = 0; green = 0; blue = 0; break;} 
      case 1: { red = 1; green = 0; blue = 0; break;} 
      case 2: { red = 0; green = 1; blue = 0; break;} 
      case 3: { red = 0; green = 0; blue = 1; break;} 
    }
    Serial.println("Color Chosen");

    for (uint8_t iFLAGled = 0; iFLAGled < LEDCOUNT; iFLAGled++ )
    {
    leds[iFLAGled].setRGB (red, green, blue);
//      neopixelWrite(PIN_NEOPIXEL, red, green, blue);
    }
  
    Serial.println("Color Displayed");

  FastLED.show();
  delay (1000);

 }
}

```

zackees, 2024-08-13:
Thanks for this, this is great.

Okay I know what the issue is: the legacy RMT driver is NOT compatible with the new RMT driver. Hence this crash.

This is happening because the new Arduino 2.3.1 IDE now forces the idf5.1 toolchain and is causing breakages across the Arduino ESP32 maker-space.

The solution being presented by the community includes downgrading their arduino esp idf package back to a previous version. You can search around for "RMT" and "FastLED" to see the fix.

If you'd like a more stable work around that doesn't involve modifying your global arduino environment then I suggest that you try out our new platform io / arduino dual repo.

https://github.com/FastLED/PlatformIO-Starter

Download and unzip it, and paste in your code into the ino or main.cpp file (you can use either).

This is what your platformio.ini file would look like:

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32]
platform = espressif32
framework = arduino
board = um_tinys3 ; you might want to change this to your board type

monitor_speed = 115200
upload_speed = 500000

upload_protocol = esptool

monitor_filters = 
	default
	esp32_exception_decoder

lib_deps =
    FastLED=https://github.com/FastLED/FastLED
```

This is what your platformIO

bsohn73, 2024-08-13:
Thank you, I will look into that.. I literally have had only a few days messing with the ESP32, I normally work with the ProMicros, so I am not even familiar with older ESP32 issues or things that have to be done.. This just seems to have coincided with things that you are just now dealing with regarding the library.

bsohn73, 2024-08-14:
Just wanted to make a comment about a work around that worked for me here.. All I did was to reinstall the ESP Arduino ExpressIF Board Package version 2.0.17 instead of the latest and everything is currently seeming to work at least with my Version of the ESP32s3 development board..

zackees, 2024-08-17:
Verified working in latest release.

closed 2024-08-17 by zackees