FastLED/FastLED #1656: FastLED 3.7.1 runtime error: CONFLICT! driver_ng is not allowed to be used with the legacy driver

DKasle, 2024-08-14:
I am seeing a runtime error while running some test code with Arduino IDE v 2.3.2, esp32 v 3.0.3, FastLED v 3.7.1 running on a windows 11 PC with a UM Tiny S3 microcontroller. The code compiles and uploads successfully, but I get this runtime error:

**_E (88) rmt(legacy): CONFLICT! driver_ng is not allowed to be used with the legacy driver

abort() was called at PC 0x420075a7 on core 0_**

A post in a similar thread on the Arduino forum 
https://forum.arduino.cc/t/conflict-driver-ng-is-not-allowed-to-be-used-with-the-legacy-driver/1279147/9
mentioned that this error was due to inclusion of the FastLED library.

Here is the test code that I used:

**#include <FastLED.h>

#include <stdio.h>
#include <stdlib.h>
#include "Arduino.h"
#include "math.h"
#include <FreeRTOS.h>
#include <SPI.h>         

/*********
  Rui Santos
  Complete project details at https://randomnerdtutorials.com  
*********/

TaskHandle_t Task1;
TaskHandle_t Task2;

// LED pins
const int led1 = 2;
const int led2 = 4;

void setup() {
  Serial.begin(115200); 
  pinMode(led1, OUTPUT);
  pinMode(led2, OUTPUT);

  //create a task that will be executed in the Task1code() function, with priority 1 and executed on core 0
  xTaskCreatePinnedToCore(
                    Task1code,   /* Task function. */
                    "Task1",     /* name of task. */
                    10000,       /* Stack size of task */
                    NULL,        /* parameter of the task */
                    1,           /* priority of the task */
                    &Task1,      /* Task handle to keep track of created task */
                    0);          /* pin task to core 0 */                  
  delay(500); 

  //create a task that will be executed in the Task2code() function, with priority 1 and executed on core 1
  xTaskCreatePinnedToCore(
                    Task2code,   /* Task function. */
                    "Task2",     /* name of task. */
                    10000,       /* Stack size of task */
                    NULL,        /* parameter of the task */
                    1,           /* priority of the task */
                    &Task2,      /* Task handle to keep track of created task */
                    1);          /* pin task to core 1 */
    delay(500); 
}

//Task1code: blinks an LED every 500 ms
void Task1code( void * pvParameters ){
  Serial.print("Task1 running on core ");
  Serial.println(xPortGetCoreID());

  for(;;){
    digitalWrite(led1, HIGH);
    delay(500);
    digitalWrite(led1, LOW);
    delay(500);
  } 
}

//Task2code: blinks an LED every 700 ms
void Task2code( void * pvParameters ){
  Serial.print("Task2 running on core ");
  Serial.println(xPortGetCoreID());

  for(;;){
    digitalWrite(led2, HIGH);
    delay(700);
    digitalWrite(led2, LOW);
    delay(700);
  }
}

void loop() {
  
}**

When I comment out the first line of the code above that includes the FastLED library, the test program runs successfully without the error, so I suspect the post on the Arduino forum was correct about this being related to the FastLED library.   It was not clear to me from the thread in the Arduino forum how the problem was solved,and I need to use the FastLED library for my current project.  Can anyone suggest a fix for this?

Thanks in advance!

samguyer, 2024-08-14:
I think this a problem for me to figure out. I just haven't had time to dig into the changes in the new ESP IDF.

edautomation, 2024-08-14:
During compilation, there are already some warnings:
`C:/Users/johnsmith/esp/v5.1.4/esp-idf/components/driver/deprecated/driver/rmt.h:18:2: warning: #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h" [-Wcpp]`

It seems there were some [breaking changes from IDF 4.x to 5.x](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/migration-guides/release-5.x/5.0/peripherals.html#rmt-driver) to the RMT peripheral driver.

F1p, 2024-08-15:
+1 on this issue, came to the same conculsion as above and same error message.

F1p, 2024-08-16:
> > I think this a problem for me to figure out. I just haven't had time to dig into the changes in the new ESP IDF.
> 
> I think i saw somewhere that someone has made a legacy wrapper around the RMT driver.

Would it be on the Adafruit NeoPixel?

https://github.com/adafruit/Adafruit_NeoPixel/pull/394 
https://github.com/adafruit/Adafruit_NeoPixel/pull/392

F1p, 2024-08-17:
Seems to be resolved for me in 3.7.3

michaelochs, 2024-08-20:
Getting this when uploading to an ESP32C6-DevKit-M1. The board just crash-loops right at boot. This is the stack trace I'm getting:
```
4087e140: 0x00000000 0x00000000 0x4087e16c 0x4080b42e 0x408014c2 0x42000000 0x420b0030 0x40810700
  #0  0x4080b42e in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:34
  #1  0x408014c2 in rv_utils_dbgr_is_attached at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/riscv/include/riscv/rv_utils.h:203
      (inlined by) esp_cpu_dbgr_is_attached at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/esp_cpu.h:506
      (inlined by) call_start_cpu0 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/cpu_start.c:307
  #2  0x40810700 in __global_pointer$ at ??:?

4087e160: 0x4087e16c 0x4081071c 0x4087e158 0x30303234 0x39663337 0x28010200 0x726f6261 0x20292874
  #0  0x4081071c in __global_pointer$ at ??:?

4087e180: 0x20736177 0x6c6c6163 0x61206465 0x43502074 0x34783020 0x37303032 0x20396633 0x63206e6f
4087e1a0: 0x2065726f 0x00000030 0x420b0000 0xfb10f0ec 0x20001090 0x420d2870 0x420d2894 0x420073fc
  #0  0x420b0000 in _flash_rodata_dummy_start at ??:?
  #1  0x420073fc in check_rmt_legacy_driver_conflict at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/deprecated/rmt_legacy.c:1386 (discriminator 3)

4087e1c0: 0x420ce000 0x420d2870 0x420d28a0 0x4200bf7e 0x0000004d 0x0000000c 0x0000004d 0x4200b60c
  #0  0x4200bf7e in do_global_ctors at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/startup.c:203
      (inlined by) start_cpu0_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/startup.c:486
  #1  0x4200b60c in esp_cache_err_int_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/soc/esp32c6/cache_err_int.c:39

4087e1e0: 0x0000004d 0x0000000c 0x0000004d 0x40801630 0x43000000 0x420d7de7 0x420a656f 0x42000000
  #0  0x40801630 in call_start_cpu0 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/cpu_start.c:706
  #1  0x420a656f in _flash_rodata_dummy_start at ??:?

4087e200: 0x43000000 0x420d7de7 0x420a656f 0x4086fa06 0x74667220 0x000b0000 0x00010020 0x00040020
  #0  0x420a656f in _flash_rodata_dummy_start at ??:?

4087e220: 0x420b0020 0x42000020 0x00027dc8 0x000a6550 0x00010000 0x00140000 0x00000000 0x40880000
  #0  0x420b0020 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_app_format/esp_app_desc.c:15
  #1  0x42000020 in CFastLED::CFastLED() at .pio/libdeps/esp32c6/FastLED/src/FastLED.cpp:36

4087e240: 0x00004964 0x4004b05c 0x0000ffff 0xffffffff 0x00000000 0x4087e3a0 0x00000000 0x4086fdb6
4087e260: 0x61202c78 0x00010000 0x00140000 0x00010000 0x200205e9 0x408014c2 0x000000ee 0x0000000d
  #0  0x408014c2 in rv_utils_dbgr_is_attached at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/riscv/include/riscv/rv_utils.h:203
      (inlined by) esp_cpu_dbgr_is_attached at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/esp_cpu.h:506
      (inlined by) call_start_cpu0 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/cpu_start.c:307

4087e280: 0x00ffff00 0x01000000 0x420b0020 0x00027dc8 0x40800000 0x00008228 0x42000020 0x000a6550
  #0  0x420b0020 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_app_format/esp_app_desc.c:15
  #1  0x40800000 in _vector_table at ??:?
  #2  0x42000020 in CFastLED::CFastLED() at .pio/libdeps/esp32c6/FastLED/src/FastLED.cpp:36

4087e2a0: 0x40808228 0x00005654 0x4080d880 0x000036e4 0x00000000 0x00000000 0x00000000 0x00000000
  #0  0x40808228 in vTaskRemoveFromUnorderedEventList at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4197
```

DKasle, 2024-08-23:
I just tried using the lates FastLED library: v 3.7.3 (instead of v 3.7.1).  The behavior of the Tiny S3 microcontroller seems unchanged, it still encounters a run time error.

samguyer, 2024-08-23:
@Rich968 I don't have an ESP32 board handy, but there might be a very simple fix. Would you be willing to try it out for me?

samguyer, 2024-08-23:
If you've got a moment, here's the potential fix: just change one of the include files in the RMT driver. Edit the file FastLED/src/platforms/esp/32/clockless_rmt_esp32.h and go to approximately line 130. Replace include "driver/rmt.h" with include "driver/rmt_tx.h"

DKasle, 2024-08-23:
Hi @samguyer ,

The change that you suggested:

Edit the file FastLED/src/platforms/esp/32/clockless_rmt_esp32.h and go to approximately line 130. Replace include "driver/rmt.h" with include "driver/rmt_tx.h"

produces the following compile error in my system:

c:\Users\dkasl\OneDrive\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/clockless_rmt_esp32.h:291:5: error: 'rmt_item32_t' does not name a type
  291 |     rmt_item32_t * mBuffer;

samguyer, 2024-08-23:
OK, so no quick fixes. I'll need to look at it more closely when I get home.
Thanks @DKasle !

Rich968, 2024-08-24:
Thanks to ChatGPT this lower level code works to control a single WS2812 LED irrespective whether CDC is enabled or disabled on boot.
#include <Arduino.h>
#include "esp32-hal-rgb-led.h"

// Define the LED pin
#define LED_PIN 21

// Function to write a color value to the WS2812 LED
void rgbLedWrite(uint8_t pin, uint8_t red_val, uint8_t green_val, uint8_t blue_val) {
#if SOC_RMT_SUPPORTED
  rmt_data_t led_data[24];
  
  // Color values in GRB order (default for WS2812B)
  uint8_t color[3] = {green_val, red_val, blue_val};

  // Initialize RMT
  if (!rmtInit(pin, RMT_TX_MODE, RMT_MEM_NUM_BLOCKS_1, 10000000)) {
    Serial.println("RMT initialization failed!");
    return;
  }

  // Prepare the LED data
  int i = 0;
  for (int col = 0; col < 3; col++) {
    for (int bit = 0; bit < 8; bit++) {
      if (color[col] & (1 << (7 - bit))) {
        led_data[i].level0 = 1;     // HIGH bit
        led_data[i].duration0 = 8;  // 0.8us
        led_data[i].level1 = 0;     // LOW bit
        led_data[i].duration1 = 4;  // 0.4us
      } else {
        led_data[i].level0 = 1;     // LOW bit
        led_data[i].duration0 = 4;  // 0.4us
        led_data[i].level1 = 0;     // HIGH bit
        led_data[i].duration1 = 8;  // 0.8us
      }
      i++;
    }
  }

  // Write the data to the LED
  rmtWrite(pin, led_data, RMT_SYMBOLS_OF(led_data), RMT_WAIT_FOR_EVER);
#else
  Serial.println("RMT is not supported on this target.");
#endif
}

void setup() {
  // Initialize serial communication
  Serial.begin(115200);

  // Set LED_PIN as an output
  pinMode(LED_PIN, OUTPUT);

  // Turn on WS2812 LED with blue color
  rgbLedWrite(LED_PIN, 0, 0, 255); // Blue color
}

void loop() {
  // No need to do anything in loop
}

moose4lord, 2024-08-24:
The UM TinyS3 is one of the ESP32 boards with a built-in WS2812 RGB LED, so it's pins_arduino.h file has the macro RGB_BUILTIN defined. This makes it incompatible with the legacy RMT driver included with IDF v5. 

The esp32-hal-gpio.c file contains this note:
```
// RGB_BUILTIN is defined in pins_arduino.h
// If RGB_BUILTIN is defined, it will be used as a pin number for the RGB LED
// If RGB_BUILTIN has a side effect that prevents using RMT Legacy driver in IDF 5.1
// Define ESP32_ARDUINO_NO_RGB_BUILTIN in build_opt.h or through CLI to disable RGB_BUILTIN
#ifdef ESP32_ARDUINO_NO_RGB_BUILTIN
#ifdef RGB_BUILTIN
#undef RGB_BUILTIN
#endif
#endif
```

Until the problems with the legacy RMT driver are resolved in IDF v5, you may want to stick with arduino_esp v2.0.17 (which uses IDF v4) for this board. Or you can hack the boards pins_arduino.h file to remove the RGB_BUILTIN macro.

Rich968, 2024-08-24:
Now written my own very simple 'library' class as FastLED is an overkill for my application.

zackees, 2024-08-24:
Heads up I just did a big behavior preserving refactor to seperate out the rtm driver stuff from the fastled cled wrapper and pushing a whole bunch of header defines into cpp defines to isolate and trim.

So some of the line numbers and file locations on our side have changed a bit.

Rich968, 2024-08-25:
I have based my WS2812 ESP32 class on esp32-hal-rgb-led.h and esp32-hal-rgb-led.cpp. All works fine. Presumably these files use the new RMT driver as no longer have any runtime driver conflict messages. (Apparently Espressif want new library code moving forward to use the new RMT driver code). Will new FastLED code use the new RMT driver code?

zackees, 2024-08-26:
Care to share?

Rich968, 2024-08-26:
Yes, feel free to modify or use etc. Here is my single header file 'library' class (thanks to ChatGPT!)  followed by a simple Arduino example cycling through various colours with default brightness.  The RMT class functions work with Espressif code in the Arduino board libraries for ESP32 directly and therefore are using the latest RMT driver. 

**First my WS2812.h library :**

#ifndef WS2812LED_H
#define WS2812LED_H

// Define the maximum brightness level (255 is full brightness)
#define MAX_BRIGHTNESS 255

class WS2812LED {
public:
  // Constructor: Initialize with the pin number
  WS2812LED(uint8_t pin) : _pin(pin) {
    pinMode(_pin, OUTPUT);
  }

  // Set the color and brightness of the LED
  void define(uint8_t red_val, uint8_t green_val, uint8_t blue_val, uint8_t brightness = MAX_BRIGHTNESS) {
    rgbLedWrite(_pin, red_val, green_val, blue_val, brightness);
  }

  // Turn off the LED
  void off() {
    define(0, 0, 0, 0);
  }

  // Predefined color methods for the rainbow - not quite a rainbow though!
  void defineRed(uint8_t brightness = MAX_BRIGHTNESS) {
    define(255, 0, 0, brightness);
  }

  void defineOrange(uint8_t brightness = MAX_BRIGHTNESS) {
    define(255, 140, 0, brightness);
  }

  void defineYellow(uint8_t brightness = MAX_BRIGHTNESS) {
    define(200, 200, 0, brightness);
  }

  void defineGreen(uint8_t brightness = MAX_BRIGHTNESS) {
    define(0, 255, 0, brightness);
  }

  void defineBlue(uint8_t brightness = MAX_BRIGHTNESS) {
    define(0, 0, 255, brightness);
  }

  void defineViolet(uint8_t brightness = MAX_BRIGHTNESS) {
    define(255, 0, 255, brightness);
  }
  
 void defineWhite(uint8_t brightness = MAX_BRIGHTNESS) {
    define(255, 255, 255, brightness);
  }

private:
  uint8_t _pin;

  // Function to adjust brightness of a color value
  uint8_t adjustBrightness(uint8_t color_val, uint8_t brightness) {
    return (color_val * brightness) / MAX_BRIGHTNESS;
  }

  // Function to write a color value to the WS2812 LED with brightness control
  void rgbLedWrite(uint8_t pin, uint8_t red_val, uint8_t green_val, uint8_t blue_val, uint8_t brightness) {
#if SOC_RMT_SUPPORTED
    rmt_data_t led_data[24];
    
    // Adjust color values based on brightness
    red_val = adjustBrightness(red_val, brightness);
    green_val = adjustBrightness(green_val, brightness);
    blue_val = adjustBrightness(blue_val, brightness);

    // Color values in RGB order 
    uint8_t color[3] = {red_val, green_val, blue_val};

    // Initialize RMT
    if (!rmtInit(pin, RMT_TX_MODE, RMT_MEM_NUM_BLOCKS_1, 10000000)) {
      Serial.println("RMT initialization failed!");
      return;
    }

 // Prepare the LED data
    int i = 0;
    for (int col = 0; col < 3; col++) {
      for (int bit = 0; bit < 8; bit++) {
        if (color[col] & (1 << (7 - bit))) {
          led_data[i].level0 = 1;     // HIGH bit
          led_data[i].duration0 = 8;  // 0.8us
          led_data[i].level1 = 0;     // LOW bit
          led_data[i].duration1 = 4;  // 0.4us
        } else {
          led_data[i].level0 = 1;     // LOW bit
          led_data[i].duration0 = 4;  // 0.4us
          led_data[i].level1 = 0;     // HIGH bit
          led_data[i].duration1 = 8;  // 0.8us
        }
        i++;
      }
    }

    // Write the data to the LED
    rmtWrite(pin, led_data, RMT_SYMBOLS_OF(led_data), RMT_WAIT_FOR_EVER);
#else
    Serial.println("RMT is not supported on this target.");
#endif
  }
};

#endif // WS2812LED_H

**Now Arduino example using it:**

#include "WS2812LED.h"

// Create an instance of the WS2812LED class on pin 21
WS2812LED myLED(21);

void setup() {
  // Leave setup empty
}

void loop() {
  // Cycle through colors with 1-second intervals
  myLED.defineRed();
  delay(1000);

  myLED.defineOrange();
  delay(1000);

  myLED.defineYellow();
  delay(1000);

  myLED.defineGreen();
  delay(1000);

  myLED.defineBlue();
  delay(1000);

  myLED.defineViolet();
  delay(1000);

  myLED.defineWhite();
  delay(1000);
}

moose4lord, 2024-08-26:
Nice. Take a look at the ESP32->RMT->RMTWriteNeoPixel example sketch. It's similar to what you have.

Rich968, 2024-08-26:
I found out about esp32-hal-rgb-led.cpp from a discussion on the ESP Arduino repository that is maintained by me-no-dev an Espressif employee. He has confirmed today that it does indeed use the new RMT driver. Yes, I did wonder whether Espressif had any Arduino WS2812 examples but decided just to play around with esp32-hal-rgb-led.cpp myself.

planevina, 2024-08-31:
I  have the same problem. And waiting for update. I don't want to change my led library, I like fastled

zackees, 2024-08-31:
Issue is non trivial but there is a work around floating around to set some magic defines to disable the onboard led

DKasle, 2024-08-31:
Hello @zackees , 

I believe you have checked in a new version of the FastLED library in the not too distant past to fix a compile issue ([FastLED 3.7.0 compile error #1628).  

Are you planning a new check in of  the FastLED library to fix this issue?

zackees, 2024-09-03:
> Hello @zackees ,
> 
> I believe you have checked in a new version of the FastLED library in the not too distant past to fix a compile issue ([FastLED 3.7.0 compile error #1628).
> 
> Are you planning a new check in of the FastLED library to fix this issue?

That was the plan. But the fix turns out to be MUCH harder than it first appeared.

The issue here is that that the legacy RMT driver does a freaking symbol check on a function on startup. This symbol is linked by default as weak and set to null. If the symbol is not NULL, then the esp platform will realize that the new RMT driver has been installed and will issue a fatal error... on startup. You don't even have to use RMT driver, merely linking to it is enough to cause the boot crash.

And guess what? It looks like the Arduino RMT library or something else is linking the old RMT driver in, always. That means we can't ever use the new RMT driver in Arduino code, which FastLED does. So we have to use a different fallback mechanism to fix this like the I2S library. This really really sucks.

I think everyone in the LED space is screwed with the IDF5.1 update for this feature.

So no, it doesn't look like a new RMT driver is on it's way. Fortunately the new I2S does have some nice feature to it including not flickering when the WIFI driver is being run. But geez, this sucks.

@samguyer

tomjuggler, 2024-09-03:
I just stumbled on this issue after spending an entire day trying to troubleshoot my brand new ESP32 S3 Super Mini's. I don't really do any embedded stuff that doesn't include FastLED in some way, so I missed the obvious for a while. 

Will getting another board without LED solve this problem for now?  (I do need the dual core S3 chip though)

tomjuggler, 2024-09-03:
> The UM TinyS3 is one of the ESP32 boards with a built-in WS2812 RGB LED, so it's pins_arduino.h file has the macro RGB_BUILTIN defined. This makes it incompatible with the legacy RMT driver included with IDF v5.
> 
> The esp32-hal-gpio.c file contains this note:
> 
> ```
> // RGB_BUILTIN is defined in pins_arduino.h
> // If RGB_BUILTIN is defined, it will be used as a pin number for the RGB LED
> // If RGB_BUILTIN has a side effect that prevents using RMT Legacy driver in IDF 5.1
> // Define ESP32_ARDUINO_NO_RGB_BUILTIN in build_opt.h or through CLI to disable RGB_BUILTIN
> #ifdef ESP32_ARDUINO_NO_RGB_BUILTIN
> #ifdef RGB_BUILTIN
> #undef RGB_BUILTIN
> #endif
> #endif
> ```
> 
> Until the problems with the legacy RMT driver are resolved in IDF v5, you may want to stick with arduino_esp v2.0.17 (which uses IDF v4) for this board. Or you can hack the boards pins_arduino.h file to remove the RGB_BUILTIN macro.
> 
> Update: there's a new example sketch included with arduino_esp v3.0.4 that shows how to workaround this issue. See examples/ESP32/RMT/Legacy_RMT_Driver_Compatible.

Did anyone get this example workaround sketch to work? ESP32S3 Super Mini. Not working - am I missing something?

tomjuggler, 2024-09-03:
I did it, following the somewhat cryptic instructions in the examples/ESP32/RMT/Legacy_RMT_Driver_Compatible sketch:
* _If the content of this file changes, it is necessary to delete the compiled Arduino IDE cache_
* _It can be done by changing, for instance, the "Core Debug Level" option, forcing the system to rebuild the Arduino Core_

Turns out you have to actually install with the "Core Debug Level" change, a simple compile doesn't suffice. 
Now to implement this in my Platformio project! Many thanks to @moose4lord for finding this simple work-around, and good luck to those re-writing FastLED to cope with the changes.

zackees, 2024-09-04:
Thanks. We are working on a new RMT driver.

DKasle, 2024-09-13:
> Thanks. We are working on a new RMT driver.

Hi @zackees ,
How is it going with the new RMT driver?  Thanks for your efforts to solve this issue!

zackees, 2024-09-13:
no uodate at this time

zackees, 2024-09-15:
Okay here's the update. RGBW being released on monday im now focusing my attention with RMT5.1.

If only you knew the challenges that I face...

DKasle, 2024-09-15:
Hi @zackees ,

Sorry to hear that this is so difficult to fix.  Please know that all your efforts are much appreciated!

zackees, 2024-09-15:
By the way this is a nasty issue. It looks like the rmt driver is sneaking in even without using it.

Unfortunately ESP32 will boot crash if it even detects you are linking into rmt 5.1 and 4.4 at the same time.

I'm focusing my attention on tackling this this next week.

puzzlelogic, 2024-09-17:
Hi all,  has this been addressed yet?  I have just tried 3.7.6 on Arduino IDE 2.3.2 and am still getting this issue.

F1p, 2024-09-17:
> Hi all, has this been addressed yet? I have just tried 3.7.6 on Arduino IDE 2.3.2 and am still getting this issue.

Can you downgrade to Arduino Core 2.0.17?

puzzlelogic, 2024-09-17:
> > Hi all, has this been addressed yet? I have just tried 3.7.6 on Arduino IDE 2.3.2 and am still getting this issue.
> 
> Can you downgrade to Arduino Core 2.0.17?

Hi,  Should have mentioned I am using ESP32-C3 at the moment and am using Espressif esp32 board v. 3.0.2.

Not sure how to downgrade the core Arduino library - do you mean go back to v 2.0.17 of the IDE?

Sorry - quite new to all this.

Thanks

F1p, 2024-09-17:
> Hi, Should have mentioned I am using ESP32-C3 at the moment and am using Espressif esp32 board v. 3.0.2.
>
> Not sure how to downgrade the core Arduino library - do you mean go back to v 2.0.17 of the [...]

[image]

puzzlelogic, 2024-09-17:
> ![image](https://private-user-images.githubusercontent.com/16503382/368291793-5fd4d381-a054-4bbe-bc5f-99e8e4953602.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Thanks - I will give that a go when I am next in front of the setup.

puzzlelogic, 2024-09-17:
> ![image](https://private-user-images.githubusercontent.com/16503382/368291793-5fd4d381-a054-4bbe-bc5f-99e8e4953602.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

That compiled and runs OK now but I do get quite a lot of warnings on compile:

FastLED\src\fastpin.h:13:32: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
 #pragma GCC diagnostic ignored "-Wvolatile"

Lots of the same.

puzzlelogic, 2024-09-17:
> ![image](https://private-user-images.githubusercontent.com/16503382/368291793-5fd4d381-a054-4bbe-bc5f-99e8e4953602.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

OK...  A small update here.  I tried upgrading everything to the latest so fastLED to 3.7.7 and ESP32 to 3.0.4 and it worked fine (ColorFlash example).  I added a few Serial prints in the code and got nothing out.  Turned on 'USB CDC on boot' to get the serial working and it crashes again.

Downgraded to esp32 2.0.17 again with USB CDC on boot enabled and it works again, with serial output but I get the 'FastLED\src\fastpin.h:13:32: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
#pragma GCC diagnostic ignored "-Wvolatile"' compile warnings again.

From my little testing it looks like the USB CDC ob boot that has issues with the later versions of esp32 and the fastLED library.

zackees, 2024-09-17:
You can ignore the warnings about the volatile. Thanks for helping debugging this.

The RMT 5.1 issue is a big problem and I'm currently spending more hours than I should trying to get it to work. I did a port of 5.1 to FastLED and it simply doesn't work, but the Arduino Core RMT 5.1 version does, which we can't use because we are platform-neutral. I strongly suspect at this point that it's the Arduino Core framework itself that is somehow causing the issue.

I haven't tried with a clean room esp idf version yet to try out our 5.1 port. But I really can't see anything wrong with it as the changes between the C -> C++ version only involve changing how structs are initialized (C allows you to skip fields, C++ doesn't so I have to zero initialize the structs then set the fields manually).

I've filed a bug with espressif to see if there is something going on with their Framework code.

It's also possible that this is an issue on my side, but I've double checked all the work a few times and I can't see anything that I'm doing wrong. I've also re-done the the example with different code and it just doesn't work. What's interesting is that the code runs fine and no errors are generated. It's just doesn't write any LEDs and nothing is appearing on the oscilloscope.

Strange.

Espressif issue that I've filed: https://github.com/espressif/idf-extra-components/issues/379

puzzlelogic, 2024-09-17:
> You can ignore the warnings about the volatile. Thanks for helping debugging this.
> 
> The RMT 5.1 issue is a big problem and I'm currently spending more hours than I should trying to get it to work. I did a port of 5.1 to FastLED and it simply doesn't work, but the Arduino Core RMT 5.1 version does, which we can't use because we are platform-neutral. I strongly suspect at this point that it's the Arduino Core framework itself that is somehow causing the issue.
> 
> I haven't tried a clean room esp idf version yet to try out our 5.1 port. But I really can't see anything wrong with it as the changes between the C -> C++ version only involve changing how structs are initialized (C allows you to skip fields, C++ doesn't so I have to zero initialize the structs then set the fields manually).
> 
> I've filed a bug with espressif to see if there is something going on with their Framework code.
> 
> It's also possible that this is an issue on my side, but I've double checked all the work a few times and I can't see anything that I'm doing wrong. I've also re-done the the example with different code and it just doesn't work. What's interesting is that the code runs fine and no errors are generated. It's just doesn't write any LEDs and nothing is appearing on the oscilloscope.
> 
> Strange.
> 
> Espressif issue that I've filed: [espressif/idf-extra-components#379](https://github.com/espressif/idf-extra-components/issues/379)

Thanks for all the hard work.  I have a project nearing its end (ish) and I decided to switch from Neopixel to fastLED - thinking it would be very minimal changes but could save some code size.

zackees, 2024-09-18:
This may help you.

https://github.com/zackees/esp-rmt-led-strip-component-idf-5-1-cpp

Melmac2, 2024-09-26:
I think this is same problem with my ESP32C3 Super Mini; it will only Serial print if FastLED.h is not included in any simple test sketch with a serial print in either setup or loop.  Unfortunately FastLED is a needed library for me.  Symptoms is the Serial Monitor will not work at all with this board selected,  as ESP32 C3 Dev Module, unless setting USB CDC on boot is enabled, but if I subsequently then include FastLED.h  to the simple sketch I get the "Conflict  driver_ng..." serial  error message only.  I updated to latest FastLED 3.7.8, ESP32 3.0.4, and Arduino IDE 2.3.3 , and see no change? My other various FastLED sketches run (drive matrix arrays), but not the serial.print components.
    Any update or suggestions?

zackees, 2024-09-27:
Can you run a test by including <iostream> and using std::cout? That has always worked for me.

zackees, 2024-09-27:
That driver message is a boot crash by the RMT driver. We are working on a fix

zackees, 2024-09-29:
RMT 5.1 prototype merged into master.

You'll need to set a define to make it work.

See this builder for details on how to enable it:

https://github.com/FastLED/FastLED/actions/workflows/build_esp32rmt.yml

zackees, 2024-09-30:
Scratch that, it's now FASTLED_RMT5

venice1200, 2024-10-01:
I am using your Library for my [tty2oled](https://github.com/venice1200/MiSTer_tty2oled) Display since for years without Problems.

Since ESP32 v3 for the Arduine IDE i got this "CONFLICT" message with ESP32 and ESP32-S3.
ESP32 was working again with FastLED 3.7 but not ESP32-S3 which compiles but reboots permantly.
Now with FastLED 3.8 the ESP32-S3 Version is also working again.

Many many Thx

zackees, 2024-10-01:
So to confirm this is the original esp32dev? If so there is an easy fix I can apply before the driver goes public

zackees, 2024-10-01:
Okay i've merged in a fix which disables DMA if the ESP32dev, 8266 or ESP32S2 is detected.

Let me know if this fixes it for you.

3.8 is still in a prerelease state so i really appreciate you bringing this to my attention before it goes live.

venice1200, 2024-10-01:
Hi,
except a few warnings during compilation the latest fix works fine for me and my ESP32 & ESP32-S3.

Cheers

```
In file included from e:\Projekte\Arduino\Arduino_MiSTer\libraries\FastLED_v3.8.0\src\crgb.cpp:1:
e:\Projekte\Arduino\Arduino_MiSTer\libraries\FastLED_v3.8.0\src\crgb.h:196:32: warning: inline function 'CRGB& CRGB::subtractFromRGB(uint8_t)' used but never defined
  196 |     FASTLED_FORCE_INLINE CRGB& subtractFromRGB(uint8_t d );
      |                                ^~~~~~~~~~~~~~~
e:\Projekte\Arduino\Arduino_MiSTer\libraries\FastLED_v3.8.0\src\crgb.h:187:32: warning: inline function 'CRGB& CRGB::addToRGB(uint8_t)' used but never defined
  187 |     FASTLED_FORCE_INLINE CRGB& addToRGB (uint8_t d);
      |                                ^~~~~~~~

...

In file included from e:\Projekte\Arduino\Arduino_MiSTer\libraries\FastLED_v3.8.0\src/pixel_iterator.h:9,
                 from e:\Projekte\Arduino\Arduino_MiSTer\libraries\FastLED_v3.8.0\src\platforms\esp\32\idf5_rmt.h:7,
                 from e:\Projekte\Arduino\Arduino_MiSTer\libraries\FastLED_v3.8.0\src\platforms\esp\32\idf5_rmt.cpp:7:
e:\Projekte\Arduino\Arduino_MiSTer\libraries\FastLED_v3.8.0\src/crgb.h:196:32: warning: inline function 'CRGB& CRGB::subtractFromRGB(uint8_t)' used but never defined
  196 |     FASTLED_FORCE_INLINE CRGB& subtractFromRGB(uint8_t d );
      |                                ^~~~~~~~~~~~~~~
e:\Projekte\Arduino\Arduino_MiSTer\libraries\FastLED_v3.8.0\src/crgb.h:187:32: warning: inline function 'CRGB& CRGB::addToRGB(uint8_t)' used but never defined
  187 |     FASTLED_FORCE_INLINE CRGB& addToRGB (uint8_t d);
      |                                ^~~~~~~~
```

F1p, 2024-10-02:
3.8 pre-release: Is there any change to how fade works? I seem to be full brightness and a slight flicker/instability visually in the LED

AtomM5 Lite - ESP32-S3

zackees, 2024-10-02:
Yes. The component brightness scaling channels has been altered in this new release to support high definition leds like the apa102 and it's possible you have discovered a bug in the new release.

F1p, 2024-10-02:
> Yes. The component brightness scaling channels has been altered in this new release to support high definition leds like the apa102 and it's possible you have discovered a bug in my code.
> 
> Can you explain what is going on and what you noticing.

Every loop, if WiFi is connected then it runs:
`
#ifdef ESP32
    leds[0] = CRGB::Green;
    leds[0].fadeLightBy(200);  // Green LED on, reduced brightness
    FastLED.show();
#endif` 

It appears it is reverting to full brightness and only fade is active for a short period seen below:

[image]

zackees, 2024-10-02:
> > Yes. The component brightness scaling channels has been altered in this new release to support high definition leds like the apa102 and it's possible you have discovered a bug in my code.
> > Can you explain what is going on and what you noticing.
> 
> Every loop, if WiFi is connected then it runs: `leds[0] = CRGB::Green; leds[0].fadeLightBy(200); FastLED.show();`

Um... this looks like expected behavior. You are setting it green then fading by an amount in the same line of code which executes together.

Why would it do anything else? Are you sure this is not user error?

zackees, 2024-10-02:
So yeah, none of the code I changed should impact fade operations on the leds themselves. It should only impact the global brightness and color values applied when the LEDs are transformed into driver pixels.

This example sketch works as expected:

```C++
#include <FastLED.h>

#define LED_PIN     2
#define COLOR_ORDER BRG
#define CHIPSET     WS2811
#define NUM_LEDS    1

#define BRIGHTNESS  128
#define FRAMES_PER_SECOND 30
#define REVERSE_DIRECTION false

CRGB leds[NUM_LEDS] = {0};

void setup() {
  delay(3000); // sanity delay
  FastLED.addLeds<CHIPSET, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS)
    .setCorrection(TypicalLEDStrip);
  FastLED.setBrightness(BRIGHTNESS);
}

void loop()
{
  EVERY_N_MILLISECONDS(1000) {
    leds[0] = CRGB::White;
  }

  EVERY_N_MILLISECONDS(8) {
    leds[0].fadeLightBy(32);
  }
  FastLED.show(); // display this frame
}
```

zackees, 2024-10-02:
Please post the full sketch.

F1p, 2024-10-03:
> Please post the full sketch.

https://github.com/F1p/Mitsubishi-CN105-Protocol-Decode/blob/f3b79eb73421f7f53b6288ae6a48f5a3f05fb862/ECODAN_Bridge.ino#L282

alba-ado, 2024-10-03:
I'm getting the following error with the IDF version 5.1.4 while compiling:

`/home/ado/Projects/Espressif/idf-test/components/FastLED/src/platforms/esp/32/led_strip/led_strip_rmt_dev.cpp:305:1: error: label 'err' defined but not used [-Werror=unused-label]
  305 | err:
      | ^~~
cc1plus: some warnings being treated as errors
`

[16 more replies]

closed 2024-10-31 by zackees