FastLED/FastLED #1657: Serious flickering problems on ESP32-C3

berkut0, 2024-08-14:
I have done some research and am convinced that the problem is with FastLED. I have tried different versions from 3.5 to 3.7.1 and the same problem. The same problem seems to migrate across repositories.

What I've tried:

1. I first [noticed this behaviour with wled](https://github.com/Aircoookie/WLED/issues/2698). I thought it might be a specific behaviour of wled.

2. Then I [tried Adalight-FastLED](https://github.com/dmadison/Adalight-FastLED) and got **exactly the same behaviour**.

3. I tried compiling with other versions of FastLED or with a library I got directly from github.

All a complete failure.

You don't need to remind me about 'power' and 'ground', the problem is ONLY with FastLED. I think there is another library involved in the wled operation and only the USB transfer somehow involves FastLED.
This is all about the Esp32-c3 board.

zackees, 2024-08-16:
What's your LED type?

Please post your code.

berkut0, 2024-08-17:
> What's your LED type?

WS2812-like strip

> Please post your code.

Just this code compiled to ESP32-C3:
https://github.com/dmadison/Adalight-FastLED
<details><summary>CODE</summary>

``` C++
/*
 *  Project     Adalight FastLED
 *  @author     David Madison
 *  @link       github.com/dmadison/Adalight-FastLED
 *  @license    LGPL - Copyright (c) 2017 David Madison
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU Lesser General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 * 
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU Lesser General Public License for more details.
 *
 * You should have received a copy of the GNU Lesser General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

#include <Arduino.h>

// --- General Settings
const uint16_t 
	Num_Leds   =  80;         // strip length
const uint8_t
	Brightness =  255;        // maximum brightness

// --- FastLED Setings
#define LED_TYPE     WS2812B  // led strip type for FastLED
#define COLOR_ORDER  GRB      // color order for bitbang
#define PIN_DATA     6        // led data output pin
// #define PIN_CLOCK  7       // led data clock pin (uncomment if you're using a 4-wire LED type)

// --- Serial Settings
const unsigned long
	SerialSpeed    = 115200;  // serial port speed
const uint16_t
	SerialTimeout  = 60;      // time before LEDs are shut off if no data (in seconds), 0 to disable

// --- Optional Settings (uncomment to add)
#define SERIAL_FLUSH          // Serial buffer cleared on LED latch
// #define CLEAR_ON_START     // LEDs are cleared on reset

// --- Debug Settings (uncomment to add)
// #define DEBUG_LED 13       // toggles the Arduino's built-in LED on header match
// #define DEBUG_FPS 8        // enables a pulse on LED latch

// --------------------------------------------------------------------

#include <FastLED.h>

CRGB leds[Num_Leds];
uint8_t * ledsRaw = (uint8_t *)leds;

// A 'magic word' (along with LED count & checksum) precedes each block
// of LED data; this assists the microcontroller in syncing up with the
// host-side software and properly issuing the latch (host I/O is
[... 158 lines cut ...]
	}
}

void timeouts(){
	// No data received. If this persists, send an ACK packet
	// to host once every second to alert it to our presence.
	if((t - lastAckTime) >= 1000) {
		Serial.print("Ada\n"); // Send ACK string to host
		lastAckTime = t; // Reset counter

		// If no data received for an extended time, turn off all LEDs.
		if(SerialTimeout != 0 && (t - lastByteTime) >= (uint32_t) SerialTimeout * 1000) {
			memset(leds, 0, Num_Leds * sizeof(struct CRGB)); //filling Led array by zeroes
			FastLED.show();
			mode = Header;
			lastByteTime = t; // Reset counter
		}
	}
}
```

</details>

Same behaviour on WLED compiled on ESP32-C3:
https://github.com/Aircoookie/WLED

zackees, 2024-08-17:
Can you do me a favor and try your project with the Adafruit Neopixel and report if that works? I've been toying around with the idea of bringing in a subset of their library as a WS2812 controller type.

zackees, 2024-08-17:
This probably has to do with the RMT driver on the ESP32 flavors. We use a DMA like engine to push out the data. There's rumor that Adafruit already has figured out how to do it correctly.

berkut0, 2024-08-17:
> Can you do me a favor and try your project with the Adafruit Neopixel and report if that works? I've been toying around with the idea of bringing in a subset of their library as a WS2812 controller type.

I made a Frankenstein's monster by crossing FastLED and another library, NeopixelBus (by makuna). Unfortunately, the result is exactly the same.

<details><summary>Frankenstein's Code</summary>

``` C++

/*
 *  Project     Adalight FastLED
 *  @author     David Madison
 *  @link       github.com/dmadison/Adalight-FastLED
 *  @license    LGPL - Copyright (c) 2017 David Madison
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU Lesser General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 * 
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU Lesser General Public License for more details.
 *
 * You should have received a copy of the GNU Lesser General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#include <Arduino.h>

// --- General Settings
const uint16_t
  Num_Leds = 300;  // strip length
const uint8_t
  Brightness = 255;  // maximum brightness

// --- FastLED Setings
#define LED_TYPE WS2812B  // led strip type for FastLED
#define COLOR_ORDER GRB   // color order for bitbang
#define PIN_DATA 6        // led data output pin
// #define PIN_CLOCK  7       // led data clock pin (uncomment if you're using a 4-wire LED type)

// --- Serial Settings
const unsigned long
  SerialSpeed = 115200;  // serial port speed
const uint16_t
  SerialTimeout = 60;  // time before LEDs are shut off if no data (in seconds), 0 to disable

// --- Optional Settings (uncomment to add)
#define SERIAL_FLUSH  // Serial buffer cleared on LED latch
// #define CLEAR_ON_START     // LEDs are cleared on reset

// --- Debug Settings (uncomment to add)
// #define DEBUG_LED 13       // toggles the Arduino's built-in LED on header match
// #define DEBUG_FPS 8        // enables a pulse on LED latch

// --------------------------------------------------------------------

#include <NeoPixelBus.h>
NeoPixelBus<NeoGrbFeature, NeoWs2812xMethod> strip(Num_Leds, PIN_DATA);

#include <FastLED.h>
// #include <Adafruit_NeoPixel.h>

// Adafruit_NeoPixel pixels(Num_Leds, PIN_DATA, NEO_GRB + NEO_KHZ800);
// struct CRBG{

// }

CRGB leds[Num_Leds];
[... 186 lines cut ...]
}

void timeouts() {
  // No data received. If this persists, send an ACK packet
  // to host once every second to alert it to our presence.
  if ((t - lastAckTime) >= 1000) {
    Serial.print("Ada\n");  // Send ACK string to host
    lastAckTime = t;        // Reset counter

    // If no data received for an extended time, turn off all LEDs.
    if (SerialTimeout != 0 && (t - lastByteTime) >= (uint32_t)SerialTimeout * 1000) {
      memset(leds, 0, Num_Leds * sizeof(struct CRGB));  //filling Led array by zeroes
      // FastLED.show();
      show();
      mode = Header;
      lastByteTime = t;  // Reset counter
    }
  }
}

```

</details>

Unfortunately I can't test the classic library Neopixel by Adafruit any more. Maybe much later. It was a bit urgent.

zackees, 2024-08-17:
You are trying to bang out the WS data via the serial port yes?

berkut0, 2024-08-17:
@zackees Adalight is a protocol. It forms a data packet that is sent through the COM port. Yes, essentially the data is sent through the serial port. The operation of the data output to the WS chips should not be affected. Specifically in this case, I replaced the board with an RP2040 based board and compiling the original Adalight FastLED code worked without any changes to the code.

vincentbraillard, 2024-08-20:
Any idea? same issues with same settings

berkut0, 2024-08-21:
Maybe related:
https://github.com/FastLED/FastLED/pull/1515
https://github.com/FastLED/FastLED/issues/1349

But changes in v3.6.0 do not solve the problem in that case
I tried 3.6.0 by the way, it compiles with errors in general, this is fixed later, but in other versions the problem persists.

zackees, 2024-08-23:
RMT driver in Arduino 3.2.1+ uses the platform esp-idf 5.1 toolchain. This broke the RMT driver.

The RMT driver was built on the the idf 4.x toolchain and works.

Arduino IDE can be downgraded back to idf 4.x, and so can PlatformIO

artkeller, 2024-08-23:
@zackees: Do you think that this RMT problem is the reason for the runtime error on ESP32-C3: “E (79) rmt(legacy): CONFLICT! driver_ng is not allowed to be used with the legacy driver” when I include FastLED and declare a normal IO port (e.g. pinMode(PORT, OUTPUT). The code runs normally without FastLED.

zackees, 2024-08-23:
Yes, this is from the esp-idf v5.1 update to the arduino core.

What is happening is that many boards are using the new RMT library to connect to a possible WS2812 status led, thus initializing the new RMT driver. For some reason both the new RMT driver and the legacy RMT driver cannot both be initialized in the same program or you get this crash.

Many other led projects are dealing with the same issue.

zackees, 2024-08-28:
FastLED can be adapted to use the new idf 5.1 driver and it will be much better for it as the new driver is designed to push large blocks of RMT-encoded data even when the interrupt is shut off for large amounts of time due to wifi interrupts and flash writing.

The IDF 4.x RMT encoder that we've implemented has two modes, a stream encoding mode that simply copies the RGB buffer (re-ordered) and then stream encodes it on a tight interrupt two symbols at a time, if memory serves correct. This works fine when wifi is not running. When wifi runs this interrupt doesn't fire fast enough, (>50uS) and the ws2812 will reset, which causes the flashing/flickering behavior.

The other mode that we have for the 4.x RMT driver is allocate-everything-upfront for the RMT buffer and populate it. This requires a ton of memory but the benefit is that no interrupt is needed to stream encode, as everything is already encoded.

The reason we haven't done the new RMT driver is because it's non-trivial and the FastLED codebase in this area was tightly coupled between the FastLED interface wrapper and the RMT driver itself. I've spent the last week making the wrapper portion dead simple and pushing implementation details into the RMT driver and the *.cpp file where it belongs.

At this stage we are pretty much ready for the new IDF 5.1 RMT driver to be implemented.

I'll gladly accept a patch from anyone that chooses to do this.

I've made the FastLED codebase extremely easy to start developing on. At this point you can just fork, clone locally and then hit the compile button and the default blink demo will compile to s3 using idf 5.1.

There are some caveats though like the idf 4.4 RMT driver runs all the RMT devices in parallel during show(), and this behavior needs to be preserved in the new idf 5.1 driver. So this behavior needs to be preserved. However the "allocate everything up front" part of the idf 4.4 driver *probably* doesn't have to be implemented on the first pass for the 5.1 driver migration because the amount of buffering in the 5.1 RMT driver seems to have vastly improved.

samguyer, 2024-08-28:
I'm going to set aside some time to understand the changes to the ESP-IDF, but I think the truth is that they don't change the fundamental underlying problem. A lot of the code I wrote in the current RMT driver is essentially taken from the internal implementation of Espressif's driver -- they aren't doing anything substantially different in terms of responding to interrupts and filling the buffers. The big difference in FastLED is that we service up to 8 outgoing channels at a time, allowing for a significant amount of parallelism.

zackees, 2024-08-28:
@samguyer this time we can make the RMT driver fully async so that it's now possible that FastLED.show() only blocks on the next call for the duration of the time necessary to finish the RMT driver output.

I noticed that you are kicking off the RMT driver during the last call to show() in the RMT driver batch. Now there is an explicit onBeginShow() and onEndShow() in the fastled ClockLess wrapper that I've implemented. This used to be getDither() / setDither() and the default still does this. My suggestion is that you kick off the RMT driver on the first call to onEndShow() in the batch and then sync block on the next first call to onBeginShow() in the batch. This would allow client logic to run concurrently to the RMT output.

samguyer, 2024-08-28:
@zackees The onBegin and onEnd will help a lot. It also fixes a long-standing bug, which is that the RMT driver locks up if you call show on a single controller (because it is waiting for all the other calls to show).

artkeller, 2024-08-28:
@zackees @samguyer Thanks guys, I can see ‘light at the end of the tunnel’ and understand the problem much better now. Good work! I'm really excited...

softhack007, 2024-08-30:
> Same behaviour on WLED compiled on ESP32-C3:
> https://github.com/Aircoookie/WLED

Just for clarification -  WLED only uses fastled for fast color utilities and palettes. In WLED all led outputs are driven by makuna/NeoPixelBus. 

In case the same behaviour exists with WLED and with another fastled based firmware (adalight-fastled) then it means there is either a problem that neither fastled nor NeoPixelBus are handling properly.

zackees, 2024-08-30:
Do you know if WLED is using the idf 5.1 driver for RMT?

softhack007, 2024-08-30:
@zackees definitely not. For esp32-c3 we use arduino-esp32 v2.0.9 (idf 4.4.4) and for classic esp32 it's still arduino-esp32 v1.x.

artkeller, 2024-08-31:
@zackees It seems that Adafruit has found the philosopher's stone - didn't Espressif test it properly? (https://github.com/espressif/arduino-esp32/blob/gh-pages/LIBRARIES_TEST.md) 
[image]

vincentbraillard, 2024-08-31:
Nice. But what doesn't work with esp?

zackees, 2024-09-03:
https://github.com/artkeller

I'm not entirely sure what they are testing. I can't find what their failure criteria is. I know that we are using the open source platformio version to test our code and we have an entire cluster of VM's that test on every CL. So maybe they are testing on official version the open source version.

artkeller, 2024-09-04:
@zackees @samguyer I see that we are not getting very far with analysing Espressif at the moment. The new driver seems to offer significant advantages and its implementation promises improvements. However, we realise that even parallel RGB strip projects are still based on earlier IDF development versions of Espressif. It is clear that collaboration should be intensified to discuss the requirements of IDF 5.1 more comprehensively and consolidate the code in other projects. This fragmentation and lack of adaptation to new hardware and drivers risks bringing OSS expertise ‘to its knees’.
Currently, I see the IDF5 wrapper for IDF 4.4 in FastLED version 3.7.3 as well as your optimistic discussions in this thread. The previously known limitations that ESP32-NG projects only work with FastLED if there are no other GPIO activities (like push buttons, serial communication, etc.) in the sketch are an obstacle. Only the log level outputs will still work.
Do you have a clear plan for implementation and joint testing? Apart from the usual lack of time, what else is missing? Do we need extended community communication and more contributors?

artkeller, 2024-09-07:
@zackees These explanations are absolutely plausible to me and are technically competent - thank you very much for your feedback! 
At the same time, I recognise in it a kind of pioneering work - so you are more or less alone in the field. And I see a really good plan. Thank you. You are probably leading a major contribution to the community! Thanks for that too!

I can see that a lot is happening in FastLED at the moment and is frequently published. It's a really excellent project in the community. I'll keep my fingers crossed that your ideas work out!

zackees, 2024-09-18:
I was able to correctly port the espressif led_strip library to C++

https://github.com/zackees/esp-rmt-led-strip-component-idf-5-1-cpp

Features stream encoding.

zackees, 2024-09-30:
RMT5 is in the master branch now. It features DMA mode and the RMT buffer has increased from 64 symbols to 1024 per channel. 

The number of channels has increased from 4 to 8. You can also allocate an RMT channel for your project and the code will gracefully detect this at runtime and scale back its RMT usage.

This code is live on Master. If you are still having this issue then I suggest you try the latest and see if it works for you.

The release will go out this week sometime when proper polish is put on everything.

zackees, 2024-09-30:
Also one more thing. FastLED.show() is now fully async as long as all RMT channels are drained. So the next release will allow you to compute the next frame while the current frame is being hardware bitbanged out by the RMT hardware.

zackees, 2024-10-12:
Looks like RMT5 fixed everyone. Closing this bug report. We did a silent release on 3.8 to fix everyone as we get ready for the big 4.0 release.

closed 2024-10-12 by zackees