FastLED/FastLED #1672: Wrong timing for WS2812B on ESP32-C6

fweiss, 2024-08-19:
The output from the ESP32-C6 to a 8x8 WS2812B array is wrong. The bit time is 0.625 microsecond, but it should be 1.25 microsecond. Consquently random colored pixels are displayed. I was trying to move existing code from ESP32 to ESP32C6, keeping the FastLED library. I'm using this code to reproduce:
```
#include <FastLED.h>
#define NUM_LEDS 64
CRGB leds[NUM_LEDS];
void setup() { FastLED.addLeds<WS2812B, 22>(leds, NUM_LEDS); }
void loop() {
	leds[0] = CRGB::White; FastLED.show(); delay(30);
	leds[0] = CRGB::Black; FastLED.show(); delay(30);
}
```

Environment is Windows 10, Arduino IDE 2.3.2, FastLED 3.7.3, Seed Studios XIAO ESP32C6, WS2812B array.

Scope trace shows the errant timing.

[image]

fweiss, 2024-08-19:
I found a work-around with the following code:
```
#include <FastLED.h>

#define FMUL (CLOCKLESS_FREQUENCY/8000000)

template <uint8_t DATA_PIN, EOrder RGB_ORDER = RGB>
class WS2812Controller400Khz : public ClocklessController<DATA_PIN, 4 * FMUL, 10 * FMUL, 6 * FMUL, RGB_ORDER> {};

template<uint8_t DATA_PIN, EOrder RGB_ORDER> class WS2812C : public WS2812Controller400Khz<DATA_PIN, RGB_ORDER> {};

#define NUM_LEDS 64
CRGB leds[NUM_LEDS];
void setup() { FastLED.addLeds<WS2812C, 22>(leds, NUM_LEDS); }
void loop() {
	leds[0] = CRGB::White; FastLED.show(); delay(30);
	leds[0] = CRGB::Black; FastLED.show(); delay(30);
}
```
Note the doubled timing parameters for ``ClocklessController``.

I suspect there's a clock setup incorrectly for the ESP32C6 which is a RISC processor. I'd love to help but need some guidance with:
- how to run a local fork of FastLED
- where in the code a ESP32C6-specific clock parameter would be
-

zackees, 2024-08-22:
Here is a platformio that will point to your own remote fork of the fastled library. You can probably just use a path instead of a github url.

```
[env:esp32c6] ; the esp32c6 part can be renamed to anything you want.
; Note that this special platform override is used to bring in the
; idf5.1 toolchain that is now included in Arduino 2.3.1+, and may
; not be necessary in the near future.
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zi
framework = arduino
; This is a popular ESP32-S3 board.
board = esp32-c6-devkitc-1

upload_protocol = esptool

; The exception decoder is a useful tool for debugging that is difficult
; to enable in the Arduino IDE. But in platformio it will decode the
; 0x0f4852c0 addreesses into the actual function names.
monitor_filters = 
	default
	esp32_exception_decoder

lib_deps =
    FastLED=https://github.com/MY_FORK/FastLED.git
```

zackees, 2024-08-22:
And also - you should be able to use intellisense to jump right into the code.

zackees, 2024-08-23:
By default we are going to be using the RMT driver on WS2812, you can find that here:

src/platforms/esp/32/clockless_rmt_esp32.cpp

fweiss, 2024-08-24:
I have narrowed down the problem a bit. I've been patching the Arduino FastLED 3.7.3 library locally for experiments. As suggested, I'll move to PlatformIO as well.

In ``clockless_rmt_esp32.cpp::151`` replace ``DIVIDER`` with 4. Then I get the correct output bit period on XIAO ESP32C6. Oddly changing ``DIVIDER`` in ``clockless_rmt_esp32.h::178`` has no effect.

There's a comment re ESP32H2 at ``clockless_rmt_esp32.h::222``, not sure if it is pertinent. Also note that the Seed Studio XIAO ESP32C6 (the one I'm using) has a 40 MHz XTAL, while the Espressif ESP32C6 Dev Kit apparently has a different clock.

I tried my "work-around" posted above on a LOLIN D32 and it dutifully doubled the bit time.

netmindz, 2024-08-26:
Doesn't really make much of a difference really @zackees as compiling the library alone isn't of any use really. You need a project that actually consumes the library, so you need a test project that references the FastLED source

zackees, 2024-08-27:
@netmindz

By default it compiles to esp32-s3 using a blink demo. It's designed exactly for the use case your are describing.

netmindz, 2024-08-28:
> @fweiss is it too much to ask for a C6 patch?

Firstly give the person chance to respond and also I don't think that kind of comment it helpful to encouraging a healthy development community.

They have done the hard part of identifying both the issue and detailed what is needed to be changed, don't then have a go at them for not being familiar with GitHub PRs

fweiss, 2024-08-28:
@zackees @netmindz I cleared my workbench and have started on a fork/PR for this.

fweiss, 2024-08-29:
I've narrowed it down some more. Using log statements and comparing running on esp32-s3 with esp32-c6, I've observed a diifference in the macro ``APB_CLK_FREQ``. On S3 it is 80,000,000 while on C6 it is 40,000,0000. Look at ``idf4_rmt.cpp`` around line 97. This explains why on C6 the bit time is half as long, since it thinks it has a slower clock. There's a comment there re H2, but I haven't seen any documents showing H2 or C6 use a 40 MHz APB clock.

Now the other question is, where is ``APB_CLK_FREQ`` defined? It's not in the FastLED source code, so I suppose it comes from the ``esp-idf`` toolchain. I will, of course dig deeper (I'm not afraid of ``esp-idf``), but if anybody has some tribal knowledg, it would be a help.

zackees, 2024-08-29:
To find this I opened up the FastLED repo in VSCode + PlatformIO and did a search for APB_CLK_FREQ.

I found it then right clicked it then it jumped right to the source code in esp idf.

~platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/soc/esp32c6/include/soc/soc.h

Using the public fork of esp-idf for our repo I clearly see that it's defined correctly. It' possible if you are working with Arduino that this value is wrong:

[image]

zackees, 2024-08-29:
Thanks for nailing it. I went ahead and manually set the RMT frequency for these platforms.

closed 2024-08-29 by zackees

fweiss, 2024-08-30:
@zackees Firstly, sorry that I appeared to not be able to navigate the codebase with VSC. I just have a lot going on.

Thanks for the effort, however nothing has changed for me. I am still seeing, **with the oscilloscope**, the 0.625 us bit times going out to the WS2812B array **and erratic flickering**. It looks to me like the change at 64f1da566d7ee2859cabd084a829d3bcc5100d8f was a no-op, it left the APB_CLK_FREQ at the errant 40 MHz.

BTW, when I change ``F_CPU_RMT_CLOCK_MANUALLY_DEFINED`` in ``idf4_rmt.cpp:106`` from 40 * 1000000 to 80 * 1000000, the output is correct.

Now that the root cause is pinpointed to the Espressif arduinoespressif library, I'll follow up with Espressif to see why they have ``APB_CLK_FREQ`` set to 40 M Hz instead of 80. I'm studying the [ESP32-C6 Technical Reference Manual](https://www.espressif.com/sites/default/files/documentation/esp32-c6_technical_reference_manual_en.pdf) and it looks well that the APB clock is normally 80 MHz, but can throttle down to 40 MHz. to conserve power. However, that shift (which I believe is not happening) would result in peripherals, such as RMT, getting bad timing. (See relevant page 1277 in the TRM).

It's up to you to reopen this issue and undo the above referenced commit. Thank you for helping me along with this issue.

zackees, 2024-08-30:
Thanks for the tip, merged, see https://github.com/FastLED/FastLED/commit/26c21b8215206c9a51d15dd5b3fc8e1d8183644f