FastLED/FastLED #1673: Support for Raspberry Pi Pico 2 - RP2350

FeuerSturm, 2024-08-21:
Hi,

I'd love to see support for the recently released Raspberry Pi Pico 2 - RP2350.
https://www.raspberrypi.com/products/rp2350/

It features 12 PIO state machines and thus driving 12 pins simultaneously would be possible.

Is this already being worked on?

Thanks!

zackees, 2024-08-23:
we accept pull requests

zackees, 2024-08-23:
Have you attempted a compile for this platform? Can you point out any platformio.ini files for the chip type.

FeuerSturm, 2024-08-24:
Currently there's only a WIP/draft PR to add initial support for the RP2350 for the arduino-pico core by Earle F. Philower III that I wasn't able to test yet as I have just received my Pico 2:
https://github.com/earlephilhower/arduino-pico/pull/2337

Unfortunately I am just a generic Arduino User and besides some testing I would not be able to help much.

maxgerhardt, 2024-08-24:
Supporting the RP2350 in the Arduino-Pico core is still in progress, as well as expanding PlatformIO platform to support it. It's still going to take a little bit of time to have all that ready.

FeuerSturm, 2024-08-28:
@zackees 
Tried to compile a sketch for RP2040 and RP2350 with FastLED 3.7.4 and I am getting those errors for BOTH:

`In file included from c:\Users\myPC\Documents\Arduino\libraries\FastLED\src/platforms/arm/rp2040/clockless_arm_rp2040.h:7,
                 from c:\Users\myPC\Documents\Arduino\libraries\FastLED\src/platforms/arm/rp2040/fastled_arm_rp2040.h:6,
                 from c:\Users\myPC\Documents\Arduino\libraries\FastLED\src/platforms.h:38,
                 from c:\Users\myPC\Documents\Arduino\libraries\FastLED\src/FastLED.h:66,
                 from C:\Users\myPC\Downloads\SRGBmods_Pico_LED_Controller_1724706442\SRGBmods_Pico_LED_Controller_1724706442.ino:13:
c:\users\myPC\documents\arduino\libraries\fastled\src\platforms\arm\common\m0clockless.h: In function 'int showLedData(volatile uint32_t*, uint32_t, const uint8_t*, uint32_t, M0ClocklessData*)':
c:\users\myPC\documents\arduino\libraries\fastled\src\platforms\arm\common\m0clockless.h:316:40: error: 'SysTick' was not declared in this scope
  316 |       uint32_t ticksBeforeInterrupts = SysTick->VAL;
      |                                        ^~~~~~~
In file included from c:\Users\myPC\Documents\Arduino\libraries\FastLED\src/FastLED.h:79:
c:\Users\myPC\Documents\Arduino\libraries\FastLED\src/fastspi.h: At global scope:
c:\Users\myPC\Documents\Arduino\libraries\FastLED\src/fastspi.h:172:23: note: '#pragma message: Forcing software SPI - no hardware SPI for you!'
  172 | #      pragma message "Forcing software SPI - no hardware SPI for you!"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Using library Adafruit TinyUSB Library at version 3.3.3 in folder: C:\Users\myPC\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.1\libraries\Adafruit_TinyUSB_Arduino 
Using library SPI at version 1.0 in folder: C:\Users\myPC\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.1\libraries\SPI 
Using library FastLED at version 3.7.4 in folder: C:\Users\myPC\Documents\Arduino\libraries\FastLED 
exit status 1
Compilation error: exit status 1`

zackees, 2024-08-28:
I just added RP2040 to the build list and there are no compiler errors here. I noticed that your log stream mentions "Arduino15", what version of Arduion IDE are you running?

[image]

maxgerhardt, 2024-08-28:
When I read 
https://github.com/FastLED/FastLED/blob/master/.github/workflows/build_rp2040.yml#L12-L16 

correctly, if you use just `board = nanorp2040connect` with standard `platform = raspberrypi`, the used Arduino core is the https://github.com/arduino/ArduinoCore-mbed one, not EarlePhilhower's https://github.com/earlephilhower/arduino-pico. That needs a special platform and either different board ore core setting. Example
```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = rpipico ; earle-philhowever variant of "pico"
framework = arduino
```
or
```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = nanorp2040connect
framework = arduino
; use EarlePhilhower core
board_build.core = earlephilhower
```
Currently when using `platform = https://github.com/maxgerhardt/platform-raspberrypi.git` you will get the previous 3.9.5 version (no RP2350 support yet), not the latest 4.0.1. But you can as of now already use the 4.0.1 core in the Arduino-IDE or on the Arduino CLI. I just need a little more time to finish supporting it in PlatformIO. In **any** case though, to use the Arduino-Pico core, you will need to use the custom `platform`, as the mainline platform is not going to merge it due to political reasons.

FeuerSturm, 2024-08-28:
> I just added RP2040 to the build list and there are no compiler errors here. I noticed that your log stream mentions "Arduino15", what version of Arduion IDE are you running?
> 
> [image]

I am using ArduinoIDE 2.3.2 on Windows 11,
latest stable FastLED lib, latest stable arduino-pico core by Earle Philhower.

Any ideas where to look for the problem?

zackees, 2024-08-28:
When this happened on the RP2040 it was an issue with `sei()` and `cli()` being defined when compiling Arduino. We fixed that. Others with the RP2040 said they only ran into this issue when compiling with debug information `-g` which Arduino does by default.

As far as the RP2035 - there isn't much information on the internet about this.

Here's a bug report to get your started.
https://github.com/FastLED/FastLED/issues/1481

If you want to try this yourself you can just clone the FastLED repo, drop this as your platformio.ini at the root:

```
[platformio]
src_dir = dev ; target is ./dev/dev.ino

[env:dev]
; This is the espressif32 platform which is the 4.1 toolchain as of 2024-Aug-23rd
; platform = espressif32
; The following platform enables the espressif32 platform to use the 5.1 toolchain, simulating
; the new Arduino 2.3.1+ toolchain.

# Developement branch of the open source espressif32 platform
platform = https://github.com/maxgerhardt/platform-raspberrypi.git

framework = arduino
board = rpipicow

upload_protocol = esptool

monitor_filters = 
	default
	esp32_exception_decoder  ; Decode exceptions so that they are human readable.

; Symlink in the FastLED library so that changes to the library are reflected in the project
; build immediatly.
lib_deps =
  FastLED=symlink://./

board_build.filesystem_size = 0.5m
```

And hit compile.

zackees, 2024-08-28:
If you are on windows then enable git to clone paths with long file names. Run a command terminal with admin privileges and paste in:

`git config --system core.longpaths true`

zackees, 2024-08-28:
Oh, I noticed now that you are using a different pico framework git repository.

zackees, 2024-08-28:
> When I read
> 
> https://github.com/FastLED/FastLED/blob/100e44360f2ebe520e0b992d2e5677c73f5258ec/.github/workflows/build_rp2040.yml#L12-L16
> 
> correctly, if you use just `board = nanorp2040connect` with standard `platform = raspberrypi`, the used Arduino core is the https://github.com/arduino/ArduinoCore-mbed one, not EarlePhilhower's https://github.com/earlephilhower/arduino-pico. That needs a special platform and either different board or core setting ([docs](https://arduino-pico.readthedocs.io/en/latest/platformio.html)). Example
> 
> ```ini
> [env:pico]
> platform = https://github.com/maxgerhardt/platform-raspberrypi.git
> board = rpipico ; earle-philhowever variant of "pico"
> framework = arduino
> ```
> 
> or
> 
> ```ini
> [env:pico]
> platform = https://github.com/maxgerhardt/platform-raspberrypi.git
> board = nanorp2040connect
> framework = arduino
> ; use EarlePhilhower core
> board_build.core = earlephilhower
> ```
> 
> Currently when using `platform = https://github.com/maxgerhardt/platform-raspberrypi.git` you will get the previous Arduino-Pico 3.9.5 version (no RP2350 support yet), not the latest 4.0.1. But you can as of now already use the 4.0.1 core in the Arduino-IDE or on the Arduino CLI. I just need a little more time to finish supporting it in PlatformIO. In **any** case though, to use the Arduino-Pico core, you will need to use the custom `platform`, as the mainline platform is not going to merge it due to political reasons.

Am I to understand you that this is the correct setting for RP2050?

```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = nanorp2040connect
framework = arduino
; use EarlePhilhower core
board_build.core = earlephilhower
```

zackees, 2024-08-28:
And can you point me to a board.json for this item?

https://github.com/FastLED/FastLED/actions/runs/10604869165/job/29392457473

It was building before, now it's not.

zackees, 2024-08-28:
Uh... I broke the RPXXXX builds. I'm reverting back.

I need recommended:

RP2350: platformio.ini + board.json
RP2335: platformio.ini + board.json

If I get this then I'll put it in our build system so we can track the what the compiler errors are try to fix them with a repro case.

maxgerhardt, 2024-08-28:
No you're mixing up the platform and package. 
>Running command: pio project init --project-dir .build/rpipico2 --board rpipico2 --project-option=platform=https://github.com/earlephilhower/arduino-pico.git "--project-option=board_build.core = earlephilhower" --project-option=board_build.filesystem_size=0.5m 

The platform should be `https://github.com/maxgerhardt/platform-raspberrypi.git`. If you want it to to use the latest Arduino-Pico *package*, you do that with `"-project-option=platform_packages=frameowork-arduinopico@https://github.com/earlephilhower/arduino-pico.git"`.

zackees, 2024-08-28:
> No you're mixing up the platform and package.
> 
> > Running command: pio project init --project-dir .build/rpipico2 --board rpipico2 --project-option=platform=https://github.com/earlephilhower/arduino-pico.git "--project-option=board_build.core = earlephilhower" --project-option=board_build.filesystem_size=0.5m
> 
> The platform should be `https://github.com/maxgerhardt/platform-raspberrypi.git`. If you want it, on top, to use the latest Arduino-Pico _package_ (4.0.1 instead of 3.x), you do that with `"-project-option=platform_packages=frameowork-arduinopico@https://github.com/earlephilhower/arduino-pico.git"`.

[image]

You've got some board.json stashed in your platform io? I noticed on windows I have issues.

I need the URL for the board.json. Our build script injects this into the ~/.platformio/boards directory using a hack.

zackees, 2024-08-29:
I've asked Earle Philhower what I'm doing wrong. But so far it's not working for platformio and I don't want switch out build system over to arduino cli yet.

zackees, 2024-08-30:
Okay this is the next compiler error, missing a definition of SysTick

@maxgerhardt 

https://github.com/FastLED/FastLED/blob/4a4c7117ff2a32fadeb404b76bb1f3baad4d5034/src/platforms/arm/common/m0clockless.cpp#L11

zackees, 2024-08-31:
Good news I was able to get the RP2350 to compile

[image]

closed 2024-08-31 by zackees

FeuerSturm, 2024-08-31:
@zackees 
RP2350 features three PIO instances compared to RP2040's two PIO instances.

Now SysTick is defined twice!

I created a pull request to fix both:
https://github.com/FastLED/FastLED/pull/1689