FastLED/FastLED #1684: Recent commits broke CMakeLists.txt

ausculta, 2024-08-29:
A recent commit renamed  src/platforms/esp/32/clockless_rmt_esp32.cpp --> src/platforms/esp/32/rmt.cpp 

Then another commit renamed src/platforms/esp/32/rmt.cpp --> src/platforms/esp/32/idf4_rmt.cpp

But CMakeLists.txt wasn't updated, so FastLED won't compile with ESP-IDF & ESP-Arduino. It should now look like this:

# FastLED
# https://github.com/FastLED/FastLED
# MIT License

cmake_minimum_required(VERSION 3.5)

set(FastLED_SRCS
  src/bitswap.cpp
  src/colorpalettes.cpp
  src/colorutils.cpp
  src/FastLED.cpp
  src/hsv2rgb.cpp
  src/lib8tion.cpp
  src/noise.cpp
  src/platforms.cpp
  src/power_mgt.cpp
  src/wiring.cpp
  src/platforms/esp/32/idf4_rmt.cpp
  )

idf_component_register(SRCS ${FastLED_SRCS}
                       INCLUDE_DIRS "src"
                       REQUIRES arduino)

project(FastLED) 

(sorry I can't raise a pull request at the moment, so raising it as an issue).
Thank you!

zackees, 2024-08-29:
That was me. I didn't realize this was needed. Do you have recommendations on how to test this in the future?

closed 2024-08-29 by zackees

zackees, 2024-08-29:
I've gone ahead and did a glob so that it picks up all files in that directory so that if we have any other cpp file changes then this won't need to be a problem again.

zackees, 2024-08-29:
I have no idea how to test this so let me know if this doesn't fix the issue.

ausculta, 2024-08-30:
The best way to test it may be to run CMake by compiling some code using the ESP-IDF (I accept that the ESP-IDF is an acquired taste!). There aren't many cpp files referenced in CMakeLists.txt, though, and this one was out of the way, so you've been a bit unlucky with that change. 
I fetched the change and did a clean build so the issue's fixed. I can't do a clean build with all yesterday's other commits as they take my binary over my partition size (1280K), but that is another issue.