FastLED/FastLED #1698: FastLED 3.7.5 explodes sketch size versus < 3.7.5 on ESP32

nomakewan, 2024-09-03:
Just updated to FastLED 3.7.5 and ran a compile and the compile failed due to taking up 113% of the available program space. Reverted to 3.7.4 and was back down to 97% where I was before.

3.7.5
`Sketch uses 1488525 bytes (113%) of program storage space. Maximum is 1310720 bytes.`
Compilation Log (Abbreviated): https://pastebin.com/96n5pWUh

3.7.4
`Sketch uses 1283293 bytes (97%) of program storage space. Maximum is 1310720 bytes.`
Compilation Log (Abbreviated): https://pastebin.com/YTGdJW59

205 kilobytes additional overhead seems a little excessive. Lots of warnings in the log too, but there were even more warnings in 3.7.4 so I suppose that much is improved.

Code: https://github.com/gpstar81/GPStar-proton-pack/tree/main/source/Attenuator
Board: Expressif Systems ESP32 Dev Module v3.0.4 (all default settings)
Arduino IDE 1.8.19

For our code to compile successfully, see the list of required libraries: https://github.com/gpstar81/GPStar-proton-pack/blob/main/ATTENUATOR_FLASHING.md#software-development-requirements

zackees, 2024-09-03:
Anyway you can help me out?

What is your chipset?

Can you comment out the new pallete high precision changes that went in as well as the new noise function?

This is probably what's taking up the room.

zackees, 2024-09-03:
I did a coarse grained seach for the offending CL and found it happened on Aug 29th. I'm doing a refined search to figure out what specific CL did this and will revert the change.

zackees, 2024-09-03:
Wow, it was because I pulled in the <iostream> and used std::court that the binary size blew up.

I've fixed this issue and will issue a new release either this week or the next that has this fix in it.

I've put in a builder to test the binary size and fail if it ever goes beyond an expected value.

closed 2024-09-03 by zackees

nomakewan, 2024-09-03:
Awesome as always, and thank you for the speedy fix!