FastLED/FastLED #1708: stroboscopic effect

saeedbs, 2024-09-08:
Thanks for this great project

I'm trying to find a way to have stroboscopic effect on pc case fans like this video : [QX fan](https://www.youtube.com/shorts/5xYnWsJUB9E)  or this: [stroboscopic effect ](https://www.youtube.com/shorts/drZOdRXKtfw)
I'm not sure but from my understanding this needs control over light frequency... is this possible with FastLED? if yes can u give some tips/example about it?

Do you think it is possible with ws2812b? 

from ws2812 datasheet: 
`Each pixel of the three primary color can achieve 256 brightness display, completed 16777216 color full color
display, and scan frequency not less than 400Hz/s.`    is this frequency what I'm looking for?

nomakewan, 2024-09-08:
This is the sort of question that would be better suited for the Reddit community: https://www.reddit.com/r/FastLED

The Issues page on Github is for reporting bugs/crashes/etc.

saeedbs, 2024-09-08:
ok sorry I'm closing this and asking there

closed 2024-09-08 by saeedbs