FastLED/FastLED #1710: Missing entries in CMakeLists.txt

ausculta, 2024-09-09:
Hello,
You're missing entries in the CMakeLists.txt file that, unfortunately, prevent successful compilation using ESP-IDF.
 src/rgb_2_rgbw.cpp is missing. Looking at the cpp files in /src, adding these 3 entries after the wiring.cpp entry fixes the problem. 

  src/rgb_2_rgbw.cpp
  src/five_bit_hd_gamma.cpp
  src/simplex.cpp

So the file now looks like this:
cmake_minimum_required(VERSION 3.5)

set(FastLED_SRCS
  src/bitswap.cpp
  src/colorpalettes.cpp
  src/colorutils.cpp
  src/FastLED.cpp
  src/hsv2rgb.cpp
  src/lib8tion.cpp
  src/noise.cpp
  src/platforms.cpp
  src/power_mgt.cpp
  src/wiring.cpp
  src/five_bit_hd_gamma.cpp
  src/rgb_2_rgbw.cpp
  src/simplex.cpp
  )

file(GLOB ESP32_SRCS "src/platforms/esp/32/*.cpp")
file(GLOB ESP32_LED_STRIP_COMPONENT_SRCS "src/platforms/esp/32/led_strip/*.cpp")
list(APPEND FastLED_SRCS ${ESP32_SRCS})

idf_component_register(SRCS ${FastLED_SRCS} ${ESP32_LED_STRIP_COMPONENT_SRCS}
                       INCLUDE_DIRS "src"
                       REQUIRES arduino)

project(FastLED)

Thanks for all the recent changes / improvements to FastLED. Much appreciated. :-)

zackees, 2024-09-09:
I went ahead and globed all the *.cpp files at the root of source and removed the stub main file.

closed 2024-09-09 by zackees

ausculta, 2024-09-09:
Yes, that works. Thanks a lot for making the change so quickly.