FastLED/FastLED #1713: Compilation error in version 3.7.5 and ESP-IDF to v4.3.2

alf45tar, 2024-09-11:
Version 3.7.5 cannot be used with ESP-IDF to v4.3.2 (https://github.com/platformio/platform-espressif32/releases/tag/v3.5.0) due to the following error.

```
In file included from /tmp/PedalinoMini/libdeps/esp32doit-devkit-v1/FastLED/src/platforms/esp/32/fastled_esp32.h:3:0,
                 from /tmp/PedalinoMini/libdeps/esp32doit-devkit-v1/FastLED/src/platforms.h:42,
                 from /tmp/PedalinoMini/libdeps/esp32doit-devkit-v1/FastLED/src/FastLED.h:67,
                 from /tmp/PedalinoMini/libdeps/esp32doit-devkit-v1/FastLED/src/noise.cpp:6:
/tmp/PedalinoMini/libdeps/esp32doit-devkit-v1/FastLED/src/platforms/esp/32/fastpin_esp32.h:7:32: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
 #pragma GCC diagnostic ignored "-Wvolatile"
                                ^
In file included from /tmp/PedalinoMini/libdeps/esp32doit-devkit-v1/FastLED/src/platforms/esp/32/idf4_rmt_impl.h:11:0,
                 from /tmp/PedalinoMini/libdeps/esp32doit-devkit-v1/FastLED/src/platforms/esp/32/idf4_rmt.cpp:9:
/tmp/PedalinoMini/libdeps/esp32doit-devkit-v1/FastLED/src/platforms/esp/32/clock_cycles.h:5:25: fatal error: hal/cpu_hal.h: No such file or directory
compilation terminated.
*** [/tmp/PedalinoMini/build/esp32doit-devkit-v1/libf2f/FastLED/platforms/esp/32/idf4_rmt.cpp.o] Error 1
```

ewowi, 2024-09-13:
I have the same :

```
In file included from .pio/libdeps/esp32dev/FastLED/src/FastLED.h:63,
                 from src/App/LedLayer.h:21,
                 from src/App/LedFixture.h:16,
                 from src/App/LedModEffects.h:14,
                 from src/main.cpp:36:
.pio/libdeps/esp32dev/FastLED/src/fastpin.h:13:32: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
 #pragma GCC diagnostic ignored "-Wvolatile"
                                ^~~~~~~~~~~~
```

But it is a warning - which is nicer to cleanup, but is not the reason for terminating

```fatal error: hal/cpu_hal.h: No such file or directory compilation terminated.```
seems to be the reason for you, this I do not have

zackees, 2024-09-16:
Okay I think I fixed it. Can't reproduce with our build system but I think I found the legacy header to include.

It's submitted to master but will be released on Monday or EOW.

Let me know if it's not fixed on master.

closed 2024-09-16 by zackees

tschamp31, 2024-09-17:
Had the same reference issue but different environment. This corrected it for me as well.
VSCode PlatformIO Same Espressif32 version.