FastLED/FastLED #1721: ESP32 bitbang (non-I2S) speed for WS2812

SinanAkkoyun, 2024-09-19:
Hey! I need to drive a lot of WS2812 LEDs at a high speed, so I need to split my display up into multiple chunks and use multiple GPIOs.

[_FL_DEFPIN(0);](https://github.com/FastLED/FastLED/blob/8ac3dd7f00e933a376530ecf86d360d167e1b82a/platforms/esp/32/fastpin_esp32.h#L75)

Could I use all of these pins at max speed (800 kbps, 30us per LED) with bit-banging? What are the limitations?

SinanAkkoyun, 2024-09-19:
I would need to drive 10 separate chunks. Or could you recommend a different microcontroller or solution? (like RMT into shift register)

zackees, 2024-09-20:
This is the only artist that I know of that has used (abused) the I2S protocol to hardware bitbang into shift registers. He calls them virtual pins, and he has managed to create 120 "virtual pins" that transpose and hardware bitbang into shift registers.

It's a board specific feature for the ESP32 S3 and has not be ported to any other board.

[[Yves-bazin](https://www.reddit.com/user/Yves-bazin/)](https://github.com/hpwit/I2SClocklessVirtualLedDriver)

GOOD LUCK IN YOUR PROJECT!!

And when you manifest your vision, please share with us on reddit of your adventures!

zackees, 2024-09-20:
If you just need 8 channels though, I would seriously consider the OctoWS2811 + Teensy 4.1 board. I've used it, it's great. And is stable.

closed 2024-09-20 by zackees

SinanAkkoyun, 2024-09-20:
Wow, you are awesome, thank you for providing all these invaluable resources!
I will try my luck :)

SinanAkkoyun, 2024-09-20:
I will! (I hope I'll find the time to finish it!)