FastLED/FastLED #1723: Wiki Page 'Interrupt problems' has an

DougF42, 2024-09-20:
On the 'Interrupt problems' page of the wiki,  in the `What does this practically mean, however?` section, this article, while discussing serial data rates states:

> At 57.6kbps you will receive 7200 bytes per second. Or, put differently, one byte every 138µs. 

This line should read:
`> At 57.6kbps you will receive 5700 bytes per second. Or, put differently, one byte every 174µs. `

The current numbers appear to be based on he assumption of 8 bits per character, which  is incorrect - the correct value is 10 bits per character (8 bits + 1 Start Bit + 1 Stop Bit).

There is also a second mention of  the ''138" value later in the same paragraph, which should adjusted accordingly.

zackees, 2024-09-21:
Feel free to issue a PR

nomakewan, 2024-09-23:
How would one issue a PR for the FastLED wiki?

DougF42, 2024-09-23:
Sorry for my ignorance -  I'm willing, but I'm not sure about how to do this for the Wiki...??  Not sure I have permissions??

zackees, 2024-09-24:
Done

DougF42, 2024-09-24:
Uh, almost - the data rate needs to be updated from` 7200 bytes per second` to` 5700 bytes per second`.

closed 2025-09-24 by zackees