FastLED/FastLED #1734: All releases past 3.7.6 are not compiling with my set up

met-hs, 2024-09-27:
Hello and thanks for your hard work and attention! Just reporting some odd thing with the new versions of the library. I can't compile the examples from the library anymore (at first I thought my code was broken, but the examples are not compiling either so something must be incompatible somewhere?). Was there a backwards incompatible change in this release? Maybe https://github.com/FastLED/FastLED/pull/1704? 

IDE Version: 2.3.3
CLI Version: 1.0.4
Arduino SAM Boards (32-Bit ARM Cortex-M3) - 1.6.12
Board - Arduino Due
FASTLED Version: 3.7.6 (3.7.5 works just fine so for now I rolled back to that version which unblocked me)

Try to compile FASTLED Example -> Blink.ino 

Result of compile and verify:
```
Using board 'arduino_due_x_dbg' from platform in folder: /Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12
Using core 'arduino' from platform in folder: /Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12

Detecting libraries used...
/Users/metanat/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON -DUSB_MANUFACTURER="Arduino LLC" -DUSB_PRODUCT="Arduino Due" -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x /private/var/folders/nl/s69x_7xd1pnclqbz7q8zdcq40000gr/T/arduino/sketches/B7176E64A549671ADD3912673CD8EBF6/sketch/Blink.ino.cpp -o /dev/null
Alternatives for FastLED.h: [FastLED@3.7.6]
ResolveLibrary(FastLED.h)
  -> candidates: [FastLED@3.7.6]
/Users/metanat/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON -DUSB_MANUFACTURER="Arduino LLC" -DUSB_PRODUCT="Arduino Due" -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x -I/Users/metanat/Documents/Arduino/libraries/FastLED/src /private/var/folders/nl/s69x_7xd1pnclqbz7q8zdcq40000gr/T/arduino/sketches/B7176E64A549671ADD3912673CD8EBF6/sketch/Blink.ino.cpp -o /dev/null
/Users/metanat/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON -DUSB_MANUFACTURER="Arduino LLC" -DUSB_PRODUCT="Arduino Due" -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x -I/Users/metanat/Documents/Arduino/libraries/FastLED/src /Users/metanat/Documents/Arduino/libraries/FastLED/src/FastLED.cpp -o /dev/null
/Users/metanat/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON -DUSB_MANUFACTURER="Arduino LLC" -DUSB_PRODUCT="Arduino Due" -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x -I/Users/metanat/Documents/Arduino/libraries/FastLED/src /Users/metanat/Documents/Arduino/libraries/FastLED/src/bitswap.cpp -o /dev/null
/Users/metanat/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON -DUSB_MANUFACTURER="Arduino LLC" -DUSB_PRODUCT="Arduino Due" -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x -I/Users/metanat/Documents/Arduino/libraries/FastLED/src /Users/metanat/Documents/Arduino/libraries/FastLED/src/colorpalettes.cpp -o /dev/null
/Users/metanat/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON -DUSB_MANUFACTURER="Arduino LLC" -DUSB_PRODUCT="Arduino Due" -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/ -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/Device/ATMEL/ -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x -I/Users/metanat/Documents/Arduino/libraries/FastLED/src /Users/metanat/Documents/Arduino/libraries/FastLED/src/colorutils.cpp -o /dev/null
/Users/metanat/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON -DUSB_MANUFACTURER="Arduino LLC" -DUSB_PRODUCT="Arduino Due" -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/libsam -I/Users/metanat/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/system/CMSIS/CMSIS/Include/  [... cut ...]
/Users/metanat/Documents/Arduino/libraries/FastLED/src/chipsets.h:88:53: error: expected ',' before ')' token
     static_assert(RGB == CONTROLLER::RGB_ORDER_VALUE);
                                                     ^
/Users/metanat/Documents/Arduino/libraries/FastLED/src/chipsets.h:88:53: error: expected string-literal before ')' token

Using library FastLED at version 3.7.8 in folder: /Users/metanat/Documents/Arduino/libraries/FastLED 
exit status 1

Compilation error: exit status 1
```

Thanks again!

met-hs, 2024-09-27:
Just in case this is the Blink.ino from examples for me:

```
/// @file    Blink.ino
/// @brief   Blink the first LED of an LED strip
/// @example Blink.ino

#include <FastLED.h>

// How many leds in your strip?
#define NUM_LEDS 1

// For led chips like WS2812, which have a data line, ground, and power, you just
// need to define DATA_PIN.  For led chipsets that are SPI based (four wires - data, clock,
// ground, and power), like the LPD8806 define both DATA_PIN and CLOCK_PIN
// Clock pin only needed for SPI based chipsets when not using hardware SPI
#define DATA_PIN 3
#define CLOCK_PIN 13

// Define the array of leds
CRGB leds[NUM_LEDS];

void setup() { 
    // Uncomment/edit one of the following lines for your leds arrangement.
    // ## Clockless types ##
    FastLED.addLeds<NEOPIXEL, DATA_PIN>(leds, NUM_LEDS);  // GRB ordering is assumed
    // FastLED.addLeds<SM16703, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<TM1829, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<TM1812, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<TM1809, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<TM1804, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<TM1803, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<UCS1903, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<UCS1903B, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<UCS1904, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<UCS2903, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<WS2812, DATA_PIN, RGB>(leds, NUM_LEDS);  // GRB ordering is typical
    // FastLED.addLeds<WS2852, DATA_PIN, RGB>(leds, NUM_LEDS);  // GRB ordering is typical
    // FastLED.addLeds<WS2812B, DATA_PIN, RGB>(leds, NUM_LEDS);  // GRB ordering is typical
    // FastLED.addLeds<GS1903, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<SK6812, DATA_PIN, RGB>(leds, NUM_LEDS);  // GRB ordering is typical
    // FastLED.addLeds<SK6822, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<APA106, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<PL9823, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<SK6822, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<WS2811, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<WS2813, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<APA104, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<WS2811_400, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<GE8822, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<GW6205, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<GW6205_400, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<LPD1886, DATA_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<LPD1886_8BIT, DATA_PIN, RGB>(leds, NUM_LEDS);
    // ## Clocked (SPI) types ##
    // FastLED.addLeds<LPD6803, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);  // GRB ordering is typical
    // FastLED.addLeds<LPD8806, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);  // GRB ordering is typical
    // FastLED.addLeds<WS2801, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<WS2803, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<SM16716, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);
    // FastLED.addLeds<P9813, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);  // BGR ordering is typical
    // FastLED.addLeds<DOTSTAR, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);  // BGR ordering is typical
    // FastLED.addLeds<APA102, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);  // BGR ordering is typical
    // FastLED.addLeds<SK9822, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);  // BGR ordering is typical
}

void loop() { 
  // Turn the LED on, then pause
  leds[0] = CRGB::Red;
  FastLED.show();
  delay(500);
  // Now turn the LED off, then pause
  leds[0] = CRGB::Black;
  FastLED.show();
  delay(500);
}
```

fermino, 2024-09-28:
It looks like C11 does not support `static_assert` without an error message. See https://stackoverflow.com/questions/58681558/expected-string-literal-before-when-using-static-assert-in-c17

fermino, 2024-09-28:
@met-hs please try this PR: #1735 ([repo here](https://github.com/fermino/FastLED/tree/patch-1)).

PR #1735 (2024-09-30): Fix static_assert call on C++11 (fixes #1734)
C/++11 does not support calling static_assert with no error message. This PR adds it. 
In all honest I did not completely understand the functioning so I just copied the message from above. 
Please feel free to edit it as deemed necessary :)

Thanks for all the awesome work in FastLED <3

EDIT: This fixes #1734.
commit 1dd05f05 (2024-09-30): Fix static_assert call on C++11 (fixes #1734) (#1735)

C/++11 does not support calling static_assert with no error message.
closed 2024-09-30 by zackees