FastLED/FastLED #1738: All releases past 3.7.5 are not compiling with ESP32/RMT driver/Expressif 1.0.x

Jueff, 2024-09-28:
Compiling the Blink sketch fails with **esp_idf_version.h: No such file or directory**

Board: ESP32DevBoard
FastLED: 3.7.8
esp32:esp32: Espressif 1.0.4

Message:
In file included from C:\...\FastLED\src/platforms/esp/32/fastled_esp32.h:12:0,
                 from C:\...\FastLED\src/platforms.h:42,
                 from C:\...\FastLED\src/FastLED.h:68,
                 from C:\...\FastLED\examples\Blink\Blink.ino:5:
C:\...\FastLED\src/platforms/esp/32/clockless_rmt_esp32.h:28:29: fatal error: esp_idf_version.h: No such file or directory
compilation terminated.
exit status 1

zackees, 2024-09-30:
Weird, is there an alternative to this header?

zackees, 2024-09-30:
Found the answer. It will be fixed in tomorrow release.

closed 2024-09-30 by zackees

zackees, 2024-10-01:
Release might be delayed for week.