FastLED/FastLED #1750: Arduino nano matter / MGM240s support

incapable, 2024-10-07:
When trying to compile after including the fastled library, the following error is shown:
"This platform isn't recognized by FastLED... yet.  See comments in FastLED/led_sysdefs.h for options."

The led_sysdefs.h indeed does not appear to specify the Silicon Labs MGM240s chip that's used for the Arduino nano matter (preview) board.

And so here's the request to, if at all possible, get support for this board/chip.

zackees, 2024-10-09:
I see that platformio supports this board a little bit:

https://github.com/platformio/platform-siliconlabsefm32

Do you happen to have platformio.ini configuration sitting around for this?

zackees, 2024-10-09:
Once we have a platformio.ini then I can put it into the test infrastructure and get a whole bunch of information about it and adding support will either be really easy or hard, depending on existing support that we have for this family type. It could be as easy sprinkling a few defines here or there to map in the pin definitions.

zackees, 2024-10-09:
Yeah it's starting to look like a no right now. If we can't get it into platformio then we can't even compile it.

How are you compiling this board?

https://github.com/SiliconLabs/arduino/issues/52#issuecomment-2117415159

zackees, 2024-10-09:
Okay i was able to generate some sort of build and it did fail. Can you determine if this matches your expectations:

*platformio*

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:hy_tinystm103tb]
platform = ststm32
board = hy_tinystm103tb
framework = arduino
build_flags = -fopt-info-all=optimization_report.txt
extra_scripts = C:\Users\niteris\dev\fastled\pre:lib\ci\ci-flags.py
```

Build error output:

```
In file included from C:/Users/niteris/dev/fastled/.build/hy_tinystm103tb/src/Apa102.ino:3:
lib\src/FastLED.h:827:2: warning: #warning "No pin/port mappings found, pin access will be slightly slower. See fastpin.h for info." [-Wcpp]
  827 | #warning "No pin/port mappings found, pin access will be slightly slower. See fastpin.h for info."
      |  ^~~~~~~
In file included from lib\src/FastLED.h:81:
lib\src/fastspi.h:163:23: note: '#pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output'
  163 | #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from lib\src/FastLED.h:64:
lib\src/fastpin.h: In instantiation of 'class FastPin<1>':
lib\src/fastspi_bitbang.h:27:49:   required from 'class AVRSoftwareSPIOutput<1, 2, 12>'
lib\src/fastspi.h:40:7:   required from 'class SPIOutput<1, 2, 12>'
lib\src/chipsets.h:327:6:   required from 'class APA102Controller<1, 2, RGB, 12, kFiveBitGammaCorrectionMode_Null, 0, 4278190080>'
lib\src/FastLED.h:468:34:   required from 'CLEDController& CFastLED::addLeds(CRGB*, int, int) [with ESPIChipsets CHIPSET = APA102; unsigned char DATA_PIN = 1; unsigned char CLOCK_PIN = 2; EOrder RGB_ORDER = RGB]'
C:/Users/niteris/dev/fastled/.build/hy_tinystm103tb/src/Apa102.ino:16:66:   required from here
lib\src/fastpin.h:274:31: error: static assertion failed: Invalid pin specified
  274 |         static_assert(validpin(), "Invalid pin specified");
      |                       ~~~~~~~~^~
lib\src/fastpin.h:274:31: note: 'FastPin<1>::validpin()' evaluates to false
lib\src/fastpin.h: In instantiation of 'class FastPin<2>':
lib\src/fastspi_bitbang.h:28:50:   required from 'class AVRSoftwareSPIOutput<1, 2, 12>'
lib\src/fastspi.h:40:7:   required from 'class SPIOutput<1, 2, 12>'
lib\src/chipsets.h:327:6:   required from 'class APA102Controller<1, 2, RGB, 12, kFiveBitGammaCorrectionMode_Null, 0, 4278190080>'
lib\src/FastLED.h:468:34:   required from 'CLEDController& CFastLED::addLeds(CRGB*, int, int) [with ESPIChipsets CHIPSET = APA102; unsigned char DATA_PIN = 1; unsigned char CLOCK_PIN = 2; EOrder RGB_ORDER = RGB]'
C:/Users/niteris/dev/fastled/.build/hy_tinystm103tb/src/Apa102.ino:16:66:   required from here
lib\src/fastpin.h:274:31: error: static assertion failed: Invalid pin specified
lib\src/fastpin.h:274:31: note: 'FastPin<2>::validpin()' evaluates to false
Compiling .pio\build\hy_tinystm103tb\libca7\src\FastLED.cpp.o
*** [.pio\build\hy_tinystm103tb\src\Apa102.ino.cpp.o] Error 1
In file included from lib\src\FastLED.cpp:2:
lib\src\FastLED.h:827:2: warning: #warning "No pin/port mappings found, pin access will be slightly slower. See fastpin.h for info." [-Wcpp]
  827 | #warning "No pin/port mappings found, pin access will be slightly slower. See fastpin.h for info."
  ```

zackees, 2024-10-09:
Confirmed the support is for arduino only.

Here is the package information for this board:

https://siliconlabs.github.io/arduino/package_arduinosilabs_index.json

And this is the relevant snippets:

```json
{
                    "category": "Silicon Labs",
                    "name": "Silicon Labs",
                    "url": "https://github.com/SiliconLabs/arduino/releases/download/2.0.0/silabs_arduino_core-2.0.0.zst",
                    "version": "2.0.0",
                    "architecture": "silabs",
                    "archiveFileName": "silabs_arduino_core-2.0.0.zst",
                    "boards": [
                        {
                            "name": "Arduino Nano Matter"
                        },
                        {
                            "name": "SparkFun Thing Plus Matter"
                        },
                        {
                            "name": "xG24 Explorer Kit"
                        },
                        {
                            "name": "xG24 Dev Kit"
                        },
                        {
                            "name": "xG27 Dev Kit"
                        },
                        {
                            "name": "BGM220 Explorer Kit"
                        }
                    ],
                    "toolsDependencies": [
                        {
                            "packager": "SiliconLabs",
                            "version": "12.2.rel1",
                            "name": "gcc-arm-none-eabi"
                        },
                        {
                            "packager": "SiliconLabs",
                            "version": "1.16.4",
                            "name": "simplicitycommander"
                        },
                        {
                            "packager": "SiliconLabs",
                            "version": "0.12.0-arduino1-static",
                            "name": "openocd"
                        }
                    ],
                    "help": {
                        "online": "https://github.com/SiliconLabs/arduino"
                    },
                    "size": 320160894,
                    "checksum": "SHA-256:75b9690f628cd9334c055ddb02ad559236732f3997ef59c87e833698830f19c7"
                },
```

incapable, 2024-10-09:
Ok that's a lot of info, I'm really new to the arduino ecosystem so forgive me for lacking some knowledge, but I'll help where I can.
As i said, Fastled fails fast on a compatible board check, that's how far i got with that.

At this point I'm still developing in the Arduino IDE, getting to know the platform, but i did encounter the platform.io libraries where i got roughly the same results you did.

> Confirmed the support is for arduino only.
You mean just the Arduino IDE environment here?

Is there anything specific I can run/find out for you?

I'm aware it's technically a preview board and third party support might be a bit troubling.

zackees, 2024-10-10:
Here's the problem. Arduino hides all this information from you and you are going to have to look really deep to figure out what it's doing.

On the plus side, Arduino supports your board and platformio does not, which means we are flying blind here.

I would normally recommend you use Platformio + VS Code to do your development, but you won't be able to hit the compile button and get it to work. However, VS code is a much better code editor and seems pretty smart at right clock to follow a symbol, which Arduino does not.

Anyway, I'll keep an eye out for pull request for a fix. I'm sorry your board doesn't work out of the box. But it's likely that some combination of defines will get it at least working.

```C++
FASTLED_ALL_PINS_HARDWARE_SPI
FASTLED_FORCE_SOFTWARE_PINS
FASTLED_FORCE_SOFTWARE_SPI
```

eMUQI, 2024-10-14:
HI, @zackees 

Could you please clarify how to use these defines?

> Anyway, I'll keep an eye out for pull request for a fix. I'm sorry your board doesn't work out of the box. But it's likely that some combination of defines will get it at least working. Good luck, happy coding and check back with us if you have a revelation on what is going on.
> 
> ```c++
> FASTLED_ALL_PINS_HARDWARE_SPI
> FASTLED_FORCE_SOFTWARE_PINS
> FASTLED_FORCE_SOFTWARE_SPI
> ```

I added the following lines before `#include <FastLED.h>`:

```
#define FASTLED_ALL_PINS_HARDWARE_SPI
#define FASTLED_FORCE_SOFTWARE_PINS
#define FASTLED_FORCE_SOFTWARE_SPI
```

Unfortunately, I encountered the same result as @incapable each time. Am I possibly misunderstanding their intended usage? Any clarification would be greatly appreciated!

zackees, 2024-10-14:
That's the correct way to do it.

Unfortunately our build system for testing platforms is platformio and this board is not supported by them nor have the makers provided a boards.json or platformio.ini file to make it work.

I attempted to make this myself but there is so little information out there that i came up empty handed.

So unfortunately I am going to have to wait until such a configuration becomes available. Then the support process can begin.

zackees, 2024-10-29:
> HI, @zackees
> 
> Could you please clarify how to use these defines?
> 
> > Anyway, I'll keep an eye out for pull request for a fix. I'm sorry your board doesn't work out of the box. But it's likely that some combination of defines will get it at least working. Good luck, happy coding and check back with us if you have a revelation on what is going on.
> > ```c++
> > FASTLED_ALL_PINS_HARDWARE_SPI
> > FASTLED_FORCE_SOFTWARE_PINS
> > FASTLED_FORCE_SOFTWARE_SPI
> > ```
> 
> I tried adding each of the following lines, one at a time, before `#include <FastLED.h>`:
> 
> ```
> #define FASTLED_ALL_PINS_HARDWARE_SPI
> #define FASTLED_FORCE_SOFTWARE_PINS
> #define FASTLED_FORCE_SOFTWARE_SPI
> ```
> 
> Unfortunately, I encountered the same result as @incapable each time. Am I possibly misunderstanding their intended usage? Any clarification would be greatly appreciated!

Because SPI uses specialized hardware, there are only one or few channels available on mcus. If you don't set the right pins then FastLED will refuse to compile.

These defines are used to get around that and force FastLED to accept either all pins as hardware, regardless of whether they are or not, or to use software bitbanging which is much slower.

zackees, 2025-01-11:
Update: looks like nano_every might be supported now:

https://community.platformio.org/t/support-for-arduino-nano-matter/44951/10

zackees, 2025-09-13:
Implemented in https://github.com/FastLED/FastLED/commit/bac25ca768389793f4fdc5ee93c600e9682a655b

It's mostly vibe coded in with claude, no idea if it works. please test it and if it's not working, re-open bug.

closed 2025-09-13 by zackees

zackees, 2025-09-13:
You got me